RobotLocomotion/drake #15177: python: passing the wrong context to OutputPort::Eval crashes the python interpreter

RussTedrake, 2021-06-11:
I've seen this time and again (a user's juptyer server / colab notebook crashes because they passed the wrong context).  I just did it to myself, so decided to make a minimal repro.  I've seen it most often when the context is an `MultibodyPlant` or `SceneGraph`, but created an example below using just a `Gain` which supports my thinking that it is a framework issue, not a system-specific issue.

```
from pydrake.all import DiagramBuilder, Gain, ConstantVectorSource

builder = DiagramBuilder()
gain = builder.AddSystem(Gain([1.0]))
source = builder.AddSystem(ConstantVectorSource([1.0]))
builder.Connect(source.get_output_port(), gain.get_input_port())
diagram = builder.Build()

context = diagram.CreateDefaultContext()

ok = gain.get_output_port().Eval(gain.GetMyContextFromRoot(context))

# This causes the python interpreter to crash:
crash = gain.get_output_port().Eval(context)
```

Expected behavior: The last line should throw an exception (probably `AssertionError`).
Current behavior: The python interpreter crashes.

cc @sherm1 as team owner.
@jwnimmer-tri -- hoping you might have a suggestion/solution.

jwnimmer-tri, 2021-06-11:
\CC @rpoyner-tri because this is supposed to be caught by our #12560 work.

Later today, I'll see what I can find out from a debugger.

jwnimmer-tri, 2021-06-11:
It segfaults in Release builds but raises a nice exception in Debug builds.

This is the line that's helping us out in Debug builds.  It's guarded by `DRAKE_ASSERT_VOID`.
https://github.com/RobotLocomotion/drake/blob/v0.30.0/systems/framework/output_port.h#L118

Perhaps none of the calls to `ValidateContext` should ever be within `DRAKE_ASSERT_VOID`.  Most of them are not guarded (only 8 are turned off in Release builds).  I'll have to go back and look why we did it that way.

PR #15181 (2021-06-16): [framework] Enable context validation for ports in Release builds
Doing this without eating a virtual function call means that we need to pass the SystemId into the Port objects so that they can perform the guard locally, without asking SystemMessageInterface.

Add some Debug-only context validation checks to CacheEntry, to help backstop the rare cases where a user might be evaluating their cache entry without first checking the Context they were given.

Closes #15177.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15181)
closed 2021-06-16 by sammy-tri