RobotLocomotion/drake #15178: Change the decision variable type from binary to continuous after the variable has been created

hongkai-dai, 2021-06-11:
As @TobiaMarcucci mentioned, when using gurobi's python interface directly, one user scenario is to change the type of the variable after it has been created, something like
```py
# Create a binary variable
x = gurobi_model.addVars(1, var_type=gurobipy.GRB.BINARY)
# Change the variable type to continuous
x.var_type = gurobipy.GRB.CONTINUOUS
```
Tobia would like to emulate the similar workflow in Drake's MathematicalProgram.

To do so in Drake, the biggest problem is that Drake creates decision variables, that both stored inside MathematicalProgram, and returned a copy of the variable to the user. If we change either the variable stored inside MathematicalProgram or the returned variable, the other one won't change, causing inconsistency, as shown below
```py
# `prog` stores the newly created variable internally, and also returns a copy of the variable to the user as `x`.
x = prog.NewBinaryVariable(1)
# Now change the variable type of `x`.
x[0].set_type(Variable.Continuous)
# The variable in the `prog` is unchanged
print(prog.decision_variable(prog.GetVariableIndex(x[0])).var_type())  # This will print binary.
```

The problem is that we returned the **copy** of the variable in `MathematicalProgram::NewContinuousVariable`** and `MathematicalProgram::NewBinaryVariable`, the reason why we decided to return the copy is that `symbolic::Variable` type is effectively immutable after creation, it doesn't have a setter method. Now if we add a setter method `set_type` to `symbolic::Variable`, we cause the issue that the returned variable and the variable stored inside MathematicalProgram can become inconsistent.

@soonho-tri @jwnimmer-tri do you think we have an easy fix here? All the fix I can think of are too complicated, like returning shared pointers in `MathematicalProgram::NewContinuousVariable`.

jwnimmer-tri, 2021-06-11:
I think if you let go of the idea of mutating variables, and instead think more about replacing some variables with new variables, you might find more suitable answers.

We have `NewContinuousVariables` and `NewBinaryVariables` -- maybe you want something like `ReplaceVariables(list_of_existing_decision_variables, list_of_new_variables)`?  The number of decision variables doesn't change, but we go through and edit any bindings that used the old ones to use the new ones, and we update the required capabilities to make sure whether or not it contains `kBinaryVariable` is still accurate.

hongkai-dai, 2021-06-11:
I think that could work, but we need to warn the user that after calling `ReplaceVariables`, the old variables cannot be used. Moreover, the returned bindings from `AddCost` and `AddConstraint` won't be used, if these bindings are associated with the old variables. I am afraid it is too hard for the user to keep track of all the variables/cost_binding/constraint_binding that would become obsolete? For example, if we have
```py
x_old = prog.NewContinuousVariable(1)
cnstr = prog.AddLinearConstraint(x_old[0] == 3)
x_new = prog.NewBinaryVariable(1)
prog.ReplaceVariables(x_old, x_new)
# Now cnstr can't be used, as it binds to the old variable x
```

jwnimmer-tri, 2021-06-11:
Oh, I see.  My expectation was that we'd mutate the Binding during `ReplaceVariables`, so that it would continue to work correctly, but I see now that we're returning copies of the binding (why?!), not references.  So, the idea is problematic, for the reason you note.

hongkai-dai, 2021-06-11:
> but I see now that we're returning copies of the binding (why?!)

Binding has a shared_ptr to the constraint, and a copy of the bound variables. We thought we would change the constraint, but not the bound variables when we created MathematicalProgram::AddConstraint function.

Maybe the right approach is to change the return type of `AddConstraint` to reference (or pointers)?

soonho-tri, 2021-06-12:
@hongkai-dai and @jwnimmer-tri ,

I have another proposal which changes the type of `type_` member:

```c++
class Variable {
  // ...
 private:
  std::shared_ptr<Type> type_;
}
```

The idea is that all variable instance with the same ID share a single type value.

jwnimmer-tri, 2021-06-12:
It's a bit of a delicate situation.  The MathematicalProgram has already updated its internal data structures based on the original type of the variable.  If we change the variable type out from under it, those structures will be incorrect and it will fail.  The same could be said for any other class in Drake that operates on variables; if they mutate out from under it, any checking it did when it first received the variable is now invalid.

At best, that means we must rely on the user to call an method on MathematicalProgram to change the type (and then MathematicalProgram would mutate the variable); ditto for other variable-using classes.  At worst, it could lead to subtle bugs in user code when they don't fully appreciate the action-at-a-distance effects of changing the type on what appears to be a _copy_ of a variable, but is really an _alias_ to some notion of a global variable.

I think we are not in the realm of "easy fix" anymore, as mentioned in the top post.

It's probably worth stepping back and considering a _fully worked_ example of what we want to accomplish, i.e., a complete program.  Is there a toy program we could consider in the discussion here?  The original example only has a snippet, so the important relationship with Binding is not shown.  I think if we had a full program in front of us, that would help more clearly explain the problem, and allow us to measure potential solutions.

hongkai-dai, 2021-06-12:
Thanks @jwnimmer-tri and @soonho-tri , I think I have a candidate plan

1. We change `symbolic::Variable::type_` to a shared pointer, so now all variables with the same ID would share the same variable type.
2. We add a **private** setter to change the variable type `void Variable::set_type(Type type)`. This private setter can only be used by some methods in MathematicalProgram.
3. In MathematicalProgram, we add functions `MathematicalProgram::ChangeVariableType(const VectorX<symbolic::Variable>& var, Type new_type)`. This function would change the variable type, and also reparse the optimization problem (like checking whether we require binary capability after the variable type is changed). This function has access to the private setter `Variable::set_type(Type new_type)`.

In order to grant access to the private setter, I think I can use the[ "passkey" idiom](https://www.spiria.com/en/blog/desktop-software/passkey-idiom-and-better-friendship-c/), so that only `MathematicalProgram::ChangeVariableType()` function has access to the private setter, and it has access to only this private setter, not other private members.

@jwnimmer-tri @soonho-tri what do you think?

@jwnimmer-tri for the complete program example, do you want one that works and runs, but doesn't change the variable type? If you would like to see a working one that also changes the variable type, I can code it using gurobipy API directly, but I can't do it with the current Drake API.

soonho-tri, 2021-06-16:
Sorry for the delay, @hongkai-dai.

I see what and how you want to achieve things. I think here we assume that we have a single MathematicalProgram instance which holds the variables that we want to change their types. However, it is possible that we have two (or more) MP instances (say `mp1` and `mp2`) which share common variables introduced via `MP::AddDecisionVariables` method. If we call `mp1.ChangeVariableType(var, new_type)`, it will update the type of `var` and update `mp1` to maintain its internal invariants. But `mp2` will not be updated, which can cause a problem.

hongkai-dai, 2021-06-17:
Thanks Soonho.

Yes that is a very good example, great catch. Unfortunately I don't have a good solution. One solution is that before we check the required capabilities of a `MathematicalProgram` instance, we loop through all variables again to make sure if `ProgramAttribute::kBinaryVariable` is set correctly. Maybe in MathematicalProgram, instead of storing `required_capabilities_`, and having the getter function
```cc
MathematicalProgram::required_capabilities(){
  return required_capabilities_;
}
```
we should instead ask `MathematicalProgram::required_capabilities()` to check all the variables and constraints.

soonho-tri, 2021-06-17:
I see. Essentially, we need to update those code written on the assumption that `symbolic::Variable` is immutable. Then we can add `Variable::set_type` method which changes the type of the variables with the same ID.

I also would like to check if we have https://mywiki.wooledge.org/XyProblem here. In the original problem, there could be other solutions than providing `Variable::set_type`. As Jeremy requested above, if you can share the original problem with an example, that should help us investigate this problem from different angles.

jwnimmer-tri, 2021-06-22:
Still curious to hear the full work-up, but here's another way we could tackle it (tossing an idea out into the wind) ...

Instead of mutating the Variable, or swapping it out for another replacement (and teaching all of the Bindings about the swap), I wonder if we could always declare it as CONTINUOUS (the default), and then _later_ tell MathematicalProgram that certain continuous variables used in in the program should be constrained to be BINARY (or INTEGER) when solving, without actually changing their intrinsic type.

So something like:
```c++
/** Changes the apparent type of decision variables as seen by the solver back-end
to be new_type instead of their current type.  This is useful for swapping between
continuous, binary, and integer variables during optimization.
*/
void MathematicalProgram::SetEffectiveDecisionVariableType(
    const VectorDecisionVariable&, Variable::Type new_type);
```

The solver back-ends would need to call a new function `prog.get_apparent_decision_variable_type(i)` instead of the plain `prog.decision_variable(i).get_type()` .

Then inside the mathematical program, we have to keep careful track of program attributes.

Aside: It looks to me like we have a pre-existing bug -- the function `MathematicalProgram::NewVariables` sets `ProgramAttribute::kBinaryVariable` when adding binary variables, but `MathematicalProgram::AddDecisionVariables` does not check for incoming binary variables, so our program attributes and `ChooseBestSolver` will be wrong in this case.

hongkai-dai, 2021-06-22:
Thanks @soonho-tri and @jwnimmer-tri . I am working on a conference deadline. Would you mind if I come back to this issue later in the week?

soonho-tri, 2021-06-22:
@hongkai-dai , no problem. Good luck on your submission!