RobotLocomotion/drake #15184: solvers: API proposal for passing structured data to generic costs/constraints

RussTedrake, 2021-06-13:
@weigao95 proposed to me that at little improvement here could make a dramatic improvement in the usability of MathematicalProgram.  After hearing his proposal and thinking about what we have currently and in the pipeline (#14971 and #14973), here is my opening suggestion.  

Note: I recommended we make it nice in Python first, since we already have `namedview`, and have discussed making something similar to that work for c++, too.

The problem that we want to solve is that authors of generic costs/constraints currently need to pack their decision variables into a vector and then unpack them inside the function implementing the cost/constraint.  In the best case, this leads to code like:
```
def angular_momentum_constraint(vars):
	q, com, Hdot, contact_force = np.split(vars, [nq, nq+3, nq+6])
			...

prog.AddConstraint(angular_momentum_constraint, lb=np.zeros(3), ub=np.zeros(3),  vars=np.concatenate((q[:,n], com[:,n], Hdot[:,n], Fn)))
```
But oftentimes, I've seen people write code referencing e.g. `vars[32]` inside the constraint implementation.

This is in stark contrast to authoring e.g. convex constraints, where the user has the nicely curated symbolic decision variables to work with.  For instance, right next to that `angular_momentum_constraint` example from my [littledog notebook](https://github.com/RussTedrake/underactuated/blob/master/examples/littledog.ipynb) is another line:
```
prog.AddConstraint(eq(H[:,n+1], H[:,n] + h[n]*Hdot[:,n]))
```
which is (almost) beautiful.

Proposal:
Add `MathematicalProgram.AddCost(func_handle, lb, ub, namedview)` as an additional entry point, and the same for `AddConstraint`.  The example above could then look like:
```
def angular_momentum_constraint(vars):
	vars.q[2] + vars.com[3], ...
			...

AngularMomentumConstraintView = NamedView('angular_momentum_constraint', 'q', 'com', 'Hdot')
prog.AddConstraint(angular_momentum_constraint, lb=np.zeros(3), ub=np.zeros(3),  vars=AngularMomentumConstraintView(q[:,n], com[:,n], Hdot[:,n], Fn))
```
I suspect we can turn that into a one-liner with a little sugar.

Alternatively (or additionally), even with the current entry point we could make a `namedview` using the string names of the decision variables as they are already passed into the method to construct a `namedview` internally. In a simpler case we would have 
```
def mycost(vars):
	vars.x0
	vars.x1
	vars.y0
	...

prog.AddCost(mycost, lb, ub, [x,y])
```
Two big caveats: 1) this would add a new requirement that the decision variable names are unique.  2) this would be more fragile in the case of the `angular_momentum_constraint` above, since I use the same constraint function on a variety of different decision variables (one for each time step), and the variable names would be different each time.

@weigao95 and I would welcome your feedback / alternative ideas.  cc @hongkai-dai

weigao95, 2021-06-14:
I think that's a nice solution. Besides, I am considering the trade-off of including the time (knot) within the namedview index? For example we might write `vars.q.1 - vars.q.0`.

As mentioned by Russ, there might be a variety of different, but very similar decision variables (one for each time step). That's a good reason to include the knot as part of the index. Furthermore, including the knot also simplify the indexing in the `MathmaticalProgram` if we decide to use namedview as the variable index (at least for sequential planning problem). On the other hand, `MathmaticalProgram` is used in many other applications without a sequential structure. Requesting unique names and knots seems a little bit strange for these applications.

Maybe we should define a new class, for instance `TrajectoryOptimization`, that uses the new index scheme and owns a `MathmaticalProgram`? This design is very similar to Hongkai's implementation of `InverseKinematics`. 

I am happy to type an initial implementation once we agree on the design.

hongkai-dai, 2021-06-14:
I like the `namedview` idea, it would make the code much more clear.

I think the second approach (using variable name) is harder. As you point out, it requires the variables having unique names. This is never enforced in our code, and in some cases we don't provide the API for the user to set the variable name (For example, in NewSosPolynomial() function, it adds the polynomial Gram matrix as decision variables, but the variable name are all hard-coded to `Symmetric` in this function).

RussTedrake, 2021-06-14:
@hongkai-dai ... thanks.  i agree.
@weigao95.. awesome.  But we should definitely *not* write a new TrajectoryOptimization class.  I would rather pull the trigger and refactor MultipleShooting into the has-a relationship.  See the slack thread here: https://drakedevelopers.slack.com/archives/C2ETTRXT7/p1586869334012700

weigao95, 2021-06-14:
> @hongkai-dai ... thanks. i agree.
> @weigao95.. awesome. But we should definitely _not_ write a new TrajectoryOptimization class. I would rather pull the trigger and refactor MultipleShooting into the [...]

I am happy with that. Then I will take a look at the C++ code (which I am more familiar with).

RussTedrake, 2021-06-20:
Related to #9497

weigao95, 2021-07-26:
I would like to document a discussion with Prof. Russ about the Hessian/Quadratic term in this new indexing scheme. 

If we have a new constraint interface where variables are indexed by (string, knot) pairs, then when a user implementing his constraints, for instance

```python
class MyConstraint(ConstraintBase):
    def __init__(self, q_key=("q", 0), v_key=("v", 0)):
        self._q_key = q_key
        self._v_key = v_key
```
When actually implementing this constraint, the user might write
```python
    def evaluate(self, problem):
        q_vec = problem.get_vector(self._q_key)
        v_vec = problem.get_vector(self._v_key)
        constraint_value, jacobian_wrt_q, jacobian_wrt_v = do_some_computation(q_vec, v_vec)
        jacobian_map = {}
        jacobian_map[self_.q_key] = jacobian_wrt_q
        jacobian_map[self_.v_key] = jacobian_wrt_v
        return constraint_value, jacobian_map
```
As shown in the code block above, the jacobian must be indexed by the (string, knot) pairs. This is different from returning a big matrix in the old `Constraint::Eval` interface (the one with Matlab, before adapting to the current `MathematicalProgram` with the autodiff vector input). This `jacobian_map` works well for jacobian, **but it can be messy for Hessian**.

After discussion with Russ, we conclude:

- We do not need to provide the interface for the Hessian matrix. Most solvers do not use Hessian (to my knowledge, only Iport has optional Hessian interfaces).
- One important use case of the Hessian matrix is to construct a quadratic term `x_T*Q*x`. However, we can implement this using the `MathematicalProgram` interface.