RobotLocomotion/drake #15206: mixed integer rotation constraint test error

jhu-gh, 2021-06-17:
After installing drake from source, tried to run mixed_integer_rotation_constraint_test, and got the following error:
`[==========] Running 244 tests from 3 test suites.
[----------] Global test environment set-up.
[----------] 48 tests from RotationTest/TestMixedIntegerRotationConstraintGenerator
[ RUN      ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/0
[       OK ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/0 (44 ms)
[ RUN      ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/1
[       OK ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/1 (20 ms)
[ RUN      ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/2
[       OK ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/2 (84 ms)
[ RUN      ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/3
[       OK ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/3 (84 ms)
[ RUN      ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/4
[       OK ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/4 (22 ms)
[ RUN      ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/5
[       OK ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/5 (22 ms)
[ RUN      ] RotationTest/TestMixedIntegerRotationConstraintGenerator.TestConstructor/6
Restoration phase is called at point that is almost feasible,
  with constraint violation 1.110223e-16. Abort.
abort: Failure at solvers/mixed_integer_rotation_constraint_internal.cc:218 in ComputeHalfSpaceRelaxationForBoxSphereIntersection(): condition 'result.is_success()' failed.
Aborted (core dumped)
`
I'm using Ubuntu 18, and followed the installation instructions on drake website.  Can anyone give a pointer what the issue could be? I'm especially interested in using the mixed_integer_rotation_constraint. Thanks.

hongkai-dai, 2021-06-17:
Your test incurs IPOPT solver. The optimization problem is convex. It seems that you don't run the test with Mosek/Gurobi enabled, and Drake falls back to IPOPT which doesn't perform nicely with this convex optimization problem.

What is the command you used for testing? I tried
```
bazel test --config gurobi --config mosek --test_tag_filters=gurobi,mosek //solvers:mixed_integer_rotation_constraint_test
```
and it works fine on my Ubuntu 18 machine. Note the flag `--config gurobi --config mosek --test_tag_filters=gurobi,mosek` enables both Gurobi and Mosek.

jhu-gh, 2021-06-17:
> Your test incurs IPOPT solver. The optimization problem is convex. It seems that you don't run the test with Mosek/Gurobi enabled, and Drake falls back to IPOPT which doesn't perform nicely with [...]

Thanks for your quick reply.

You're correct, it is because of the solver, I was running the binary directly without any flags, tried your command and it worked! Thanks.

closed 2021-06-17 by jhu-gh