RobotLocomotion/drake #15213: We have `SolverType::kDReal`, solver class `DrealSolver`, and `solver.SolverName="dReal"`

RussTedrake, 2021-06-20:
The capitalization is inconsistent.  Proposal: should we at least make `SolverType` into `kDreal` to match the class name?

(I'm writing the python bindings now, and will PR them soon)

soonho-tri, 2021-06-20:
> should we at least make SolverType into kDreal to match the class name?

Yes, I agree.

jwnimmer-tri, 2021-06-20:
Since SolverType only exists for backwards compatibility and should not be used in new code (#15214), renaming `SolverType::kDreal` without deprecation defeats the purpose, and going through deprecation seems like more work that it's worth.

If we find the naming conventions of SolverType distracting, then we should  deprecate and remove it entirely, not try to fix it.

closed 2021-06-20 by RussTedrake