RobotLocomotion/drake #15216: pydrake: Segfault when using `RigidTransform_[Expression].multiply(VectorXDecisionVariables)`

RussTedrake, 2021-06-20:
This one is a blocker for me.  Would really appreciate some help:
```
from pydrake.all import RigidTransform_, MathematicalProgram, Expression

prog = MathematicalProgram()
p_A = prog.NewContinuousVariables(3)
X_WA = RigidTransform_[Expression]()

print(X_WA.multiply([0,0,0]))
print(X_WA.multiply(p_A)). # <-- this line segfaults
```
results in 
```
[<Expression "0"> <Expression "0"> <Expression "0">]
zsh: segmentation fault  python3 test.py
```

RussTedrake, 2021-06-20:
Update: It seems that it's a `Variable` vs `Expression` issue.  `X_WA.multiply(p_A+0)` works just fine.

EricCousineau-TRI, 2021-06-22:
Yup, can repro on simplified form: https://github.com/EricCousineau-TRI/drake/tree/issue-15216-wip
```py
p_A_var = np.array([Variable("x0"), Variable("x1"), Variable("x2")])
to_sym = np.vectorize(Expression)
p_A_sym = to_sym(p_A_var)
X_WA = RigidTransform_[Expression]()
print(X_WA.multiply(p_A_sym))
print(X_WA.multiply(p_A_var))  # <-- this line segfaults
```

EricCousineau-TRI, 2021-06-22:
Seems like there's an issue with implicit casting on the pybind11 side: 6b9d70b9db
Most likely, an error is raised, but something in our pybind11 fork fumbles with it, and it gets lost.

rpoyner-tri, 2025-03-19:
This one still exists. In debug builds, it yields some Eigen assertion, so it can probably be tracked down.