RobotLocomotion/drake #15229: IPOPT solver test uses hard-coded constants

ggould-tri, 2021-06-22:
See:

https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-big-sur-unprovisioned-clang-bazel-nightly-release/133/console

This mac build failed because upstream ipopt has changed the values of its constants by adding a new constant ( https://github.com/coin-or/Ipopt/commit/5381b9a4908757dd54a2591d5f323e53814e888c#diff-96bd8208972238902c77b11d84b51981fbb4e4f2cdc73c057f5e8b9c21f29e02R38 ) and our test uses the numerical value of the constant in order to avoid including a header that may not be present.

The correct solution per @jwnimmer-tri is:
```
The form of the fix would be to edit the BUILD file for the test so say something like:
copts = select({
        "//tools:no_ipopt": [
            "-DDRAKE_IPOPT_OFF",
        ],
        "//conditions:default": [
            "-DDRAKE_IPOPT_ON",
        ],
    }),
and then in the test program, do something like if DRAKE_IPOPT_ON, then #include the ipopt header and add some new constexpr globals for the constants, or if DRAKE_IPOPT_OFF, then define the same constants but set them to zero (without including the header), which is fine because the test will be compiled but not run. (edited) 
```
However as a temporary buildcop fix we will disable the two tests that redefine IPOPT constants.

jwnimmer-tri, 2021-06-28:
@hongkai-dai would you like me to take over this issue?

hongkai-dai, 2021-06-28:
> @hongkai-dai would you like me to take over this issue?

Thanks Jeremy, that would be great.

PR #15249 (2021-06-28): [solvers] Remove hard-coded IPOPT enums from tests
Closes #15229.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15249)
closed 2021-06-28 by SeanCurtis-TRI