RobotLocomotion/drake #15233: [multibody] Add unit test for derivatives on hydroelastic contact forces

SeanCurtis-TRI, 2021-06-23:
Resolving issue #14136 included all of the work on the geometry side to get derivatives in hydroelastic contact. It included a token effort to give evidence that the derivatives propagate through to the computed forces. However, the test is only indicative and needs to be made more robust.

See the comments in #15219 for details on how to formulate a more detailed test that fully considers the expected derivative values.

amcastro-tri, 2021-12-06:
anything you think we'd need to test @SeanCurtis-TRI that #14963 does not cover? the traction calculator is covered in that PR. I do agree however that we do not seem to have an end-to-end unit test.

SeanCurtis-TRI, 2021-12-06:
My best sense of this issue is that we should have confidence that the derivatives on the forces through hydroelastic contribution are "correct". The various tests on the geometry side:

  - [mesh-half space intersection](https://github.com/RobotLocomotion/drake/blob/master/geometry/proximity/test/mesh_half_space_intersection_test.cc#L986-L1560)
  - [mesh-mesh intersection](https://github.com/RobotLocomotion/drake/blob/master/geometry/proximity/test/mesh_intersection_test.cc#L999-L1609)
  - [mesh-plane intersection](https://github.com/RobotLocomotion/drake/blob/master/geometry/proximity/test/mesh_plane_intersection_test.cc#L1322-L1908)

All have test code that examines various ways in which inputs to these methods will have the expected impact on the derivatives on the output. As long as we have tests that show that the force evaluation (based on a given contact surface) has done the same, we should be content. If you feel the tests on the traction calculator achieve that, it's up to you.

jwnimmer-tri, 2024-11-01:
We should either write the test as part of the feature development because it is surely important, or else just wait to write tests until we encounter a real bug.  Closing this as "not planned" -- having "missing unit test" bugs open for years is of no value.

closed 2024-11-01 by jwnimmer-tri