RobotLocomotion/drake #15234: DrakeLcm default queue behaviour is confusingly lossy

ggould-tri, 2021-06-23:
Context: https://github.com/lcm-proj/lcm/issues/345 LCM queue overflow works the opposite of how you would expect, discarding the most recent rather than least recent message.

This is not usually especially problematic, because the LCM message queue size of 30 is seldom reached in practice.

However LCM *also* has per-subscriber queues, and we default those to size 1 (`result->native_subscription_->setQueueCapacity(1)`) in `DrakeLcm` and those queues also have this first-wins behaviour.  This means that a call to `HandleSubscriptions` will by default only ever process the least recent message for each subscription and will silently discard the rest.

This behaviour might arguably be as-documented (though you'd be hard pressed to realize the implication) but it is profoundly perverse.

Absent a fix to the upstream queue bug, it is up to us to prevent queue overflow.  We should increase our default per-subscription queue size or better leave it at its default (equal to the global queue size).  Failing that, we should document the perverse behaviour and reference the upstream bug in `DrakeLcmInterface::Subscribe` and possibly elsewhere.

ggould-tri, 2021-06-23:
Another implication of this is that `drake::lcm::Subscriber::message()` never meets its documented behaviour ("returns the most recently received message") and, because `Subscriber` does not expose its underlying subscription, cannot be worked around to do so.

ggould-tri, 2021-06-24:
Assigning this to the dynamics team because https://drake.mit.edu/issues.html#team doesn't list network/LCM and @sherm1 probably knows how this should actually be assigned.

sherm1, 2021-06-24:
Reassigning to @jwnimmer-tri for disposition.

jwnimmer-tri, 2021-06-28:
I'll work on a more detailed reply later on, but let me ask for a little more information in the meantime:

A fundamental invariant of using LCM is that you call `handleTimeout` (which wrapped as `DrakeLcm::HandleSubscriptions`, which is wrapped as `CalcNextUpdateTime`) at least as frequently as your highest-frequency incoming message.  Can you provide any background on how it came to pass this this was not true in your situation?

ggould-tri, 2021-06-28:
The incoming messages are status messages about the progress of process setup, similar to logging; they occur at the rate at which the other end (controller) makes progress through it setup procedure, which is not known to the receiver.

If the answer is that a process should not emit messages at more than a known-in-advance rate, we should probably put a rate limiter into DrakeLcm to fail fast if you emit messages too quickly?