RobotLocomotion/drake #15244: DeclareVectorInputPort error

yangL11, 2021-06-28:
hey, every one, i want to test the examples of pendulum and design a new plant.h in one team project, but as shown below, "this->DeclareVectorInputPort" runs wrong, but with the same code in my own test project, it runs all well, the difference of which is that cmakelist are different, team project's was written by someone else, i am not sure if cmakelist of exporting drake library is the cause, or some other errors which i don't know, could you help me with some advice??

codes : 
PendulumPlant(): systems::LeafSystem<T>(systems::SystemTypeTag<PendulumPlant>{}){
          this->DeclareVectorOutputPort("state", &PendulumPlant::CopyStateOut,{this->all_state_ticket()});
          this->DeclareContinuousState(PendulumState<T>(), 1 , 1 ,0 );
          this->DeclareVectorInputPort("tau",systems::BasicVector<T>(1));
    };

yangL11, 2021-06-28:
sorry, to correct one mistake: this->DeclareVectorInputPort("tau",systems::BasicVector<T>(1)); this is right one

SeanCurtis-TRI, 2021-06-28:
Several thoughts for you to consider:

1. This question may really be better suited for stackoverflow. As you pointed out, the difference in the two efforts is code that isn't part of Drake. So, seeking help on configuring an external cmake project would be really well suited for stackoverflow so multiple users can benefit.
2. When presenting a problem, you should also include error messages. It's not clear what "runs wrong" means in this context. Being explicit about the error message will enable others to help resolve the problem.

I'm going to close this issue (and hope you'll repost in stackoverflow with additional information). If you strongly feel that this reflects a Drake defect, feel free to re-open the issue and provide further details.

closed 2021-06-28 by SeanCurtis-TRI