RobotLocomotion/drake #15298: Expression vs Eigen 3.4 memset

jwnimmer-tri, 2021-07-01:
In the Eigen 3.4 preview branch, it sometimes treats Expression as a primitive type that is safe to use for bitwise operations (`memset`, `memcpy`, bit_xor, etc.).  Since Expression is a class (and embeds shared_ptr), this is obviously not going to end well.

For example, with this snippet we have `Packet` set to `Expression`:
```c++
template<typename Packet>
Packet pzero(const Packet&) {
  Packet b;
  memset((void*)&b, 0, sizeof(b));
  return b;
}
```
See https://gitlab.com/libeigen/eigen/-/blob/1f6b1c1a1fba66dd2e09b674bbeeaf8bbd4f11f6/Eigen/src/Core/GenericPacketMath.h#L218-224 for the full code.

Even worse, Eigen also now `reinterpret_casts` the type to do bitwise opetions:

https://gitlab.com/libeigen/eigen/-/blob/1f6b1c1a1fba66dd2e09b674bbeeaf8bbd4f11f6/Eigen/src/Core/GenericPacketMath.h#L257-267

This can be seen by testing `env CC=clang-9 CXX=clang++-9 bazel test --config asan --copt=-O0 //math:rotation_matrix_test --test_arg=--gtest_filter=RotationMatrix.SymbolicProjectionTest` with Eigen commit `4780d8dfb2b0e9bcff063c80f4ffce71d9d7a725` -- see #15142 for details.

jwnimmer-tri, 2021-07-01:
Here is the ASan output, to give a sense of the call stack:
```
    #6 0x55b860623654 in drake::symbolic::Expression::~Expression() /proc/self/cwd/bazel-out/k8-opt/bin/common/_virtual_includes/_symbolic_private_headers_impl/drake/common/symbolic_expression.h:197:25
    #7 0x55b8607de8ef in drake::symbolic::Expression Eigen::internal::pandnot<drake::symbolic::Expression>(drake::symbolic::Expression const&, drake::symbolic::Expression const&) /proc/self/cwd/external/eigen/Eigen/src/Core/GenericPacketMath.h:310:45
    #8 0x55b8607de512 in drake::symbolic::Expression Eigen::internal::pselect<drake::symbolic::Expression>(drake::symbolic::Expression const&, drake::symbolic::Expression const&, drake::symbolic::Expression const&) /proc/self/cwd/external/eigen/Eigen/src/Core/GenericPacketMath.h:315:27
    #9 0x55b8607ddf02 in drake::symbolic::Expression Eigen::internal::pminmax_impl<1>::run<drake::symbolic::Expression, drake::symbolic::Expression (*)(drake::symbolic::Expression const&, drake::symbolic::Expression const&)>(drake::symbolic::Expression const&, drake::symbolic::Expression const&, drake::symbolic::Expression (*)(drake::symbolic::Expression const&, drake::symbolic::Expression const&)) /proc/self/cwd/external/eigen/Eigen/src/Core/GenericPacketMath.h:352:18
    #10 0x55b8607ddd01 in drake::symbolic::Expression Eigen::internal::pmax<1, drake::symbolic::Expression>(drake::symbolic::Expression const&, drake::symbolic::Expression const&) /proc/self/cwd/external/eigen/Eigen/src/Core/GenericPacketMath.h:402:10
    #11 0x55b8607dd6fe in Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1>::operator()(drake::symbolic::Expression const&, drake::symbolic::Expression const&) const /proc/self/cwd/external/eigen/Eigen/src/Core/functors/BinaryFunctors.h:176:12
    #12 0x55b8607df61e in Eigen::internal::redux_novec_unroller<Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1>, Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> >, 0, 2>::run(Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> > const&, Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1> const&) /proc/self/cwd/external/eigen/Eigen/src/Core/Redux.h:110:12
    #13 0x55b8607dd860 in Eigen::internal::redux_novec_unroller<Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1>, Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> >, 0, 4>::run(Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> > const&, Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1> const&) /proc/self/cwd/external/eigen/Eigen/src/Core/Redux.h:110:17
    #14 0x55b8607dd540 in Eigen::internal::redux_novec_unroller<Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1>, Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> >, 0, 9>::run(Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> > const&, Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1> const&) /proc/self/cwd/external/eigen/Eigen/src/Core/Redux.h:110:17
    #15 0x55b8607dd24e in drake::symbolic::Expression Eigen::internal::redux_impl<Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1>, Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> >, 0, 2>::run<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> >(Eigen::internal::redux_evaluator<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> > const&, Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1> const&, Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> const&) /proc/self/cwd/external/eigen/Eigen/src/Core/Redux.h:222:12
    #16 0x55b8607dd0dd in drake::symbolic::Expression Eigen::DenseBase<Eigen::CwiseUnaryOp<Eigen::internal::scalar_abs_op<drake::symbolic::Expression>, Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const> >::redux<Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1> >(Eigen::internal::scalar_max_op<drake::symbolic::Expression, drake::symbolic::Expression, 1> const&) const /proc/self/cwd/external/eigen/Eigen/src/Core/Redux.h:418:10
[... cut ...] ression, 3, 3, 0, 3, 3> > const&, drake::symbolic::Expression*) /proc/self/cwd/bazel-out/k8-opt/bin/math/_virtual_includes/geometric_transform/drake/math/rotation_matrix.h:846:24
    #22 0x55b860797bc3 in drake::math::RotationMatrix<drake::symbolic::Expression>::ProjectToRotationMatrix(Eigen::Matrix<drake::symbolic::Expression, 3, 3, 0, 3, 3> const&, drake::symbolic::Expression*) /proc/self/cwd/bazel-out/k8-opt/bin/math/_virtual_includes/geometric_transform/drake/math/rotation_matrix.h:636:9
```

jwnimmer-tri, 2021-07-01:
One work-around would be to specialize the internal packet functions to opt-out of the memset and bit casts.  This patch gets the test passing:

```patch
--- a/common/symbolic_expression.h
+++ b/common/symbolic_expression.h
@@ -1218,6 +1218,28 @@ struct ScalarBinaryOpTraits<double, drake::symbolic::Expression, BinaryOp> {
   typedef drake::symbolic::Expression ReturnType;
 };
 
+namespace internal {
+template<> EIGEN_DEVICE_FUNC inline ::drake::symbolic::Expression
+ptrue(const ::drake::symbolic::Expression&) { return 1; }
+template<> EIGEN_DEVICE_FUNC inline ::drake::symbolic::Expression
+pzero(const ::drake::symbolic::Expression&) { return 0; }
+template<> EIGEN_DEVICE_FUNC inline ::drake::symbolic::Expression
+pand(const ::drake::symbolic::Expression& a,
+     const ::drake::symbolic::Expression& b) {
+  return static_cast<int>(a.Evaluate()) & static_cast<int>(b.Evaluate());
+}
+template<> EIGEN_DEVICE_FUNC inline ::drake::symbolic::Expression
+por(const ::drake::symbolic::Expression& a,
+    const ::drake::symbolic::Expression& b) {
+  return static_cast<int>(a.Evaluate()) | static_cast<int>(b.Evaluate());
+}
+template<> EIGEN_DEVICE_FUNC inline ::drake::symbolic::Expression
+pxor(const ::drake::symbolic::Expression& a,
+     const ::drake::symbolic::Expression& b) {
+  return static_cast<int>(a.Evaluate()) ^ static_cast<int>(b.Evaluate());
+}
+}  // namespace internal
+
 }  // namespace Eigen
 #endif  // !defined(DRAKE_DOXYGEN_CXX)
 
```

However, this doesn't seem like something we should need to opt-out of.

Our `Eigen::NumTraits<Expression>` clearly says `RequireInitialization = 1`.  It seems like a bug in Eigen for it to be ignoring that.

jwnimmer-tri, 2021-07-01:
+@soonho-tri could I ask you to investigate further / confirm the above, and then help decide on next steps here?  It seems like we should make a min repro test case and submit upstream to Eigen?

sherm1, 2021-07-01:
Doesn't seem like this would work for Eigen's AutoDiffScalar either since the derivative part can be a VectorX.

jwnimmer-tri, 2021-07-01:
There might be a trick already in play there -- a memset-to-zero is a valid AutoDiffXd (zero double value, zero num_derivs, null heap pointer), and a memset-the-first-byte-to-0xff is also valid (it only frobs the double value) and since we never heap allocate we never need to free, and it magically works.

jwnimmer-tri, 2021-07-01:
To connect the dots, see also #12453 for related information.

soonho-tri, 2021-07-01:
@jwnimmer-tri , I've started creating a small test case. But so far I couldn't reproduce the problem yet. I'll keep digging it.

jwnimmer-tri, 2021-07-05:
Here is a single-file minimal repro:
https://github.com/jwnimmer-tri/drake/commits/eigen-34-debug-wip

soonho-tri, 2021-07-08:
I think there is an existing issue on this problem: https://gitlab.com/libeigen/eigen/-/issues/2245

cantonios, 2021-07-08:
There are two related problems in Eigen with custom types:
- [#2245](https://gitlab.com/libeigen/eigen/-/issues/2245), which deals with zeroing-out large arrays and should now be fixed on master with [!555](https://gitlab.com/libeigen/eigen/-/merge_requests/555)
- [#2201](https://gitlab.com/libeigen/eigen/-/issues/2201), which deals with bitwise packet operations for custom types.

I re-opened the latter now, but the original thinking was that you would need to specialize all boolean/bitwise operations anyways (including `ptrue`, `pnot`, `pand`, etc...) for your custom type, unless it was POD and would just happen to work with the generic implementation.  The original `memset` code has been around for a few years.  If we modified to assert for non-POD types (or if `RequireInitialization == 0`), this may break existing code where it was already working with the generic impl.

We could introduce a breaking change for Eigen 3.4, so I am now leaning towards asserting to produce a compile-time failure.  You would still need to specialize all the bitwise functions for your type though.

~Antonio (one of the Eigen maintainers)

jwnimmer-tri, 2021-07-08:
I appreciate the response!  We'll work on adding the specializations.

The reason that I thought we were _not_ supposed to specialize the `ptrue` etc. was because they are in the `Eigen::internal` namespace.  I guess I misinterpreted the intention of what that meant.

cantonios, 2021-07-08:
Yeah... that has come up.  We don't really want people to be doing that, but we also don't provide an alternative when adding custom types.

If it can wait, we can put in bitwise functionals similar to `std::bit_and` that could be used.  I'm not sure that could make it into 3.4 though.

jwnimmer-tri, 2021-07-08:
I'm happy to work with whatever we have available in 3.4.  We can `#ifdef` it into something else in 3.5 if it changes later on.

jwnimmer-tri, 2021-07-21:
Our `fixed_fem` unit tests appear to be showing showing the packet operations (like `pandnot`, it seems) writing `0xff` bytes atop our `AutoDiffScalar` derivatives pointers as well, which `free()` is unhappy about.

cantonios, 2021-07-21:
This should be fixed by [this MR](https://gitlab.com/libeigen/eigen/-/merge_requests/562).  Unfortunately most of the other Eigen maintainers are currently away, delaying a merge.  I think we can include this in 3.4.

jwnimmer-tri, 2021-07-22:
Thank you for the pointer!  I cherry-picked it into our integration branch.  As best I can tell at the moment, it has solved _all_ of the packet-related compatibility issues that we were seeing.

jwnimmer-tri, 2021-08-03:
The upstream MR has landed, so I'm going to call this closed.

There are other new problems with our integration branch, and I haven't yet root-caused them, so I can't 100% prove that this is fully resolved, but since the cherry-picked fix proved our a couple weeks ago, I think we're on safe gound.

closed 2021-08-03 by jwnimmer-tri

jwnimmer-tri, 2021-08-10:
@cantonios thanks for your help here!  I can finally confirm now that our project's entire test suite is 100% passing in all 80+ of our different build configurations.  Eigen 3.4 seems solid now, in our testing.