RobotLocomotion/drake #15316: solvers: Ipopt segfaults when a problem has a quadratic objects + a Lorentz cone constraint

RussTedrake, 2021-07-05:
This runs fine on Mosek, but causes IPOPT to segfault:
```
from pydrake.all import MathematicalProgram, Solve

prog = MathematicalProgram()
x = prog.NewContinuousVariables(2)

prog.AddLorentzConeConstraint(x)
prog.AddQuadraticCost(x.dot(x))

result = Solve(prog)
print(result.is_success())
print(result.get_solver_id().name())
```

hongkai-dai, 2021-07-05:
The problem in IPOPT solver handling Lorentz cone constraint. If I remove the quadratic cost I still see the seg fault. Looking into the bug.

hongkai-dai, 2021-07-05:
The reason why it segfault is due to how we impose Lorentz cone constraint `z₀ ≥ sqrt(z₁² + ... + zₙ₋₁²)` as generic nonlinear constraint. We provide 3 formulations in https://github.com/RobotLocomotion/drake/blob/dd7b1b81d45c502a0d09d03999fde5e43588c5a5/solvers/constraint.h#L306-L310
, and the default is the "Convex" formulation `z₀ - sqrt(z₁² + ... + zₙ₋₁²) ≥ 0`

There are issues with these formulations

1. For this "convex" formulation, when z = 0 (which is typically the initial guess), the constraint is non-differentiable, due to sqrt(z₁² + ... + zₙ₋₁²). We return NAN as the gradient value, and that NAN cause IPOPT to segfault. If I change the initial guess then the program runs fine.
2. For the "nonconvex" formulation, the constraint z₀²-(z₁²+...+zₙ₋₁²) ≥ 0 has 0 gradient when z=0, so it will get stuck at the initial guess.
3. I switched to kConvexSmooth, which modifies the gradient slightly. Now Ipopt successfully finds the solution with initial guess being 0.

So I think one solution is to change the default eval_type of LorentzConeConstraint to kConvexSmooth. And add that optional argument to `AddLorentzConeConstraint` function to specify `eval_type`.

PR #15318 (2021-07-05): Switch default eval_type of LorentzConeConstraint to ConvexSmooth.
Also add test for ipopt and snopt with Lorentz cone constraint

Resolves #15316

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15318)
closed 2021-07-05 by hongkai-dai