RobotLocomotion/drake #15320: More granular control on ChooseBestSolver

hongkai-dai, 2021-07-06:
Currently `ChooseBestSolver` has a linear decision chain, it mains a list of solvers, checks if the first solver in the list is available and satisfies the program requirement. If yes it calls the first solver, otherwise it moves to the second solver, and so on.

The issue is that between two solvers solverA and solverB, the order of choosing these solvers should not be fixed, they should depend on mathematical program property. For example, IPOPT usually gives a more accurate solution than SCS for linear program, but for program with second order cone constraint, SCS should be preferable to IPOPT, as seen from #15316 .

I propose that at the beginning of `ChooseBestSolver`, we first determine the type of the optimization problem, and for some problem types, we use its own list of solvers. For example, if we determine the optimization is a linear program, then we use the list `Mosek, Gurobi, CLP, IPOPT, SCS`; if the optimization is second order cone program, then we use the list `Mosek, Gurobi, SCS, CSDP`. If the program doesn't fall into these special types (LP or SOCP) where we have designated list of solvers, then it falls back to the current implementation.

cc @jwnimmer-tri @RussTedrake

RussTedrake, 2021-07-06:
I agree with the need for more granular control.

In particular, I had the experience of writing a mathematical program which I knew was convex.  But after calling the `ChooseBestSolver` via `solvers::Solve`, the default solver got stuck in a local minima!  @hongkai-dai outlined the reasoning above, but let me just emphasize the detail:  

For conic constraints in particular, some solvers handle them explicitly (revealing their convexity), but when we hand them to a more general purpose solver like IPOPT, they see a slightly different formulation which can have zero-gradients / local minima.  Since IPOPT turned out to be faster/more accurate than the open-source convex solvers for some convex optimization problems, it moved itself up in the list of `ChooseBestSolver`.  But then I ended up with local minima in my convex optimization.

I feel that for conic constraints (and presumably SDP constraints, perhaps even others), where we fall back to a generic constraint implementation when the solver doesn't handle things explicitly, we have introduced this potential problem.  I think the proposed solution above definitely helps.  I still don't like that the user can be so surprised.

At the least, I would proposed that the user be able to get access to the classification that we discovered (e.g. something like `result.get_problem_type() == kSecondOrderConeProgram`) so that they can confirm that the problem came out as they expected.  And perhaps even a sugar method that says `IsConvexProgram(result.get_problem_type())`, since that would be what I'd need in my current use case: I have subroutines which are allowed to add a variety of constraints, and want to confirm that they resulted in calling a solver than will not get stuck in local minima.

One last subtlety to point out.  There is a difference between:
1) whether all of the costs and constraints are independently convex,
2) whether the resulting program ends up as one fo the standard convex program types,
3) whether we currently have a convex solver (especially open-source) that supports it.  

For an example distinguishing (1) - (2): if i add a quadratic objective and then a conic constraint, then this would typically not be considered an SOCP, though it can certainly be reformulated as one using a slack variable.

jwnimmer-tri, 2021-07-06:
> The issue is that between two solvers solverA and solverB, the order of choosing these solvers should not be fixed, they should depend on mathematical program property.

In case it helps, I'll second the idea that `ChooseBestSolver` is permitted to become arbitrarily complex, by adding more rules or heuristics or whatever is needed to ensure that the "best" solver _is_ actually the best choice.

We will need to be sure that its unit test appropriately covers the new complexity, but that shouldn't be too difficult.

> ... user be able to get access to the classification that we discovered ...

A different way to help the user be confident would be to tell the `MathematicalProgram` that the solution must be convex, perhaps something like `program.set_required_convexity(true)`, though we should think for a bit about the best function name.

If the user called that function, then as soon as that program reaches a solver that is going to do a generic fallback, the solver would immediately throw an exception.  We would not need to remember to check the result at the end, because we expressed our intent right from the beginning.

hongkai-dai, 2021-07-07:
> If the user called that function, then as soon as that program reaches a solver that is going to do a generic fallback, the solver would immediately throw an exception

There is some subtlety here. If we have a quadratic optimization problem (which is convex) and we set `program.set_required_convexity(true)`, we still would like to fall back to generic nonlinear solver like SNOPT or IPOPT, instead of using a convex solver (like SCS). For this type of convex optimization problem (QP), the generic nonlinear solvers SNOPT/IPOPT generally performs better than an inaccurate convex solver (like SCS). But when we have certain type of convex programs (like second-order cone problem), then we would prefer SCS over SNOPT/IPOPT, as the generic nonlinear solver can get stuck (and even seg fault).

PR #15541 (2021-08-30): Choose best solver based on program type.
Resolves #15320

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15541)
closed 2021-08-30 by hongkai-dai