RobotLocomotion/drake #15341: Clp gives the wrong answer (claiming success) for a trivial LP

RussTedrake, 2021-07-08:
This one was tough to find! 

```
from pydrake.all import MathematicalProgram, Solve, IpoptSolver

np.set_printoptions(formatter={'float': lambda x: "{0:0.2f}".format(x)})
ipopt = IpoptSolver()

A = np.array([[1.0, 0.0, 0.0],
 [0.0, 1.0, 0.0],
 [0.0, 0.0, 1.0],
 [-1.0, -0.0, -0.0],
 [-0.0, -1.0, -0.0],
 [-0.0, -0.0, -1.0],
[ 9.99400723e-01, -1.75186298e-21,  3.46149414e-02],
 [ 9.82328785e-01,  1.82238474e-20, -1.87163452e-01],
 [ 9.68599420e-01, -2.48626556e-01, -1.27003703e-12],
 [ 9.68599420e-01,  2.48626556e-01 ,-1.27003703e-12]])
b = np.array([1.0, 0.5, 1.0, -0.0, 0.5, -0.0, 0.46783912,  0.39023174,  0.50647967,  0.50647967])

prog = MathematicalProgram()
x = prog.NewContinuousVariables(3)
r = prog.NewContinuousVariables(1)[0]

A_row_norm = np.linalg.norm(A, axis=1)
lhs = A.dot(x) + A_row_norm * r
prog.AddLinearConstraint(le(lhs, b))
prog.AddLinearConstraint(r >= 0)
prog.AddLinearCost(-r)

result1 = ipopt.Solve(prog)
assert result1.is_success()
print(result1.get_solver_id().name())
print(result1.GetSolution(x))
print(result1.is_success())
print(result1.get_solution_result())

result2 = Solve(prog)
assert result1.is_success()
print(result2.get_solver_id().name())
print(result2.GetSolution(x))
print(result2.is_success())
print(result2.get_solution_result())
```

gives 
```
IPOPT
[0.23 -0.00 0.33]
True
SolutionResult.kSolutionFound
CLP
[0.00 0.00 0.00]
True
SolutionResult.kSolutionFound
```

The answer from Ipopt is correct,.  

Note that the small numerical values in the coefficients seem to have something to do with it.  If I set the two coefficients that are ~ `1e-21` to zero, then CLP once again gives the correct answer.

RussTedrake, 2021-07-08:
And btw -- Clp is the default open-source solver chosen for this problem.

RussTedrake, 2021-07-08:
I've updated the example above to print to console (n.b. it took me 20+ minutes to figure *that* out; i'll open a separate issue).  

The output is now:
```
IPOPT
[0.23 -0.00 0.33]
True
SolutionResult.kSolutionFound
Clp0024I Matrix will be packed to eliminate 1 small elements
Clp1001I Initial range of elements is 1.8223847e-20 to 1
Clp1002I Range of elements is 1.3499573e-10 to 7.4076417e+09
Clp1002I Range of elements is 2.6556569e-08 to 37655466
Clp1003I Final range of elements is 7.0525138e-14 to 100
Clp0022I Absolute values of scaled rhs range from 10.83627 to 3.1033744e+09, minimum gap 1e+100
Clp0020I Absolute values of scaled objective range from 1.2574433e-08 to 1.2574433e-08
Clp0000I Optimal - objective value 0
CLP
[0.00 0.00 0.00]
True
SolutionResult.kSolutionFound
0
```

if I remove the 1e-20 term (only), then the output becomes:
```
IPOPT
[0.23 0.00 0.33]
True
SolutionResult.kSolutionFound
Clp0024I Matrix will be packed to eliminate 1 small elements
Clp1001I Initial range of elements is 1.270037e-12 to 1
Clp1002I Range of elements is 1.1269592e-06 to 887343.57
Clp1002I Range of elements is 0.0010615833 to 941.98916
Clp1003I Final range of elements is 0.00011269592 to 100
Clp0022I Absolute values of scaled rhs range from 0.90285848 to 449421.48, minimum gap 1e+100
Clp0020I Absolute values of scaled objective range from 0.00011269592 to 0.00011269592
Clp0006I 0  Obj 0 Dual inf 0.00011259592 (1)
Clp0006I 4  Obj -0.22823558
Clp0017I Looking optimal with tolerance of 1e-07
Clp0000I Optimal - objective value -0.22823558
CLP
[0.23 0.00 0.33]
True
SolutionResult.kSolutionFound
0
```
which is correct.

RussTedrake, 2021-07-08:
So I definitely don't think this is MathematicalProgram's fault, but i do think that if we put Clp as our default LP solver than we somehow need to address this.  I don't know if we should upstream an issue to Clp?  If we should ask Clp about the scaled objective range at the solution and not trust it when it's 1e-8?  If we should choose a different solver for the default LP solver?

hongkai-dai, 2021-07-08:
Sorry for all the trouble you have to go through to locate the pitfall. I will create a program using only CLP code, without Drake's code, and file the issue on CLP upstream.

The easiest solution is probably to change the default solver to IPOPT. BTW I am working on #15320 , the first step is the PR #15340 to tell the type of the optimization program.

jwnimmer-tri, 2021-11-11:
@hongkai-dai do you have a link to the CLP upstream issue that you could mention here?

RussTedrake, 2021-12-21:
I've just hit this again.  CLP reports success, but gives a terrible answer:
```
min z
|(v α - x)ᵢ| ≤ z as -z ≤ vᵢ α - xᵢ ≤ z as vᵢ α - z ≤ xᵢ && xᵢ ≤ vᵢ α + z
0 ≤ αᵢ ≤ 1
∑ᵢ αᵢ = 1

Decision variables:z(0) a(0) a(1) a(2) a(3)

LinearCost z(0)
LinearConstraint
-inf <= ( - z(0) + 5.5511151231257827e-17 * a(0) + a(1) + 5.5511151231257827e-17 * a(2) + a(3)) <= 0.99
-inf <= ( - z(0) + a(1) + 2 * a(2) + 2 * a(3)) <= 1.99
LinearConstraint
0.99 <= (z(0) + 5.5511151231257827e-17 * a(0) + a(1) + 5.5511151231257827e-17 * a(2) + a(3)) <= inf
1.99 <= (z(0) + a(1) + 2 * a(2) + 2 * a(3)) <= inf
LinearEqualityConstraint
(a(0) + a(1) + a(2) + a(3)) == 1
BoundingBoxConstraint
0 <= a(0) <= 1
0 <= a(1) <= 1
0 <= a(2) <= 1
0 <= a(3) <= 1

alpha_sol =   0 0.5 0.5   0
z_sol = 0.49
```

hongkai-dai, 2021-12-21:
@jwnimmer-tri I am trying to reproduce this using native CLP code, without the Drake MathematicalProgram interface. On ubuntu, does Drake use the system CLP? I see the pkg_config_paths is set to /usr/local/opt/clp/lib/pkgconfig (https://github.com/RobotLocomotion/drake/blob/6f602c3d57c787396e2560edefe17234ea33fc6b/tools/workspace/clp/repository.bzl#L15), but on my ubuntu machine, I don't have a folder /usr/local/opt. Could you tell me how to compile and link against the native CLP? (I can try to install CLP using coin-or softwares, but last time when I install it on my old computer I messed up my CLP installation, so I want to be extra careful this time when installing it).

jwnimmer-tri, 2021-12-21:
> On Ubuntu, does Drake use the system CLP

Yes.

https://github.com/RobotLocomotion/drake/blob/6f602c3d57c787396e2560edefe17234ea33fc6b/setup/ubuntu/source_distribution/packages-focal.txt#L3

> I see the `pkg_config_paths` is set to `/usr/local/opt/clp/lib/pkgconfig` ...

The parameter name is slightly misleading.  These are _extra_ paths.  On Ubuntu, `pkg-config` will _always_ search for the OS-installed packages.

hongkai-dai, 2021-12-22:
@jwnimmer-tri sorry to bother you again, I tried to create the code using upstream CLP code
```
#include "ClpSimplex.hpp" 
#include <limits>

int main() {
  ClpSimplex model;
  model.setLogLevel(1);
  const int numcols = 5;
  const int numrows = 5;
  const double kInf = std::numeric_limits<double>::infinity();
  double collb[5] = {0, 0, 0, 0, 0};
  double colub[5] = {1, 1, 1, 1, kInf};
  int start[6] = {0, 3, 8, 13, 18, 22};
  int index[22] = {0, 2, 4, 0, 1, 2, 3, 4, 0, 1, 2, 3, 4, 0, 1, 2, 3, 4, 0, 1, 2, 3};
  const double eps = 5.5511151231257827e-17;
  double val[22] = {eps, eps, 1, 1, 1, 1, 1, 1, eps, 2, eps, 2, 1, 1, 2, 1, 2, 1, -1, -1, 1, 1};
  double rowlb[5] = {-kInf, -kInf, 0.99, 1.99, 1};
  double rowub[5] = {0.99, 1.99, 0.99, 1.99, 1};
  double obj[5] = {0, 0, 0, 0, 1};
  model.loadProblem(numcols, numrows, start, index, val, collb, colub, obj, rowlb, rowub);
  model.primal();
  return 0;
}
```
and I name this file as `clp_test.cc`

Then I tried to compile this file as
```
➜  ~ g++ -I/usr/include/coin/ -L/usr/lib/x86_64-linux-gnu/libClp.a clp_test.cc -o clp_test
```
but I receive "undefined reference" error in the linker. I also tried to use `-lClp` and `-L/usr/lib/x86_64-linux-gnu/libClp.so`, but still got the same error.

I tried `dpkg -L coinor-libclp-dev` and got this
```
/.
/usr
/usr/include
/usr/include/coin
/usr/include/coin/CbcOrClpParam.cpp
/usr/include/coin/CbcOrClpParam.hpp
/usr/include/coin/ClpCholeskyBase.hpp
/usr/include/coin/ClpCholeskyDense.hpp
/usr/include/coin/ClpCholeskyPardiso.hpp
/usr/include/coin/ClpConfig.h
/usr/include/coin/ClpConstraint.hpp
/usr/include/coin/ClpConstraintLinear.hpp
/usr/include/coin/ClpConstraintQuadratic.hpp
/usr/include/coin/ClpDualRowDantzig.hpp
/usr/include/coin/ClpDualRowPivot.hpp
/usr/include/coin/ClpDualRowSteepest.hpp
/usr/include/coin/ClpDummyMatrix.hpp
/usr/include/coin/ClpDynamicExampleMatrix.hpp
/usr/include/coin/ClpDynamicMatrix.hpp
/usr/include/coin/ClpEventHandler.hpp
/usr/include/coin/ClpFactorization.hpp
/usr/include/coin/ClpGubDynamicMatrix.hpp
/usr/include/coin/ClpGubMatrix.hpp
/usr/include/coin/ClpInterior.hpp
/usr/include/coin/ClpLinearObjective.hpp
/usr/include/coin/ClpMatrixBase.hpp
/usr/include/coin/ClpMessage.hpp
/usr/include/coin/ClpModel.hpp
/usr/include/coin/ClpNetworkMatrix.hpp
/usr/include/coin/ClpNode.hpp
/usr/include/coin/ClpNonLinearCost.hpp
/usr/include/coin/ClpObjective.hpp
/usr/include/coin/ClpPEDualRowDantzig.hpp
/usr/include/coin/ClpPEDualRowSteepest.hpp
/usr/include/coin/ClpPEPrimalColumnDantzig.hpp
/usr/include/coin/ClpPEPrimalColumnSteepest.hpp
/usr/include/coin/ClpPESimplex.hpp
/usr/include/coin/ClpPackedMatrix.hpp
/usr/include/coin/ClpParameters.hpp
/usr/include/coin/ClpPdcoBase.hpp
/usr/include/coin/ClpPlusMinusOneMatrix.hpp
/usr/include/coin/ClpPresolve.hpp
/usr/include/coin/ClpPrimalColumnDantzig.hpp
/usr/include/coin/ClpPrimalColumnPivot.hpp
/usr/include/coin/ClpPrimalColumnSteepest.hpp
/usr/include/coin/ClpQuadraticObjective.hpp
/usr/include/coin/ClpSimplex.hpp
/usr/include/coin/ClpSimplexDual.hpp
/usr/include/coin/ClpSimplexNonlinear.hpp
/usr/include/coin/ClpSimplexOther.hpp
/usr/include/coin/ClpSimplexPrimal.hpp
/usr/include/coin/ClpSolve.hpp
/usr/include/coin/Clp_C_Interface.h
/usr/include/coin/Idiot.hpp
/usr/include/coin/OsiClpSolverInterface.hpp
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libClp.a
/usr/lib/x86_64-linux-gnu/libClpSolver.a
/usr/lib/x86_64-linux-gnu/libOsiClp.a
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/clp.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/osi-clp.pc
/usr/share
/usr/share/doc
/usr/share/doc/coinor-libclp-dev
/usr/share/doc/coinor-libclp-dev/copyright
/usr/lib/x86_64-linux-gnu/libClp.so
/usr/lib/x86_64-linux-gnu/libClpSolver.so
/usr/lib/x86_64-linux-gnu/libOsiClp.so
/usr/share/doc/coinor-libclp-dev/changelog.Debian.gz
```

How should I fix this linking error?

jwnimmer-tri, 2021-12-22:
When linking, you must list the inputs in top-down, topological order.  So you should say `clp_test.cc -lClp` not `-lClp clp_test.cc`.  You also need `CoinUtils`.

```
jwnimmer@call-cps:~/tmp/15341$ g++ -I/usr/include/coin/ -L/usr/lib/x86_64-linux-gnu/ clp_test.cc -lClp -lCoinUtils -o clp_test
jwnimmer@call-cps:~/tmp/15341$ ./clp_test
Clp0006I 0  Obj 0 Primal inf 1.3287555e+08 (3) Dual inf 5.7071068e+12 (5)
Clp0006I 5  Obj 5.5511151e-19
Clp0000I Optimal - objective value 5.5511151e-19
```

hongkai-dai, 2021-12-22:
Thanks @jwnimmer-tri for the help!

Now I believe the problem is in Drake's code, not in Clp upstream. I can solve this program using Clp, and the solution is
`alpha = [0.005, 0, 0.005, 0.99], z = 0`. 
I think this is the right answer. I will try to look into why Drake's ClpSolver produces a different result.

hongkai-dai, 2021-12-22:
OK, after more investigation, I think it is actually CLP's fault. I filed an issue to CLP in https://github.com/coin-or/Clp/issues/217

PR #16281 (2021-12-23): Call scaling() before solving in ClpSolver.
This handles the poor scaling in the problem data.

Resolves #15341

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16281)
closed 2021-12-23 by hongkai-dai