RobotLocomotion/drake #15342: Usability of MathematicalProgram solver output features

RussTedrake, 2021-07-08:
I was having problems with ClpSolver, and wanted to see it's output.  It took me 20+ minutes to figure out how, and involved reading a surprising amount of source code.
- The options are very hard to find.  In the end, the magic incantation was 
```
options = SolverOptions()
options.SetOption(CommonSolverOption.kPrintToConsole, 1)
prog.SetSolverOptions(options)
```
this is as opposed to the `prog.SetSolverOption(foo)` I was suspecting.  This much was discoverable on doxygen, but with a bunch of clicks.  And it requirement me knowing to look at the SolverOptions on MathematicalProgram, and guessing that it might be inside the CommonSolverOptions item that appeared. My initial search started on the ClpSolver page, which is effectively blank.
- The solver output does not appear in jupyter.  I had to switch to a py file running in the console to see any output. (but of course I had to realize that this was the problem; as a user, I simply saw no effect of the option that I passed to the solver; just like my other attempts like below)
- Because the output did not appear, I initially tried printing to a file instead.  But Clp silently ignores that solver option.  Again, I had to read the Clp source code to discover this.  I would say that if a solver ignores one of the commonsolveroptions that has been set, we should get a log message, at least?
- The only example I see in the codebase of calling this from python is in `mathematicalprogram_test.py`.  Can we add it to the debugging mathematical programs tutorial?

jwnimmer-tri, 2021-07-08:
See also #12867.

---
On the Jupyter front:

Ideally, all of our bound solvers would route their logs into `drake::log()` and then that output would end up in the obvious places for the user, with the conventional severity filtering, but many of them are so crusty that the _only_ thing they can do is print to file descriptor 1, which is useless in Jupyter.

Because the Jupyter console is an oddball, I could see the case for Drake _noticing_ when a user has tried to `kPrintToConsole = 1` while running under Jupyter, and then giving them a big warning (or exception) instead.

hongkai-dai, 2021-07-14:
> So I think if you mis-spell an option, it is silently ignored.

It could be worse. If you mis-spell the option name, it silently ignored. If you mis-spell the option value, the program can abort without any warning. @TobiaMarcucci recently encountered the silent aborting due to passing a wrong value to Mosek solver option.

I will work on this for each solver, that if the option name is mis-spelled or value is incorrect, then we should throw a meaningful error message.

jwnimmer-tri, 2021-11-02:
Note that my recent #16019 improved this by epsilon.  The notebook tutorial for debugging uses the correct spellings for everything now.

I think it's also worth highlighting the original call to action, which still applies:

```python
options = SolverOptions()
options.SetOption(CommonSolverOption.kPrintToConsole, 1)
prog.SetSolverOptions(options)
```

This is difficult to discover (even with the tutorial), and is still awkward to use.

Ideally, we could set the common solver options _incrementally_ on the mathematical program, even if its just `prog.get_mutable_options.SetOption(kPrintToConsole, 1)`.

Possibly, for the one specific case of "print to console", we should have some simple boolean sugar, such as `prog.set_print_to_console(True)`.

jwnimmer-tri, 2022-04-26:
To help make the issues database more intelligible, I think we should keep this issue focused on the immediately usability (the spelling and discoverability for how to use our common solver options to turn on printing).  That problem is somewhat mitigated by having a tutorial about it (debug_mathematical_program), but we can still consider API improvements, e.g., adding a sugar function onto MathematicalProgram directly.

For the problem that most solvers' "print to console" goes to file descriptor 1 (which is useless for online notebooks), we should open a separate issue, if need be.

For the problem that not all of our solvers implement all the common options yet, we should track that compatibility inventory under #12867, which has already begun that job (and I'll go update it now with the latest).