RobotLocomotion/drake #15344: Make `drake::copyable_unique_ptr` const-correct.

RussTedrake, 2021-07-08:
Per a discussion on slack.  @jwnimmer-tri originally said "The `std::unique_ptr` has const-get that returns mutable-pointer, but `drake::copyable_unique_ptr` diverges from that pattern and stays const-correct."  But upon inspection:
```
copyable_unique_ptr<ConvexSet> c = std::make_unique<MutablePoint>(Vector2d{1., 2.});
const copyable_unique_ptr<ConvexSet>& c_ref = c;
static_assert(std::is_const_v<decltype(*c_ref)>);
```
fails the static assert.  To which Jeremy responded "Eww, I missed that `copyable_unique_ptr` inherits from `std::unique_ptr<T>` and fails to shadow `operator*` to preserve const-correctness.  We should look into fixing this..."

This would improve my use of `copyable_unique_ptr` in `ConvexSets`.

jwnimmer-tri, 2021-07-08:
@SeanCurtis-TRI and @sherm1 could you weigh in here on whether the _intention_ of `copyable_unique_ptr` was to have `operator*` propagate the const, like we have `.get()` vs `.get_mutable()` in its API?

And if that answer is "yes", could I convince one of you to take the issue and push forward on a fix?

RussTedrake, 2021-07-08:
Can we also add 
```
  explicit copyable_unique_ptr(T& ref) noexcept
      : std::unique_ptr<T>(ref.Clone()) {}
```
?? That makes constructing lists of `copyable_unique_ptr` much nicer.

RussTedrake, 2021-07-08:
I've opened #15345 if you care to see my specific use case.

SeanCurtis-TRI, 2021-07-08:
I'll take a look.

As to adding a `noexcept` on the constructor...I'd imagine that it transitively requires `noexcept` on `T::Clone()` as well, right?

RussTedrake, 2021-07-08:
I wasn't trying to make a statement about `noexcept`.  That part, I'm afraid, was just copy-pasta from the constructor above.  I figured you would make it perfect.

SeanCurtis-TRI, 2021-07-08:
Ah, you'd just like to copy it from a reference.  :)  Whether it's `noexcept` or not is not something you care about.  Got it.

SeanCurtis-TRI, 2021-07-08:
So, I'm convinced that distinguishing between `operator*()` and `operator*() const` is a good thing. Technically, the distinction has nothing to do with the concept of a copyable `unique_ptr`. The reason we added the distinction between `get()` and `get_mutable()` was so that we could swap `copyable_unique_ptr` with an eventual `copy_on_write_ptr` in the future. And by introducing the mutable-non-mutable distinction now, it would inherently do good things after the swap without extensive reasoning. In that spirit, the fact that `unique_ptr::operator*() const` happily returns a non-const reference (like `unique_ptr::get() const` returning a non-const pointer) means that it needs equal treatment.

jwnimmer-tri, 2021-07-08:
The short-term fix can be shadowing the operator, so that _most_ code will behave as we want.

However, even with that, the user can still upcast to unique_ptr (and indeed, sometimes that will happen automatically) and drop the propagated const.  I believe the longer-term fix must be to replace the public inheritance with a conversion operator to unique_ptr instead.  That should fully seal the abstraction.

SeanCurtis-TRI, 2021-07-08:
I've submitted the PR with the short-term fix. I'm waiting to see how CI goes. There might be some usage somewhere where a `const copyable_unique_ptr` is having its value mutated. CI will let me know.

> ... conversion operator to unique_ptr instead

And, presumably, the conversion would itself create a copy so that the `unique_ptr` (with all of its `const` wackiness) would own its own copy as is right and proper. "Convert" in that context feels like the wrong term. And possibly redundant. Here's how I see it:

```c++
copyable_unique_ptr<Foo> my_foo = make_unique<Foo>(param1, param2);
// If I know T is copy constructible
unique_ptr<Foo> foo_copy = make_unique<Foo>(*my_foo);
// If I know T clones.
unique_ptr<Foo> foo_copy2 = my_foo->Clone();
// If we have the "conversion" method.
unique_ptr<Foo> foo_copy3 = my_foo.make_unique();
```

So, the *real* value of a method (e.g., `unique_ptr<T> make_unique() const`) would be to mask the copying mechanism of `T`.

jwnimmer-tri, 2021-07-08:
I say "convert" only because its the term of art https://en.cppreference.com/w/cpp/language/cast_operator.

I suspect that the conversion operator could be overloaded on the value category of `*this` to avoid a copy/clone.

PR #15346 (2021-07-08): Extend copyable_unique_ptr const correctness
In order to be compatble with a future copy-on-write unique pointer, `copyable_unique_ptr` made a distinction between `get()` and `get_mutable()`. The idea being that it would be nice to swap copyable with copy on write with the absolute minimum number of changes. So, by declaring that `const` distinction now, callers will think about mutable access.

However, this allowed getting a non-const *reference* via dereferencing, even from a `const copyable_unique_ptr`. Therefore, we distinguish derefencing a `copyable_unique_ptr` based on the *container's [... cut]
closed 2021-07-08 by SeanCurtis-TRI