RobotLocomotion/drake #15356: Use pkg-config to find yaml-cpp

mwoehlke-kitware, 2021-07-09:
Several externals are found using `pkg-config`. Unfortunately, yaml-cpp is not among them; the location is hard-coded.

This makes it impossible to use a yaml-cpp from any source other than `apt`, which is needed for #1183.

jwnimmer-tri, 2021-07-10:
> Use pkg-config to find yaml-cpp

Issue #13335 and #13309 contain some related background information.  As of last year, pkg-config for yaml-cpp was broken on Focal.  I haven't checked whether or not it's still broken.

> This makes it impossible to use a yaml-cpp from any source other than apt, which is needed for #1183.

My current premise for #1183 is that we will use a different `WORKSPACE` file when compiling for pip / manylinux.  Jamie and I thought it would be implausible to incrementally tweak our current `WORKSPACE` to the point where it was satisfactory.  The new `WORKSPACE` would certainly opt back in to the repository rules for _many_ of the existing externals, but would definitely need to replace some of them outright, at least for now.

Perhaps in the longer term, we could work to fold them back into a single, more general `WORKSPACE` that met both needs, but using for a forked `WORKSPACE` seemed like the smartest path to reach the manylinux goal effectively.

mwoehlke-kitware, 2021-07-12:
> My current premise for #1183 is that we will use a different WORKSPACE file

@jwnimmer-tri, how does one do that?

jwnimmer-tri, 2021-07-12:
For starters, the I'll bet the boring thing would work: have a `workspace-pip.bzl` committed somewhere within Drake (probably at the root?), and then when running in pip-building mode we do `cp workspace-pip.bzl WORKSPACE` before we begin.

mwoehlke-kitware, 2021-08-19:
Okay, given that the current approach (n.b. #15636) seems to be hammering a different hard-coded location, is this still relevant?

jwnimmer-tri, 2021-08-19:
Yes, in any case I think a pip-specific customization will end up as the answer here.  I'm OK if you want to close this issue.

closed 2021-08-31 by mwoehlke-kitware