RobotLocomotion/drake #15361: Missing a constant term in ParseRotatedLorentzConeConstraint

RussTedrake, 2021-07-10:
I think we are missing a constant term in the parsing of the quadratic form in the rotated Lorentz cone constraint....

See the parsing code [here](https://github.com/RobotLocomotion/drake/blob/e59b7fc18dbe80b827d07e4a3283a0c87eda7021/solvers/create_constraint.cc#L590), which calls [this decomposition method](https://github.com/RobotLocomotion/drake/blob/e59b7fc18dbe80b827d07e4a3283a0c87eda7021/math/quadratic_form.h#L29)

The premise of this decomposition is 
```
 * Rewrite a quadratic form xᵀQx + bᵀx + c to
 * (Rx+d)ᵀ(Rx+d)
 * where
 * RᵀR = Q
 * Rᵀd = b / 2
```
But this does form is not rich enough to reconstruct the original quadratic form.  Aren't we missing an offset term `c - dᵀd`?

@hongkai-dai , @TobiaMarcucci ?

RussTedrake, 2021-07-10:
NB -- it looks like https://gitlab.com/libeigen/eigen/-/issues/1479 is closed, so we could also resolve the TODO in quadratic_form.cc

hongkai-dai, 2021-07-10:
> it looks like https://gitlab.com/libeigen/eigen/-/issues/1479 is closed, so we could also resolve the TODO in quadratic_form.cc

I still find issues in using Eigen's `LDLT` decomposition, especially when detecting whether a matrix is positive semi-definite or not.

RussTedrake, 2021-07-10:
After much slack discussion, @hongkai-dai has figured it out.  And I was wrong.  There is no bug.  🤦   We are going to update the documentation slightly so that I don't confuse myself again in 6 months.

RussTedrake, 2021-07-11:
Resolved by #15363.

closed 2021-07-11 by RussTedrake