RobotLocomotion/drake #15366: Support L2-Norm costs in MathematicalProgram

RussTedrake, 2021-07-11:
Proposed roadmap:
- [ ] Introduce `class L2NormCost : public Cost` which implements the generic constraint. 
- [ ] Add `MathematicalProgram::AddL2NormCost(A,b,vars)` after [the old (incorrectly-named) version](https://github.com/RobotLocomotion/drake/blob/master/solvers/mathematical_program.h#L1084) clears deprecation.
- [ ] Teach convex solvers to handle this cost by either:
   - Each solver could opt-in to supporting the `L2NormCost` object, much as they do a `QuadraticCost` object.  (this requires adding another list of `L2NormCost`s to `MathematicalProgram`), or
   - Pass an option to `MathematicalProgram::AddL2NormCost(A,b,vars,use_slack_and_conic_constraint)` that would push the power (burden?) to the user to spell this cost in the convex way that is only better for some optimizers.

For bonus points:
- [ ] Add sugar in `ParseCost(Expression)` to detect the simple form of `sqrt(quadratic)` and create the constraint.
- [ ] Add `MathematicalProgram::AddL2NormCost(expression)`

cc @hongkai-dai

jwnimmer-tri, 2021-11-11:
> Add MathematicalProgram::AddL2NormCost(A,b,vars) after the old version clears deprecation.

As of #15701, this prerequisite is complete.

RussTedrake, 2021-11-12:
@hongkai-dai -- might you be willing to add this to your (admittedly also long) list?

closed 2024-06-25 by hongkai-dai