RobotLocomotion/drake #15406: py mp: Binding_{Evaluatlor} sould really by Binding[Evaluator]

EricCousineau-TRI, 2021-07-16:
Been meaning to fix this.

Currently, we have things like `Binding_LinearCost` which was a holdout before the following code got introduced (#7757).
Will be easy to swap it out.

\cc @hongkai-dai @RussTedrake

RussTedrake, 2021-07-17:
I'm just about to start using this often (I've chosen to accept only the Binding variant of AddCost in my GraphOfConvexSets, instead of the way MathematicalProgram basically replicates all of the Binding constructors in it's public API.  

Since we haven't actually bound the constructor for Binding_foo in pybind yet, I think you could make this change now withtout deprecation(?), but I'm about to start having code that constructs Binding_foo objects hopefully in the next day or two!

EricCousineau-TRI, 2021-07-19:
I think I understand - make the change, then add the deprecation?

EricCousineau-TRI, 2021-07-19:
But that being said, would it be too hard to account for the deprecations and use the spelling?

PR #15409 (2021-07-22): py mp: Use template Binding[{Type}]; deprecate old Binding_{Type}
`pydrake/__init__.py`:
- Do not repeat leading `_` for ExecuteExtraPythonCode
- Allow forwarding of private `_` variables from cc modules

deprecation:
- Fix (Py 2 -> Py 3) for empty `AttributeError` messages
- Add `ModuleShim._install(*, auto_all)`

Resolves #15406

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15409)
closed 2021-07-22 by EricCousineau-TRI