RobotLocomotion/drake #15424: ERROR: Analysis of target '//tools/workspace:check_licenses' failed; build aborted: no such package '@vtk//': java.io.IOException: Error downloading

huihuishen, 2021-07-19:
when I start "bazel build //..." ,it shows out this error, Have annybody met this problem?
[https://drake-packages.csail.mit.edu/vtk/vtk-8.2.0-embree-3.5.1-ospray-1.8.2-python-2.7.15-python-3.6.7-qt-5.9.5-bionic-x86_64.tar.gz, https://s3.amazonaws.com/drake-packages/vtk/vtk-8.2.0-embree-3.5.1-ospray-1.8.2-python-2.7.15-python-3.6.7-qt-5.9.5-bionic-x86_64.tar.gz] to /home/hui/.cache/bazel/_bazel_hui/6cfa01c7c8d4c516137c7fa30c090870/external/vtk/vtk-8.2.0-embree-3.5.1-ospray-1.8.2-python-2.7.15-python-3.6.7-qt-5.9.5-bionic-x86_64.tar.gz: All mirrors are down: [GET returned 403 Forbidden]

jwnimmer-tri, 2021-07-19:
What revision of Drake are you using?  From the URLs in the error message, it looks like you might be trying to use an untagged revision of Drake prior to v0.10.0 from 2019, and our mirror for those dependencies have expired.

huihuishen, 2021-07-19:
Thanks Reply！！
I am trying to compile with the file from https://github.com/guzhaoyuan/drake,since there are some tutorials in https://drake.guzhaoyuan.com/to-get-started. According to your words,the right way for me is to use the new repository?

huihuishen, 2021-07-19:
> What revision of Drake are you using? From the URLs in the error message, it looks like you might be trying to use an untagged revision of Drake prior to v0.10.0 from 2019, and our mirror for those dependencies have expired.

Thanks Reply！！
I am trying to compile with the file from https://github.com/guzhaoyuan/drake,since there are some tutorials in https://drake.guzhaoyuan.com/to-get-started. According to your words,the right way for me is to use the new repository?

jwnimmer-tri, 2021-07-19:
It looks like that is forked from e043e812e4db13cd2c9237be7dadf4189b36bc01.

So, yes -- if you are able to merge or rebase your fork onto one of the official numbered https://github.com/RobotLocomotion/drake/releases tags, then the build should work.

jwnimmer-tri, 2021-07-20:
Closing, since this is basically a bug report against https://github.com/guzhaoyuan/drake not https://github.com/RobotLocomotion/drake.

@huihuishen using Drake from https://github.com/RobotLocomotion/drake/releases/tag/v0.32.0 should work better.  Our website at https://drake.mit.edu/ has extra information.

closed 2021-07-20 by jwnimmer-tri