RobotLocomotion/drake #15427: python bindings for PiecewisePolynomial trajectory construction with matrices

RussTedrake, 2021-07-19:
Our trajectory bindings did not yet support the workflow were trajectories had matrix coefficients.  This was raised in https://stackoverflow.com/questions/68316305/how-to-create-polynomial-trajectories-of-linear-affine-systems-in-python-i-e-ti/68340741?noredirect=1#comment120819751_68340741 .

The fundamental issue is that our trajectory creation routines uses types for disambiguation; the conversion from numpy arrays becomes ambiguous.  See, for example, the [ZeroOrderHold methods](https://drake.mit.edu/doxygen_cxx/classdrake_1_1trajectories_1_1_piecewise_polynomial.html#a06278ca4588ba1adb004b84fdfc62f3c).  Here are three possible solutions:
1) we teach the c++ std::vector version how to handle the Eigen::MatrixXd case.  This makes the c++ code worse, I think, so I avoided it.
2) we teach the python bindings (only) how to runtime divert from one case to another; I've prototyped that [here](https://github.com/RobotLocomotion/drake/compare/master...RussTedrake:matrix_trajectories?expand=1) to see what it would look like.  it's not pretty.
3) we introduce a new name for the matrix variants in c++ (presumably deprecating the old ones), and add python bindings for those distinct names.  That would mean, e.g. `PiecewisePolynomial::ZeroOrderHold(VectorXd, MatrixXd)` and `PiecewisePolynomial::ZeroOrderHoldMatrix(std::vector<T>, std::vector<MatrixX<T>>)`.  I think that the current VectorXd workflow is the much more common one, so I think it would be the one that deserves the shorter name.

@jwnimmer-tri -- what do you think?

jwnimmer-tri, 2021-07-19:
My intuition says that the pydrake binding should accept a multi-dimensional `numpy` array directly, and within the binding it should unzip the multi-dimensional array into a `const std::vector<MatrixX<T>>& samples` to pass along to C++ function that takes that signature (for ZOH, that's the first overload listed in the header file).

Your option (2) starts along those lines, but:

(a) I don't see why it needs decide which C++ method to call?  It seems like the more general matrix-valued sample constructor can always handle vector-valued samples.  The narrower one seems like sugar, not functionally different.  (This is borne out in the cc file, where the vector one calls the matrix one.)  This seems easy enough to fix on your branch.

(b) The row/col major stuff is awkward with `std::vector`, as shown in your branch.

For (b), I wonder if the below signature for pydrake works?  Maybe by telling pybind11 that we want Eigen storage order, the shape slicing will better by default?

```c++
def_static(
    "ZeroOrderHold",
    [](const std::vector<T>& breaks,
       const Eigen::Matrix<Eigen::MatrixXd, Eigen::Dynamic, 1>& samples) {
      ...
    }
```

In any case, what matters if the test cases and semantics we want.  I don't think "ugly" is a reason to reject bindings choices -- it's all ugly under the hood.  I don't think spelling overloads differently as per (3) takes us in a good direction, and I don't even understand how (1) gets you to what you want.  I would say find the best flavor of (2) you can manage, and call it good enough.

jwnimmer-tri, 2021-07-19:
Or maybe overloading on `const vector<vector<T>>&`  and `const vector<vector<vector<T>>>&`  is enough to infer a proper shape?  Those would match the C++ shapes directly, I think, with the bindings code just doing some extra copying.

RussTedrake, 2021-07-19:
Thanks.  I'll give that a spin.

RussTedrake, 2021-07-20:
`vector<vector<T>>&` worked well, and it turns out I can still use `vector<MatrixX<T>` alongside it.  I'll PR the ZeroOrder Hold binding now for review, and we can follow-up with the other methods.

PR #15442 (2021-07-26): [py/trajectories] bind matrix versions of PiecewisePolynomial
Takes the pattern established in #15433, and implements it for the remaining construction methods.

Resolves #15427.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15442)
commit 90136090 (2021-07-26): [py/trajectories] bind matrix versions of PiecewisePolynomial (#15442)

* [py/trajectories] bind matrix versions of PiecewisePolynomial

Takes the pattern established in #15433, and implements it for the remaining construction methods.

Resolves #15427.
closed 2021-07-26 by RussTedrake