RobotLocomotion/drake #15434: [framework] Cryptic error from AbstractValue

RussTedrake, 2021-07-20:
In system_test.cc we have 
```
  std::unique_ptr<AbstractValue> DoAllocateInput(
      const InputPort<T>& input_port) const override {
    if (input_port.get_index() == 0) {
      return AbstractValue::Make<std::string>();
    } else {
      return std::make_unique<Value<BasicVector<T>>>(TestTypedVector<T>{});
    }
  }
```
when I wrote
```
  const auto& typed_input = test_sys_.AddAbstractInputPort();
  std::unique_ptr<AbstractValue> input_value =
      test_sys_.AllocateInputAbstract(typed_input);
  DRAKE_EXPECT_NO_THROW(input_value->get_value<BasicVector<double>>());
```
I get the error message
```
Failed
Expected: Does not throw:
  input_value->get_value<TestTypedVector<double>>()
Actual: Throws std::logic_error
  AbstractValue: a request to cast to 'drake::systems::MyVector<double,1>' failed because the actual type was 'drake::systems::MyVector<double,1>'.
```
which led me to look for possible ways the type could have been multiply defined.  It turns out the proper invocation was
```
  DRAKE_EXPECT_NO_THROW(input_value->get_value<BasicVector<double>>());
```

I'd say that's not a great error message.

jwnimmer-tri, 2021-07-20:
It looks like there might be a paste-o in your above.  The two examples appear to be identical?

```c++
  // What you wrote.
  DRAKE_EXPECT_NO_THROW(input_value->get_value<BasicVector<double>>());
  // Proper invocation.
  DRAKE_EXPECT_NO_THROW(input_value->get_value<BasicVector<double>>());
```

I suspect the "what you wrote" was actually `get_value<TypedTestVector<double>>`?

I suppose `AbstractValue::ThrowCastError` should report the `static_type_info()` in addition to the the `type_info()`, at least if the two are not identical.  Because of the sugar we have in `(Input|Output)Port::Eval` most end users would not encounter this, but it seems easy to improve in any case.

RussTedrake, 2021-07-21:
I've fixed the paste-o (you were correct).  Thanks.

jwnimmer-tri, 2021-11-12:
Note that part of the problem here is that the `TestTypedVector` itself is broken:

https://github.com/RobotLocomotion/drake/blob/3b5fbf929f1eb1e439c70db22bd3a5b3ab13a523/systems/framework/test_utilities/my_vector.h#L46-L52

rpoyner-tri, 2021-11-12:
> Note that part of the problem here is that the `TestTypedVector` itself is broken:
> 
> https://github.com/RobotLocomotion/drake/blob/3b5fbf929f1eb1e439c70db22bd3a5b3ab13a523/systems/framework/test_utilities/my_vector.h#L46-L52

Is this regret part of this issue, or another altogether?

jwnimmer-tri, 2021-11-12:
I guess what I was saying is that I'm not sure this issue was ever likely to impact any users.  We have some completely bone-headed code in a unit test sample file, which tripped up Russ.

We _should_ purge that sample code so that nobody else cargo-cults it (though hopefully the TODO is enough to dissuade them), but I'm fine closing this issue with the current PR, and leaving that purge for another day.

PR #16089 (2021-11-12): Value: fix types in bad cast message
Closes: #15434

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16089)
closed 2021-11-12 by jwnimmer-tri