RobotLocomotion/drake #15449: DecomposePSDmatrixIntoXtransposeTimesX vs Eigen 3.4

jwnimmer-tri, 2021-07-21:
Part of #15142.

In our Eigen 3.4 integration testing, we see this:

```c++
GTEST_TEST(TestDecomposePSDmatrixIntoXtransposeTimesX, Test4) {
  // Y is a rank 1 psd matrix.
  Eigen::Matrix3d Y;
  // clang-format off
  Y << 1, 2, -3,
       2, 4, -6,
       -3, -6, 9;
  // clang-format on
  CheckDecomposePSDmatrixIntoXtransposeTimesX(Y, 1E-15);
}
```

Output:

```
[ RUN      ] TestDecomposePSDmatrixIntoXtransposeTimesX.Test4
math/test/quadratic_form_test.cc:21: Failure
Expected equality of these values:
  qr.rank()
    Which is: 1
  X.rows()
    Which is: 2
[  FAILED  ] TestDecomposePSDmatrixIntoXtransposeTimesX.Test4 (0 ms)
```

To reproduce:
```
git fetch upstream pull/15142/head && git checkout FETCH_HEAD
bazel test //math:quadratic_form_test
```

jwnimmer-tri, 2021-07-21:
+@hongkai-dai this isn't super urgent, but when you have time it would be helpful if you could take a look and offer advice here?  Maybe try to figure how where and how it's going wrong?

jwnimmer-tri, 2021-07-21:
The `//systems/controllers:linear_quadratic_regulator_test` is also segfaulting during `R_cholesky.solve` and might be related.

hongkai-dai, 2021-07-26:
Sorry for the late reply. For the failure in `quadratic_form_test`, I changed the tolerance to 2E-15 and now it works
```
diff --git a/math/test/quadratic_form_test.cc b/math/test/quadratic_form_test.cc
index fe844f152a..907d4e4d26 100644
--- a/math/test/quadratic_form_test.cc
+++ b/math/test/quadratic_form_test.cc
@@ -59,7 +59,7 @@ GTEST_TEST(TestDecomposePSDmatrixIntoXtransposeTimesX, Test4) {
        2, 4, -6,
        -3, -6, 9;
   // clang-format on
-  CheckDecomposePSDmatrixIntoXtransposeTimesX(Y, 1E-15);
+  CheckDecomposePSDmatrixIntoXtransposeTimesX(Y, 2E-15);
 }
 
 GTEST_TEST(TestDecomposePSDmatrixIntoXtransposeTimesX, Test5) {
```
This function uses an Eigen value decomposition.

For the linear quadratic regulator, it uses a robust Choleskly decomposition LDLT. My experience was that Eigen's robust Cholesky decomposition is not numerically robust. Fixing this problem probably requires a separate PR to replace the LDLT solver to a different solver (possibly an eigen value decomposition).

jwnimmer-tri, 2021-07-26:
Aha, for some reason I didn't consider changing the `zero_tol`.  I'll try that out and then PR it.

PR #15488 (2021-07-27): [math] Loosen test tolerance for quadratic_form
We found that the prior value of 1E-15 was too tight with Eigen 3.4.

Closes #15449.
Towards #15142.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15488)
closed 2021-07-27 by rpoyner-tri