RobotLocomotion/drake #15454: How to show custom Multibody::Frame in Drake::Visulizer??

huihuishen, 2021-07-22:
> Before you post:
Hello, Is it possible to display custom frame to see if the position of the frame  is correct?

> * Have you reviewed the documentation on
[Getting Help](https://drake.mit.edu/getting_help.html)?
> * Have you reviewed existing GitHub issues and StackOverflow posts?
> * Have you rerun `install_prereqs.sh` on your current commit of `drake`?

> If you have encountered a bug (installation, segfaults, etc.), please try to
post reproduction steps and as much relevant output as possible. Consider
posting long snippets as a [Gist](https://gist.github.com/).

jwnimmer-tri, 2021-07-22:
The https://drake.mit.edu/getting_help.html page linked from our issue template says "If you are seeking assistance ... please post a question on StackOverflow with the drake tag."  Therefore, this question belongs on StackOverflow please, not as a GitHub issue.  We like to reserve GitHub issues for bugs and feature requests only.

closed 2021-07-22 by jwnimmer-tri

huihuishen, 2021-07-22:
> The https://drake.mit.edu/getting_help.html page linked from our issue template says "If you are seeking assistance ... please post a question on StackOverflow with the drake tag." Therefore, this [...]

ok,thank you very much!