RobotLocomotion/drake #15461: GetActuationFromArray(.), GetPositions(ModelInstanceIndex), etc. should provide non-heap allocating versions

edrumwri, 2021-07-23:
We desire to access parts of vectors/state using already allocated Eigen vectors. It's presently very, very cumbersome to access subvectors of actuation vectors corresponding to model instances without allocating heap.

I suggest adding functions having the signature `GetActuationFromArray(ModelInstanceIndex, const Eigen::Ref<const VectorX<T>>&, EigenPtr<VectorX<T>>*)`, etc. to the API.

jwnimmer-tri, 2021-11-12:
@edrumwri I believe many of these are resolved by #15900.

Do you still want `GetActuationFromArray`, or others that do not yet exist?

xuchenhan-tri, 2022-04-27:
Following up on this issue -- @edrumwriare there still acute needs for more sugars for more getters?

sherm1, 2022-04-27:
cc @edrumwri

edrumwri, 2022-11-22:
This issue has been resolved for us.

closed 2022-11-22 by edrumwri