RobotLocomotion/drake #15462: MBP input actuation should be providable as a single vector (sugar)

edrumwri, 2021-07-23:
When performing inverse dynamics calculations (involving B*u), the output is "u". That u must be demultiplexed in order to feed it into a MBP for which the plant has actuators grouped into model instances - this would be the case for multi-arm robots working in concert, for example. Not only is that demultiplexing expensive (see #15461), it's cumbersome as well and prevents, for example, Drake's inverse dynamics controller from working "out of the box" on such plants.   

Per discussion with @amcastro-tri, a convenience port that packs actuation for all model instances into a single vector (to be symmetric with packed positions+velocities as already done in get_state_output_port()) would seem to be the right solution.

jwnimmer-tri, 2021-07-24:
Could you speak to what you'd expect to happen in case the user connects _both_ the stacked input vector port as well as some (or all) of the per-model input ports?  Should that be an error, or should the plant sum over both inputs?

amcastro-tri, 2021-07-26:
Also, is your use case @edrumwri with the inverse dynamics controller? IMO the API for that class is just broken. That class should provide an API to perform inverse dynamics on a specific model instance so that its output is already consistent with the simulation model. I am asking in case the solution really is on the `InverseDynamicsController` instead of MBP?

edrumwri, 2021-07-30:
> Could you speak to what you'd expect to happen in case the user connects _both_ the stacked input vector port as well as some (or all) of the per-model input ports? Should that be an error, or should the plant sum over both inputs, or ...?

I think it would be best if the per-model ports were deprecated long-term (I have never found them to be convenient) and, short-term, if an error were thrown if both the stacked and some/all per-model ports were connected/fixed. So my description of this as "sugar" was misleading.

edrumwri, 2021-07-30:
> Also, is your use case @edrumwri with the inverse dynamics controller? IMO the API for that class is just broken. That class should provide an API to perform inverse dynamics on a specific model [...]

No- we don't use the inverse dynamics controller.

jwnimmer-tri, 2021-07-30:
> I think it would be best if the per-model ports were deprecated long-term ...

In our TRI simulations, we frequently use per-model-instance controllers that rely on the per-model-instance actuation input.  I find it unlikely that we would deprecate those input ports.  (We'd need to write a model-instance mux; seems best to just let the plant keep doing it, especially if it is more efficient for cache invalidation in cases where the controllers are multi-rate.)

It still seems plausible, though, to have it be an error to try to use both flavors at once.  Users who want per-model can do that; users who want anything else would use the new, unified input.

edrumwri, 2021-07-31:
> In our TRI simulations, we frequently use per-model-instance controllers that rely on the per-model-instance actuation input. I find it unlikely that we would deprecate those input ports. (We'd need [...]

It's no skin off of our nose either way. We have only found the per-model-instance actuation to only be a hindrance but of course, YMMV.

xuchenhan-tri, 2022-04-28:
I'll take a look at this in case this issue is still of interest.