RobotLocomotion/drake #15463: [colab] Drake is broken on google colab

RussTedrake, 2021-07-23:
Probably related to #1183.  Blocking #13683.

Our [setup_drake_colab](https://github.com/RobotLocomotion/drake/blob/0a588a5b31a27dd556c78896a8ad755d6cefe2a5/tools/install/colab/setup_drake_colab.py) script is currently broken, and has been broken since Colab upgraded to python 3.7.  

As noted here: https://github.com/RussTedrake/underactuated/issues/451, for my course notes, I've been living on a peg revision of drake built by @jamiesnape using [this fork of drake](https://github.com/RobotLocomotion/drake-python3.7).  

If colab were to upgrade to 3.8, then the issue would be resolved.  But according to https://github.com/googlecolab/colabtools/issues/1880 the colab teams' most recent post says "Not sure, but probably on Dec 26, 2021 drop support for Python 3.7 and update to Ubuntu 20.04 Python 3.8".

I am teaching a course this fall that is hoping to use drake on colab.  I need to either:
1) resolve this issue,
2) continue to limp along on manually built python3.7 binaries,
3) find a colab alternative.  (would need to decide that asap)

jwnimmer-tri, 2021-07-24:
> ... the colab teams' most recent post ...

Aside: Is segatrade a colab maintainer?  Given https://github.com/googlecolab/colabtools/issues/1422, my inference of status was a community member citing some relevant numpy/scipy dates, not an official word.

(Not that it changes anything about our inability to anticipate colab platform upgrades.)

RussTedrake, 2021-10-24:
Now that we have `pip install drake`, we should advertise this as the solution for colab.  And probably update the `setup_drake_colab` script to do the pip install and print a message to the console saying "you't don't need this script anymore.. just call pip install".

PR #16073 (2021-11-15): [colab] Replace installer script with an error stub
Closes #15463.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16073)
closed 2021-11-15 by RussTedrake