RobotLocomotion/drake #15478: systems: Unclear how to (quickly) check / assert caching is enabled by default?

EricCousineau-TRI, 2021-07-26:
I briefly skimmed through:
- https://drake.mit.edu/doxygen_cxx/group__cache__design__notes.html
- Current `drake`, looking for usages of `EnableCaching()`
but did not see any clear indicator that caching was enabled by default.

I then looked through PRs, and found #10979, where the only evidence was that `DisableCaching()` was removed from `LeafSystem::DoAllocateContext()`.
From perusing implementation, I also see that it *could* be inferred from the default value of `CacheEntryValue::flags_{kValueIsOutOfDate}` (vs. what I assume would *disable* by default, `kCacheEntryIsDisabled`).
https://github.com/RobotLocomotion/drake/blob/v0.32.0/systems/framework/cache.h#L600-L630

Suggestion is to add docs somewhere indicating that caching is enabled by default; perhaps to `@ref cache_design_notes`?

@sherm1 When you have a chance, could you advise? I'm happy to make a stab at adding the docs.

\cc @calderpg-tri (per Anzu PR 7339)

jwnimmer-tri, 2021-07-26:
FWIW if someone asked me where to find the documentation for the default enabled/disabled setting, the place I would look is in the API comments for `SystemBase::DisableCaching` and/or `SystemBase::EnableCaching`.

EricCousineau-TRI, 2021-07-26:
Aye; but, yeah, I had briefly looked at `ContextBase::{EnableCaching,DisableCaching}`, and did not see mention of "By default, caching is {enabled|disabled}".

sherm1, 2021-07-26:
I'll add some clarifying comments.

EricCousineau-TRI, 2021-07-26:
Thanks! Reflecting in assignment

PR #15480 (2021-07-26): Clarify that caching is enabled by default
Add some documentation to clarify that caching is enabled by default in Drake.

Resolves #15478

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15480)
closed 2021-07-26 by sherm1