RobotLocomotion/drake #15496: mbp: Indicate which bodies are influenced by DoFs / joints?

EricCousineau-TRI, 2021-07-28:
In Anzu, I'd like to be able to produce a list of all bodies kinematically affected by a set of joints (or via a "DoF mask"). This is intended to be used for for collision avoidance.

At present, to workaround, I will effectively build my own map of "kinematic path to world" (hax!).

Ideally, it'd be nice to at least have `MultibodyTreeTopology::GetKinematicPathToWorld()` exposed (in some form).

Possible alternatives:
- Given a `Joint` / `JointIndex`, list all bodies kinematically affected by this joint.
- Given a set of joints, list all bodies kinematically affected by this joint.
- Perhaps something already exposed that I missed?

\cc @calderpg-tri @siyuanfeng-tri

EricCousineau-TRI, 2021-07-28:
Assigning @sherm1 for disposition, if possible!

xuchenhan-tri, 2022-04-26:
@joemasterjohn would this be easy to achieve?

amcastro-tri, 2022-04-28:
question @EricCousineau-TRI with "Given a Joint / JointIndex, list all bodies kinematically affected by this joint.", do yo mean the two bodies connected to this joint?

To be more precise, what API/APIs would you like to have?

EricCousineau-TRI, 2022-04-28:
The use case is for collision avoidance - ideally, we only compute distances / gradients for bodies that actually can be affected by given robot arm(s) / DoFs.

Something like this (or perhaps more intelligent / efficient)? 
This is what we have in Anzu, under `robot_bridge_common.{h,cc}`:
(I think this overfit to tree topology, but that's prolly fine for now?)
```cc
// TODO(eric.cousineau): Use a better Drake-based topology query (#15496).
/**
 * Computes all bodies that are outboard of the joints dictated by
 * `active_dof`.
 * @pre active_dof has at least one true entry, and is compatible with `plant`.
 * @pre All joints of `plant` are either 0- or 1-dof.
 * @pre All joints only affect the velocity jacobian of the *outboard* (child)
 * bodies.
 */
std::unordered_set<drake::multibody::BodyIndex>
ComputeBodiesKinematicallyInfluencedByActiveDof(
    const drake::multibody::MultibodyPlant<double>& plant,
    const std::vector<bool>& active_dof) {
  DRAKE_DEMAND(plant.num_positions() == plant.num_velocities());
  DRAKE_DEMAND(
      static_cast<int>(active_dof.size()) == plant.num_positions());
  DRAKE_DEMAND(CountNumActiveDof(active_dof) > 0);
  using BodySet = std::unordered_set<BodyIndex>;
  using JointSet = std::unordered_set<JointIndex>;
  using JointToBodies = std::map<JointIndex, BodySet>;

  JointSet active_joints;
  JointToBodies joint_to_immediate_child_bodies;
  JointToBodies joint_to_transitive_child_bodies;

  for (JointIndex i{0}; i < plant.num_joints(); ++i) {
    const Joint<double>& joint = plant.get_joint(i);
    if (joint.num_positions() == 0) {
      // Topology accounted for welds using `GetBodiesWeldedTo()`.
      continue;
    }
    DRAKE_THROW_UNLESS(joint.num_positions() == 1);

    // Account for active joints.
    if (active_dof[joint.position_start()]) {
      active_joints.insert(i);
    }

    // Get immediate parent bodies.
    BodySet immediate_parent_bodies;
    for (auto* body : plant.GetBodiesWeldedTo(joint.parent_body())) {
      immediate_parent_bodies.insert(body->index());
    }

    // Get child bodies.
    BodySet& immediate_child_bodies =
        joint_to_immediate_child_bodies[i];
    for (auto* body : plant.GetBodiesWeldedTo(joint.child_body())) {
      immediate_child_bodies.insert(body->index());
      joint_to_transitive_child_bodies[i].insert(body->index());
    }

    // Check all prior joints for parenthood. Use this to naively compute
    // transitive child bodies.
    for (JointIndex i_prev{0}; i_prev < i; ++i_prev) {
      auto& child_bodies_prev = joint_to_transitive_child_bodies[i_prev];
      if (HasIntersection(immediate_parent_bodies, child_bodies_prev)) {
        MergeInto(
            immediate_child_bodies, &joint_to_transitive_child_bodies[i_prev]);
      }
    }
  }

  // Accumulate all influenced bodies.
  BodySet influenced_bodies;
  for (JointIndex i : active_joints) {
    MergeInto(joint_to_transitive_child_bodies[i], &influenced_bodies);
  }

  DRAKE_DEMAND(influenced_bodies.size() > 0);
  return influenced_bodies;
}
```

xuchenhan-tri, 2022-04-28:
I'd imagine the numerical alternative is insufficient because it gives the wrong information in an unlucky configuration. I think exposing topology information is a reasonable and long-standing request. This is related to #11307 and #9747. 

Looks like there had been a valiant effort in #10068 but somehow it was stalled? Skimming over that PR it seems like the goal there is not only to query the topology but also modify it? Is it possible that we take baby steps and provide read-only results first? That would still seem like a good improvement to me.

sherm1, 2022-04-28:
The MultibodyGraphModeler etc code in #10068 is in pretty good shape. What we lacked is a plan for putting it to use in Drake.

amcastro-tri, 2022-04-28:
FTR, we do have `MultibodyGraph` in Drake master, which encodes Body/Joint graph topology. I think for this particular request the `MultibodyGraph` would be the ideal place where to implement it.
We might even consider exposing the `MultibodyGraph` in public APIs for this kind of topological only queries?

xuchenhan-tri, 2022-05-17:
I can take a poke at this in the near future (next quarter) if no one is working on it already.

PR #17800 (2022-09-07): [multibody] Compute bodies affected by joints
Closes #15496.

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17800)
closed 2022-09-07 by xuchenhan-tri