RobotLocomotion/drake #15524: scene_graph_inspector: Possible to query `CollisionFiltered(FrameId, FrameId)`?

EricCousineau-TRI, 2021-07-29:
Context: For #15026 (\cc @azeey @jennuine)
From Slack: https://drakedevelopers.slack.com/archives/C2WBPQDB7/p1627584416099400

`SceneGraphInspector` has `CollisionFiltered(GeometryId, GeometryId)`, but does not have `CollisionFiltered(FrameId, FrameId)`.
For workflows such as `multibody_plant_subgraph`, it'd be nice to copy over collision filtering with the same semantics from the source to the destination `SceneGraph`.

For example (and paraphrasing Sean's example):
- User adds `frame_1`, with `geometry_a` and `geometry_b`.
- User declares filtering within geometry set containing only `frame_1`.
- User then adds `geometry_c`.
- I, uh, think this implies `geometry_{a,b,c}` all have their collisions filtered?

*Note*: This assumes that collision filtering has different, uh, "transient" properties of filtering w.r.t. frames. If the "transient" semantics are the same, then perhaps we close this issue hehe

SeanCurtis-TRI, 2021-07-29:
1. I'm not sure what you mean by "transient" in this case. There's a current PR that uses the word "transient" w.r.t. collision filters to mean something very specific, and I don't believe that it's what you mean here. My best assumption is that this is referencing the fact that any frame-based declaration of collision filtering (e.g., via `collision_filter_group` in URDF/SDF files or declarations of filters at via the API) become lost and only the *effect* of the declarations on the set of filtered *geometry pairs* is preserved. Looking at the final effect and inferring the commands (and corresponding *intent*) that might have led to it is not well-formed. The function is not generally invertible.

2. I believe your example is too toy-like to illustrate what needs to be illustrated. (It may be fine and merely requires further elaboration)
    - Presumably frame 1 with geometries a and b are part of one subgraph.
    - Where does geometry c come from? A different sub graph? The same?

As I think aloud, as it were, it seems to me the only issue you have is collision filters *between* subgraphs. As I see it, the subgraph is an atomic component in your workflow. A subgraph is either added in its entirety or not. As such, any thing relating the internal workings of the subgraph are preserved. In this case, just making sure the effect of filter declarations is preserved is sufficient.

I can imagine that one can construct a plant from multiple sub graphs and then add relationships between components between those subgraphs (e.g., filter collisions). That can be considered either as two subgraphs with some metadata, or a hierarchical subgraph. If that new graph were considered to be a subgraph itself, then one would assume that using as a whole would likewise benefit from the preservation of "internal" relationships. If, on the other hand, you're thinking you can add the two constituent subgraphs to a third plant and, somehow, the metadata declarations can go along with it "for free", we're back to the inability to infer those filters. You do use the word "replayed" which suggests that you expect to maintain such metadata and are willing to exercise it again. From that perspective, `CollisionFilterDeclarations` will be your friend.

All of that babbling is to illustrate that I'm not sure what the actual *problem* is that this issue is addressing.

SeanCurtis-TRI, 2021-07-29:
I just realized that everything I wrote was to the larger question for which the `CollisionFiltered(FrameId, FrameId)` is supposed to serve.  Ignoring the application domain for a minute, let's talk purely about this function.

As stated above, all collision filters are stored *purely* on a *geometry*-pair basis. We have no way of knowing if two geometries are filtered because they were explicitly declared to be filtered or because their parent frames were used as stand ins for them. Thus, looking at the current state can provide *no* guarantees about inferring what the original command was (let alone the original intent).

What it can do is consider the sets of geometries affixed to the same frames. With respect to the two sets it can make the following determination and report accordingly:
  - There are no filtered pairs between the sets. (unfiltered)
  - All pairs between the sets are filtered. (filtered)
  - Some (but not all) pairs are filtered. (partially filtered)

I'm a bit worried at reporting "filtered" as a result, because I'd hate for the user to come away with the belief that the *frames* are filtered and if they were to introduce any geometry to those frames that somehow that geometry would "inherit" the frames' filtered status. I believe that can best be alleviated via documentation and a different method name: `AllGeometryiesFiltered(FrameId, FrameId)` or some such thing.

EricCousineau-TRI, 2021-07-29:
Yup, I had forgotten the "purely on a geometry-pair bases" in this part of your docs (as Jeremy pointed out on Slack):
https://github.com/RobotLocomotion/drake/blob/ef428c3616579fc3e3c62a9461ca3914e9ce4445/geometry/collision_filter_manager.h#L64-L76
(sorry for not putting the "Note" more prominently!)

Since I misunderstood; given this then yeah, this issue can be closed - there's definitely no need to try and infer the intent of collision filtering (I hope).
Thanks, and sorry for the noise!

closed 2021-07-29 by EricCousineau-TRI