RobotLocomotion/drake #15549: binding for le, ge (and presumably others) fails to match expression if it's the second argument.

RussTedrake, 2021-08-02:
Discovered here: https://stackoverflow.com/questions/68615278/runtimeerror-you-should-not-call-bool-nonzero-on-formula

```
from pydrake.all import Variable, le, ge
x = Variable("x")

print(2 <= x**2)
print(x**2 <= 2)
print(le(x**2, 2))
#print(le(2, x**2))  <== this crashes

print(ge(x**2, 2))
#print(ge(2, x**2))  <== this too.
```
cc @EricCousineau-TRI

soonho-tri, 2021-08-03:
@EricCousineau-TRI , those `le` and `ge`s, where do we define them?

soonho-tri, 2021-08-03:
Never mind. I think they are from these lines:
https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/_math_extra.py#L34-L46

soonho-tri, 2021-08-03:
FYI, It works when we have:

```python
print(operator.le(2, x**2))
```

But it doesn't work when we have the following line:

```python
print(le(2, x**2))  # short-hand of np.vectorize(operator.le)(2, x**2)
```

and the error message is:
```
RuntimeError: You should not call `__bool__` / `__nonzero__` on `Formula`. 
If you are trying to make a map with `Variable`, `Expression`, or `Polynomial` as keys 
(and then access the map in Python), please use pydrake.common.containers.EqualToDict`.
```

soonho-tri, 2021-08-03:
@EricCousineau-TRI , my understanding is that we found corner cases where https://github.com/RobotLocomotion/drake/pull/11171 , the workaround for #8315, does not work well. Could you investigate from here?

soonho-tri, 2021-08-03:
FYI, the runtime error comes from this line (on my mac which is using python3.9 + numpy-1.21),
https://github.com/numpy/numpy/blob/maintenance/1.21.x/numpy/lib/function_base.py#L2201.

In this case, `func = <built-in function le>` and `inputs = [2, pow(x, 2)]`. I don't understand why `func(*inputs)` throws this runtime error.

nolie-rolie, 2021-08-03:
I wanted to add on some more information. I hope this is helpful. 

```
p = p_now + h*J_G.dot(v)[3:6]
  for i in range(p.shape[0]):
    # prog.AddConstraint(ge(p[i], lower_bound[i])) # <-- TypeError
    # prog.AddConstraint(le(p[i], upper_bound[i])) # <-- TypeError
    # prog.AddConstraint(le(lower_bound[i], p[i])) # <-- RuntimeError
    # prog.AddConstraint(ge(upper_bound[i], p[i])) # <-- RuntimeError
    prog.AddConstraint(p[i] >= lower_bound[i]) # <-- working
    prog.AddConstraint(p[i] <= upper_bound[i]) # <-- working
    # prog.AddConstraint(lower_bound[i] <= p[i]) # <-- RuntimeError
    # prog.AddConstraint(upper_bound[i] >= p[i]) # <-- RuntimeError
```

Not sure why, but in my case, 'ge()' and 'le()' did not work regardless of the order of the arguments. Also, leading the inequality sign with a float rather than an expression caused a RuntimeError. Here is a link to my [colab notebook](https://colab.research.google.com/drive/1RxFtWnqgaZgMtGmAnRTiDRERossY2lL0#scrollTo=yl128hSMdGJW).

jwnimmer-tri, 2021-08-04:
The first step is writing more unit tests, so that we have a wide measurement of the current, undesired behavior.  I've start that under #15558 -- open for increments pushes / merges from others.

jwnimmer-tri, 2021-08-04:
Here's a cleaned-up backtrace for the first test failure in that branch:
```
#0  drake::pydrake::<lambda(const drake::symbolic::Formula&)>::operator()(const drake::symbolic::Formula &) const (__closure=0x10f7098) at bindings/pydrake/symbolic_py.cc:530
#11 in slot_nb_bool () at ../Objects/typeobject.c:6049
#12 in PyObject_IsTrue (v=<pydrake.symbolic.Formula at remote 0x7fffd9b492b0>) at ../Objects/object.c:1284
#13 PyObject_IsTrue (v=<optimized out>, v=<optimized out>) at ../Objects/object.c:1263
#14 in OBJECT_less (args=<optimized out>, dimensions=<optimized out>, steps=<optimized out>, __NPY_UNUSED_TAGGEDfunc=<optimized out>) at numpy/core/src/umath/loops.c.src:2811
#15 in trivial_three_operand_loop (op=op@entry=0x7fffffff9a70, innerloop=0x7ffff5aabcd0 <OBJECT_less>, innerloopdata=0x0) at numpy/core/src/umath/ufunc_object.c:1339
#16 in execute_legacy_ufunc_loop (arr_prep_args=0x0, arr_prep=0x7fffffff96d0, buffersize=8192, order=NPY_KEEPORDER, dtypes=0x7fffffff95d0, op=0x7fffffff9a70, trivial_loop_ok=<optimized out>, ufunc=0xbaf520) at numpy/core/src/umath/ufunc_object.c:1700
#17 PyUFunc_GenericFunction (ufunc=ufunc@entry=0xbaf520, args=args@entry=(<numpy.ndarray at remote 0x7fffd9b43850>, <pydrake.symbolic.Expression at remote 0x7fffd9b3f230>), kwds=kwds@entry=0x0,  op=op@entry=0x7fffffff9a70) at numpy/core/src/umath/ufunc_object.c:2749
#18 in ufunc_generic_call (ufunc=0xbaf520, args=(<numpy.ndarray at remote 0x7fffd9b43850>, <pydrake.symbolic.Expression at remote 0x7fffd9b3f230>), kwds=0x0) at numpy/core/src/umath/ufunc_object.c:4411
#21 in do_richcompare (op=0, w=<pydrake.symbolic.Expression at remote 0x7fffd9b3f230>, v=<numpy.ndarray at remote 0x7fffd9b43850>) at ../Objects/object.c:708
#22 PyObject_RichCompare () at ../Objects/object.c:756
#23 in gentype_richcompare (self=<optimized out>, other=<pydrake.symbolic.Expression at remote 0x7fffd9b3f230>, cmp_op=0) at numpy/core/src/multiarray/scalartypes.c.src:1135
#24 in double_richcompare (self=<numpy.float64 at remote 0x7ffff2cf1b70>, other=<pydrake.symbolic.Expression at remote 0x7fffd9b3f230>, cmp_op=<optimized out>) at numpy/core/src/umath/scalarmath.c.src:1532
#25 in do_richcompare (op=0, w=<pydrake.symbolic.Expression at remote 0x7fffd9b3f230>, v=<numpy.float64 at remote 0x7ffff2cf1b70>) at ../Objects/object.c:708
#26 PyObject_RichCompare () at ../Objects/object.c:756
#27 in op_lt () at ../Modules/_operator.c:107
#29 in do_call_core (kwdict=0x0, callargs=(<numpy.float64 at remote 0x7ffff2cf1b70>, <pydrake.symbolic.Expression at remote 0x7fffd9b3f230>),  func=<built-in method lt of module object at remote 0x7ffff68ded18>) at ../Python/ceval.c:5116 f=Frame 0x11faed8, for file /usr/lib/python3/dist-packages/numpy/lib/function_base.py, line 2769, in _get_ufunc_and_otypes (self=<vectorize(pyfunc=<built-in method lt of module object at remote 0x7ffff68ded18>, cache=False, signature=None, _ufunc=None, __doc__='lt(a, b) -- Same as a<b.', otypes=None, excluded=set(), _in_and_out_core_dims=None) at remote 0x7fffd9eccfd0>, args=[<numpy.ndarray at remote 0x7fffd9b43530>, <numpy.ndarray at remote 0x7fffd9b43800>], inputs=[<numpy.float64 at remote 0x7ffff2cf1b70>, <pydrake.symbolic.Expression at remote 0x7fffd9b3f230>])) at ../Python/ceval.c:754
#33 in fast_function.lto_priv () at ../Python/ceval.c:4992 f=Frame 0x11facc8, for file /usr/lib/python3/dist-packages/numpy/lib/function_base.py, line 2809, in _vectorize_call (self=<vectorize(pyfunc=<built-in method lt of module object at remote 0x7ffff68ded18>, cache=False, signature=None, _ufunc=None, __doc__='lt(a, b) -- Same as a<b.', otypes=None, excluded=set(), _in_and_out_core_dims=None) at remote 0x7fffd9eccfd0>, func=<built-in method lt of module object at remote 0x7ffff68ded18>, args=(<numpy.ndarray at remote 0x7fffd9b43530>, <numpy.ndarray at remote 0x7fffd9b43800>))) at ../Python/ceval.c:754
#47 in PyObject_Call (kwargs=0x0, args=(<numpy.ndarray at remote 0x7fffd9b43530>, <numpy.ndarray at remote 0x7fffd9b43800>), func=<method at remote 0x7ffff2d1c0c8>) at ../Objects/abstract.c:2261
#48 slot_tp_call () at ../Objects/typeobject.c:6207 func=<vectorize(pyfunc=<built-in method lt of module object at remote 0x7ffff68ded18>, cache=False, signature=None, _ufunc=None, __doc__='lt(a, b) -- Same as a<b.', otypes=None, excluded=set(), _in_and_out_core_dims=None) at remote 0x7fffd9eccfd0>) at ../Objects/tupleobject.c:131
#50 _PyObject_FastCallKeywords () at ../Objects/abstract.c:2496 f=Frame 0x11fa4c8, for file drake/bindings/pydrake/test/symbolic_test.py, line 492, in _check_algebra (self=<TestSymbolicExpression(_testMethodName='test_array_algebra', _outcome=<_Outcome(expecting_failure=False, result=<TextTestResult(failfast=False, failures=[], errors=[], testsRun=1, skipped=[], expectedFailures=[], unexpectedSuccesses=[], shouldStop=False, buffer=False, tb_locals=False, _stdout_buffer=None, _stderr_buffer=None, _original_stdout=<_io.TextIOWrapper at remote 0x7ffff7f51630>, _original_stderr=<_io.TextIOWrapper at remote 0x7ffff7f51708>, _mirrorOutput=False, stream=<_WritelnDecorator(stream=<_io.TextIOWrapper at remote 0x7ffff7f51708>) at remote 0x7fffd9b3c4e0>, showAll=False, dots=True, descriptions=True, _testRunEntered=True, _moduleSetUpFailed=False, _previousTestClass=<type at remote 0x11efee8>) at remote 0x7fffd9b3cf98>, result_supports_subtests=True, success=True, skipped=[], expectedFailure=None, err...(truncated)) at ../Python/ceval.c:754
```

jwnimmer-tri, 2021-08-04:
Here's the stack for the (simpler) test_15549 first failure:

```
#0  drake::pydrake::<lambda(const drake::symbolic::Formula&)>::operator()(const drake::symbolic::Formula &) const (__closure=0x10f7378) at bindings/pydrake/symbolic_py.cc:530
#11 in slot_nb_bool () at ../Objects/typeobject.c:6049
#12 in PyObject_IsTrue (v=<pydrake.symbolic.Formula at remote 0x7fffd9b49330>) at ../Objects/object.c:1284
#13 PyObject_IsTrue (v=<optimized out>, v=<optimized out>) at ../Objects/object.c:1263
#14 in OBJECT_less_equal (args=<optimized out>, dimensions=<optimized out>, steps=<optimized out>, __NPY_UNUSED_TAGGEDfunc=<optimized out>) at numpy/core/src/umath/loops.c.src:2811
#15 in trivial_three_operand_loop (op=op@entry=0x7fffffff9c60, innerloop=0x7ffff5aabdb0 <OBJECT_less_equal>, innerloopdata=0x0) at numpy/core/src/umath/ufunc_object.c:1339
#16 in execute_legacy_ufunc_loop (arr_prep_args=0x0, arr_prep=0x7fffffff98c0, buffersize=8192, order=NPY_KEEPORDER, dtypes=0x7fffffff97c0, op=0x7fffffff9c60, trivial_loop_ok=<optimized out>,  ufunc=0xb19280) at numpy/core/src/umath/ufunc_object.c:1700
#17 PyUFunc_GenericFunction (ufunc=ufunc@entry=0xb19280, args=args@entry=(<numpy.ndarray at remote 0x7fffd9b425d0>, <pydrake.symbolic.Expression at remote 0x7fffd9b492f0>), kwds=kwds@entry=0x0,  op=op@entry=0x7fffffff9c60) at numpy/core/src/umath/ufunc_object.c:2749
#18 in ufunc_generic_call (ufunc=0xb19280, args=(<numpy.ndarray at remote 0x7fffd9b425d0>, <pydrake.symbolic.Expression at remote 0x7fffd9b492f0>), kwds=0x0) at numpy/core/src/umath/ufunc_object.c:4411
#21 in do_richcompare (op=1, w=<pydrake.symbolic.Expression at remote 0x7fffd9b492f0>, v=<numpy.ndarray at remote 0x7fffd9b425d0>) at ../Objects/object.c:708
#22 PyObject_RichCompare () at ../Objects/object.c:756
#23 in gentype_richcompare (self=<optimized out>, other=<pydrake.symbolic.Expression at remote 0x7fffd9b492f0>, cmp_op=1) at numpy/core/src/multiarray/scalartypes.c.src:1135
#24 in long_richcompare (self=<numpy.int64 at remote 0x7ffff2cf1b58>, other=<pydrake.symbolic.Expression at remote 0x7fffd9b492f0>, cmp_op=<optimized out>) at numpy/core/src/umath/scalarmath.c.src:1532
#25 in do_richcompare (op=1, w=<pydrake.symbolic.Expression at remote 0x7fffd9b492f0>, v=<numpy.int64 at remote 0x7ffff2cf1b58>) at ../Objects/object.c:708
#26 PyObject_RichCompare () at ../Objects/object.c:756
#27 in op_le () at ../Modules/_operator.c:108
#29 in do_call_core (kwdict=0x0, callargs=(<numpy.int64 at remote 0x7ffff2cf1b58>, <pydrake.symbolic.Expression at remote 0x7fffd9b492f0>),  func=<built-in method le of module object at remote 0x7ffff68ded18>) at ../Python/ceval.c:5116
#31 in PyEval_EvalFrameEx (throwflag=0,  f=Frame 0x11fa598, for file /usr/lib/python3/dist-packages/numpy/lib/function_base.py, line 2769, in _get_ufunc_and_otypes (self=<vectorize(pyfunc=<built-in method le of module object at remote 0x7ffff68ded18>, cache=False, signature=None, _ufunc=None, __doc__='le(a, b) -- Same as a<=b.', otypes=None, excluded=set(), _in_and_out_core_dims=None) at remote 0x7fffd9ecce80>, args=[<numpy.ndarray at remote 0x7fffd9b42620>, <numpy.ndarray at remote 0x7fffd9b42580>], inputs=[<numpy.int64 at remote 0x7ffff2cf1b58>, <pydrake.symbolic.Expression at remote 0x7fffd9b492f0>])) at ../Python/ceval.c:754
#32 _PyEval_EvalCodeWithName.lto_priv.1824 () at ../Python/ceval.c:4166
#35 in _PyEval_EvalFrameDefault () at ../Python/ceval.c:3351
#36 in PyEval_EvalFrameEx (throwflag=0,  f=Frame 0x11fa388, for file /usr/lib/python3/dist-packages/numpy/lib/function_base.py, line 2809, in _vectorize_call (self=<vectorize(pyfunc=<built-in method le of module object at remote 0x7ffff68ded18>, cache=False, signature=None, _ufunc=None, __doc__='le(a, b) -- Same as a<=b.', otypes=None, excluded=set(), _in_and_out_core_dims=None) at remote 0x7fffd9ecce80>, func=<built-in method le of module object at remote 0x7ffff68ded18>, args=(2, <pydrake.symbolic.Expression at remote 0x7fffd9b492f0>))) at ../Python/ceval.c:754
#37 _PyEval_EvalCodeWithName.lto_priv.1824 () at ../Python/ceval.c:4166
#40 in _PyEval_EvalFrameDefault () at ../Python/ceval.c:3351
#41 in PyEval_EvalFrameEx (throwflag=0,  f=Frame 0x11fa158, for file /usr/lib/python3/dist-packages/numpy/lib/function_base.py, line 2739, in __call__ (func=<built-in method le of module object at remote 0x7ffff68ded18>, vargs=(2, <pydrake.symbolic.Expression at remote 0x7fffd9b492f0>))) at ../Python/ceval.c:754
```

jwnimmer-tri, 2021-08-04:
It looks like rich compare of `int` vs `Expression` would work, but because of the `np.vectorize`, we're trying to rich compare `np.int64` and `Expression`, and since _both_ objects define rich comparison functions, the lhs (np.int64) wins, and we lose.

The same holds if we use "2.0" (float) in the test case, instead of "2" (int).

jwnimmer-tri, 2021-08-04:
Proof of concept that a fix is within reach:

```patch
diff --git a/bindings/pydrake/_math_extra.py b/bindings/pydrake/_math_extra.py
index acbc230c5..77dc826ae 100644
--- a/bindings/pydrake/_math_extra.py
+++ b/bindings/pydrake/_math_extra.py
@@ -24,6 +24,7 @@ As an example::
 # See `ExecuteExtraPythonCode` in `pydrake_pybind.h` for usage details and
 # rationale.
 
+import functools
 import operator
 
 import numpy as np
@@ -31,6 +32,17 @@ import numpy as np
 from pydrake.autodiffutils import AutoDiffXd as _AutoDiffXd
 from pydrake.symbolic import Expression as _Expression
 
+def _best_effort_rich_compare(a, b, *, oper):
+    try:
+        return oper(a, b)
+    except RuntimeError as e:
+        if "not call `__bool__` / `__nonzero__` on `Formula`" in str(e):
+            if isinstance(a, _Expression):
+                return oper(a, _Expression(b))
+            elif isinstance(b, _Expression,):
+                return oper(_Expression(a), b)
+        raise
+
 # As mentioned in top-level, add generic logical operators as ufuncs so that we
 # may do comparisons on arrays of any scalar type, without restriction on the
 # output type. These are added solely to work around #8315, where arrays of
@@ -38,12 +50,12 @@ from pydrake.symbolic import Expression as _Expression
 # type is not bool.
 # N.B. Defined in order listed in Python documentation:
 # https://docs.python.org/3.6/library/operator.html
-lt = np.vectorize(operator.lt)
-le = np.vectorize(operator.le)
-eq = np.vectorize(operator.eq)
-ne = np.vectorize(operator.ne)
-ge = np.vectorize(operator.ge)
-gt = np.vectorize(operator.gt)
+lt = np.vectorize(functools.partial(_best_effort_rich_compare, oper=operator.lt))
+le = np.vectorize(functools.partial(_best_effort_rich_compare, oper=operator.le))
+eq = np.vectorize(functools.partial(_best_effort_rich_compare, oper=operator.eq))
+ne = np.vectorize(functools.partial(_best_effort_rich_compare, oper=operator.ne))
+ge = np.vectorize(functools.partial(_best_effort_rich_compare, oper=operator.ge))
+gt = np.vectorize(functools.partial(_best_effort_rich_compare, oper=operator.gt))
 
 
 def _indented_repr(o):
```

When we're comparing Expression vs FooType or FooType vs Expression, we must always call into the Expression implementation of rich comparison, never FooType.  Therefore, if one argument is Expression but the other is not, we need to manually promote to Expression and then dispatch.

Here, I've done it with an exception fallback.  There are probably more correct or performant ways to do it, maybe @soonho-tri can work on the python spelling of that?

soonho-tri, 2021-08-04:
@jwnimmer-tri , thanks for the analysis! I'll take a stab tomorrow.

EricCousineau-TRI, 2021-08-05:
Nice on the exception fallback!

If it's a bit hard to figure out performant route for pydrake Expression, I'm totally up for reviewing non-performant case (#15558).

jwnimmer-tri, 2021-08-05:
I think the most difficult part here will actually be the unit testing.  The number of combinations of operand scalar types (float, AutoDiff, Expression, np.float64, np.int64), matrixness (single scalar vs vector of scalars), left/right operand order swaps, and which tranche of comparison operators we are calling (ScalarAlgebra vs VectorAlgebra) makes for a huge test set.  I briefly looked into enhancing the current tests to cover these, but I think it the approach to testing these needs a rethink from the ground up, to parametrically combine the options instead of listing them all out somewhat by hand.

As we've seen, it's basically impossible to reason about what np.vectorize is going to do with any given query (the in/equality comparison operators are particularly hairy, and we haven't even talked about them here yet), so we have no glass-box testing recourse, we just have to throw the _entire_ permutation of input arguments are the wall and see what sticks.

EricCousineau-TRI, 2021-08-06:
Aye; I don't think the combinatorics should be too hard to list out since it's just Python.

Do you want me to take a stab at it?

jwnimmer-tri, 2021-08-10:
> Aye; I don't think the combinatorics should be too hard to do via metaprogramming since it's Python.

Right, writing out the combinatoric looping with itertools or whatever shouldn't be too difficult.  What seemed like the biggest challenge, to me, would be annotating the expected results across all of the permutations.  _Shrug._  Maybe it is easy?

> Do you want me to take a stab at it?

I'm not sure who the "you" was directed to here, but just to be clear, this issue isn't assigned to me, and I'm not actively working on it.  My interest above was to get my head around the nature of the problem, and to help unwedge anyone else who is trying to work on it.

soonho-tri, 2021-08-10:
@EricCousineau-TRI , it'd be great if you can take this issue!

EricCousineau-TRI, 2021-08-20:
Can address, but would prefer to defer until late September, if possible.

RussTedrake, 2021-08-20:
I was really hoping to land it before September (I asked for it in the last release).  The psets that found the issue will happen in early September.  Might someone else take a crack at it?

RussTedrake, 2021-08-22:
@ToffeeAlbina-TRI -- can you please find someone to take this?

EricCousineau-TRI, 2021-08-24:
Per convo w/ Toffee, I'll go ahead and do it (simplest that way).

PR #15659 (2021-08-25): py math: Add patch for rich comparison operators; test combinatorics
Tests both symbolic types and AutoDiffXd
numpy_compare: Add soft_sub_test workaround
deprecation: Catch old-style deprecation elementwise != comparison

Resolves #15549

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15659)