RobotLocomotion/drake #15560: [symbolic] `Ref<VectorX<Expression>>` does not match `VectorX<Variable>`

RussTedrake, 2021-08-04:
I went to resolve #13735, which is *almost* trivial; and ironically would have been last year.  But my iris code served as a canary revealing a subtlety about casting from `Variable` to `Expression` through `Eigen::Ref`.  My almost PR is [here](https://github.com/RobotLocomotion/drake/compare/master...RussTedrake:fix_13735?expand=1).

This works fine:
```
  Vector2<Variable> vec{Variable("x"), Variable("y")};
  const Eigen::Ref<const VectorX<Expression>>& ref1 = vec.cast<Expression>();
```
but
```
  Vector2<Variable> vec{Variable("x"), Variable("y")};
  const Eigen::Ref<const VectorX<Expression>>& ref2 = vec;
```
fails with
```
common/test/symbolic_expression_test.cc: In member function 'virtual void drake::symbolic::{anonymous}::SymbolicExpressionTest_RefVectorVariable_Test::TestBody()':
common/test/symbolic_expression_test.cc:2670:55: error: invalid initialization of reference of type 'const Eigen::Ref<const Eigen::Matrix<drake::symbolic::Expression, -1, 1, 0, -1, 1> >&' from expression of type 'drake::Vector2<drake::symbolic::Variable> {aka Eigen::Matrix<drake::symbolic::Variable, 2, 1, 0, 2, 1>}'
   const Eigen::Ref<const VectorX<Expression>>& ref2 = vec;
```

Of course, I could use the `.cast<Expression>` everywhere.  But it seems if we want to use `Eigen::Ref` throughout, and think of it as a replacement for `VectorX<T>&`, then we should support this better?

jwnimmer-tri, 2021-08-04:
>This works fine:
> ```
> Vector2<Variable> vec{Variable("x"), Variable("y")};
> const Eigen::Ref<const VectorX<Expression>>& ref1 = vec.cast<Expression>();
> ```

Are you sure that doesn't create a dangling reference to destroyed memory?  I know that in some cases, Ref will take over temporary ownership, but it's not clear to me why this use would be one of them.  (Maybe it is; I haven't looked carefully).

> ... But it seems if we want to use Eigen::Ref throughout, ...

What is the referent of "throughout", and/or where is this idea coming from?  I don't seem to recall anyone on the team discussing such a convention anywhere recently.

jwnimmer-tri, 2021-08-04:
> What is the referent of "throughout", and/or where is this idea coming from?

Ah, well at least for #13735, the situation is about input arguments to functions.  From the the examples in the snippets above, I misunderstood that we were talking about local variables.

RussTedrake, 2021-08-04:
Yes, I might have skewed things by trying to make the trivial example.  I was passing `VectorX<Variable>` successfully into `SetPositions(const Context<Expression>&, const VectorX<Expression>&)`, but changing it to `SetPositions(const Context<Expression>&, const Eigen::Ref<const VectorX<Expression>>&)` caused the same code to fail.

jwnimmer-tri, 2021-08-04:
Previously, was making a copy (via a conversion constructor) during the call.  Now, that copy must be made explicit at the call site (via "cast") because turning it into Ref would requite two conversions in a row (Mat/Var => Mat/Exp => Ref/Mat/Exp), which is beyond what C++ does during function calls.

RussTedrake, 2021-08-05:
TIL.  Thanks!

closed 2021-08-05 by RussTedrake