RobotLocomotion/drake #15573: IBEX Solver

soonho-tri, 2021-08-05:
I'm in the process of adding [IBEX](https://github.com/ibex-team/ibex-lib) Solver. See the first PR, #15552 for more details and some examples.

TODO list:
 - [ ] Add Python bindings.
 - [ ] Add IBEX to https://drake.mit.edu/doxygen_cxx/group__solvers.html.
 - [ ] Add IBEX to `ChooseBestSolver` (or write a comment about why it can't be).
 - [ ] Repair handling of solver options.
 - [ ] Refine `IbexSolver::ProgramAttributesSatisfied`. This solver only works with costs and constraints that we can extract their symbolic forms.

RussTedrake, 2021-08-12:
I'll add 
- [ ] Resolve the TODO to support lorentz cone and rotated lorentz cone constraints
- [ ] Resolve the TODO to expose (at least) the tolerance and timeout parameters.

RussTedrake, 2021-08-12:
FYI -- current default solver parameters (which I found by setting a breakpoint in the debugger):
- `rel_eps_f = 0.001`
- `abs_eps_f = 1e-7`
- `eps_h = 1e-8`
- `rigor = false`
- `timeout = 30`
- `eps_x = 0`.
And there is some relevant documentation (which was annoying to find!) here: http://www.ibex-lib.org/doc/optim.html?highlight=eps#objective-precision-criteria

soonho-tri, 2021-08-12:
> Resolve the TODO to support lorentz cone and rotated lorentz cone constraints

=> https://github.com/RobotLocomotion/drake/pull/15607

soonho-tri, 2021-08-19:
> Repair handling of solver options.
=> https://github.com/RobotLocomotion/drake/pull/15612

jwnimmer-tri, 2021-11-18:
The remaining work here is to consider whether `IbexSolver` should participate in `ChooseBestSolver`.  If no, then document in its header file why not.  If yes, then add it to `Choose...` and therefore also the mathematical programming doxygen overview.

@hongkai-dai or @RussTedrake can you weigh in on whether it's a useful participant in `Choose...`?

RussTedrake, 2021-11-19:
I would say it definitely deserves to appear in https://drake.mit.edu/doxygen_cxx/group__solvers.html .  

There is not reason it should not appear in `ChooseBestSolver`, but I suspect it will be at the bottom of the list in all cases (so in practice never get called?).  The reason to use ibex over a different solver would be if you want global optimization (or feasibility) with the possibility of rigorous guarantees; but we don't have the ability to express that desire in `ChooseBestSolver`.  In terms of more common metrics (speed/robustness), it will likely almost always be worse.

jwnimmer-tri, 2023-01-28:
Closed as "not planned" via #18156.

closed 2023-01-28 by jwnimmer-tri