RobotLocomotion/drake #15596: New install of pydrake not working (possibly...fmt update)

vikr53, 2021-08-10:
_Issue:_ Recently (~12 hrs ago), fmt v8.0.1 was made the default version installed through Homebrew ([https://github.com/Homebrew/homebrew-core/commits/master/Formula/fmt.rb](https://github.com/Homebrew/homebrew-core/commits/master/Formula/fmt.rb)). 

So, when /opt/drake/share/drake/setup/install_prereqs is run, the latest versions, fmt@8.0.1 and spdlog@1.9.1 are being installed. However, it seems the provided shared object library, _module_py.cpython-39-darwin.so requires fmt@7.x (as is shown in the following error),

```
>>> import pydrake.common
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/drake/lib/python3.9/site-packages/pydrake/common/__init__.py", line 7, in <module>
    from ._module_py import *
ImportError: dlopen(/opt/drake/lib/python3.9/site-packages/pydrake/common/_module_py.cpython-39-darwin.so, 2): Library not loaded: /usr/local/opt/fmt/lib/libfmt.7.dylib
  Referenced from: /opt/drake/lib/python3.9/site-packages/pydrake/common/_module_py.cpython-39-darwin.so
  Reason: image not found
```

_What I tried:_ First, I attempted to install fmt@7.1.3 by the fetching the old fmt@7.1.3 brew formula commit and installing it locally,
```
curl https://raw.githubusercontent.com/Homebrew/homebrew-core/f281a5c6130f9e333a768a990d1a2a6f9af694d9/Formula/fmt.rb > fmt.rb
brew install ./fmt.rb
```
This worked. But, since spdlog@1.9.1 depends on fmt@8.x (and doesn't seem to be backwards compatible), spdlog breaks:
```
>>> import pydrake.all
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/drake/lib/python3.9/site-packages/pydrake/all.py", line 31, in <module>
    from .autodiffutils import *
ImportError: dlopen(/opt/drake/lib/python3.9/site-packages/pydrake/autodiffutils.cpython-39-darwin.so, 2): Library not loaded: /usr/local/opt/fmt/lib/libfmt.8.dylib
  Referenced from: /usr/local/opt/spdlog/lib/libspdlog.1.dylib
  Reason: image not found
```
Similarly, I tried to fetch the old spdlog@1.7.0 brew formula and install it locally but was faced with the following error,
`Error: spdlog: Calling `cellar` in a bottle block is disabled! Use `brew style --fix` on the formula to update the style or use `sha256` with a `cellar:` argument instead.`

This is likely because the formula is over a year old and syntax has changed.

_Questions/Idea:_ Where is _module_py.cpython-39-darwin.so being built? Is it possible to change its dependence on fmt@7.x to fmt@8.x? (I'm not sure of the pervasiveness of fmt@7.x)

jwnimmer-tri, 2021-08-10:
I think you've correctly identified the problem.

Homebrew has changed their library ABI with the fmt update, so all existing binary programs that use fmt need to be recompiled from source to match the new ABI.

Since I see `/opt/drake` in your path, I assume you're using one of our https://drake.mit.edu/from_binary.html releases, instead of building from source.

Because homebrew updated early this morning (but after our nightly builds), our nightly release pipeline hasn't had a chance to rebuild our release tarball.  The nightly builds that run tonight (i.e., stamped as 20210811 or newer) should be compatible with the new fmt (and will be incompatible with the old fmt).  The new binary will appear at https://drake-packages.csail.mit.edu/drake/nightly/drake-20210811-mac.tar.gz once the build machines do their thing.

A workaround for today would be to build https://drake.mit.edu/from_source.html yourself.

jwnimmer-tri, 2021-08-10:
I realize now that for doing a "build from source" for pydrake users, the instructions linked below would more directly relevant than my other link above:

https://drake.mit.edu/python_bindings.html#building-the-python-bindings

jwnimmer-tri, 2021-08-11:
Our continuous integration shows that this problem has been resolved, as of today's binaries.

I've also tagged those binaries as an official release (https://drake.mit.edu/release_notes/v0.33.0.html).

Hopefully this is resolved now, but please let us know if there are still any problems.

closed 2021-08-11 by jwnimmer-tri