RobotLocomotion/drake #15604: AutoDiff-related functions violate Drake styleguide

sherm1, 2021-08-12:
It has bothered me forever that we've kept the original, pre-styleguide names of several commonly-used AutoDiff-manipulating functions. For one thing it sets a poor example of the importance of adhering to a styleguide!

I propose to deprecate the old names and replace them with new, styleguide-compatible names as follows:

old name | new name | notes
-------- | -------- | --------
autoDiffToGradientMatrix | ExtractGradientMatrixFromAutoDiff
autoDiffToValueMatrix | ExtractValueMatrixFromAutoDiff
initializeAutoDiff | InitializeAutoDiff | fix output parameter
initializeAutoDiffGivenGradientMatrix | InitializeAutoDiffGivenValueAndGradientMatrix | fix output parameter
initializeAutoDiffTuple | InitializeAutoDiffTuple
resizeDerivativesToMatchScalar | ResizeDerivativesToMatchScalar | fix output parameter
gradientMatrixToAutoDiff | SetAutoDiffDerivativesFromGradientMatrix | fix output parameter
totalSizeAtCompileTime | total_size_at_compile_time | make internal?
totalSizeAtRunTime | total_size_at_run_time | make internal?

Did I miss any?

WDYT? Counterproposals or reasons why we shouldn't do this?

rpoyner-tri, 2021-08-12:
Being late to the party, I had assumed those names had crept over from
Eigen somehow. I don't see a big downside to offering both names, and/or
deprecating the GSG violators. Any historians able to speak for the
existing names?

-- 
Rick Poyner
Manager, Dynamics and Simulation
Robotics Program
Toyota Research Institute

jwnimmer-tri, 2021-08-12:
Those functions predate our adoption of GSG, e.g., https://github.com/RobotLocomotion/drake/commit/6bdf3a3569de11bd72388d962afc58ccc430cc9b was made into a header-file function as of 2015.

sherm1, 2021-08-12:
@hongkai-dai, @RussTedrake I think you use these frequently -- any thoughts on the name changes?

RussTedrake, 2021-08-12:
As long as you change my call sites in drake, I think it's great to update to GSG.

jwnimmer-tri, 2021-08-17:
> resizeDerivativesToMatchScalar | ResizeDerivativesToMatchScalar | fix output parameter

As with gradientMatrixToAutoDiff, this function is _also_ unused within Drake and Anzu.  Why not deprecate for removal (vs rename) likewise?

sherm1, 2021-08-17:
> resizeDerivativesToMatchScalar -> trash

Great, will do. 

Q: what is the best strategy for pain reduction in Anzu? Should I merge the new names first without deprecating the old ones so that we can convert Anzu first, then follow up with non-disruptive deprecations?

jwnimmer-tri, 2021-08-17:
> Q: what is the best strategy for pain reduction in Anzu? Should I merge the new names first without deprecating the old ones so that we can convert Anzu first, then follow up with non-disruptive deprecations?

That's a fine plan, but it's really just up to you.  If you're making the Anzu PR(s) yourself, then you can choose the way that's easiest for you.  If you're going to try to recruit a buddy to do the Anzu PRs alongside you, then you can choose between the two of you what's easiest.

sherm1, 2021-08-22:
After looking at the call sites in #15615, Jeremy wrote:
> Do we need the "FromAutoDiff" suffix on these function names? They are very widely used, and the amount of bad line wrapping induced by a 33-letter long function name is fairly acute. Maybe ExtractValueMatrix and ExtractGradientMatrix are clear enough on their own? Similarly, InitializeAutoDiffFromValueMatrix weighs in at 33 letters and InitializeAutoDiffFromValueAndGradientMatrix is much longer. Are they sufficiently the same operation that we could just say InitializeAutoDiff for both, and allow overloading to do its thing? They seem to be the same to me, with the caveat of the gradients being an identity matrix when not supplied. I'm honestly surprised that nobody who pinged the issue objected to the names. Maybe if they looked over this code, they would object?

I am neutral about short-vs-long names here but would like comment from users. I picked the long, more-explicit names because I think AutoDiff is a source of confusion for Drake users (and code readers) so being explicit would outweigh the inconvenience. But I'm not sure.

Thoughts please @RussTedrake @hongkai-dai @edrumwri @rpoyner-tri or other AutoDiff consumers?

RussTedrake, 2021-08-22:
I favor the shorter names, too.  
- `ExtractGradientMatrix` seems clear enough.
- `ExtractValueMatrix` would be fine, but keep in mind that we already have `ExtractDoubleOrThrow` which by the way tends to confuse me (#12650) and for AutoDiff the [docs](https://drake.mit.edu/doxygen_cxx/namespacedrake.html#aa9721f3a5897b05ef2168f7a6d886dc8) actually say "never throws".   So I think your proposed method would be identical?
- `InitializeAutoDiff` could have a single argument version and a two-argument version (which takes the gradient); it seems like that would still be perfectly clear.
- `InitializeAutoDiffTuple` strikes me as very reasonable.

hongkai-dai, 2021-08-22:
I think `InitializeAutoDiffFromValueMatrix` and `InitializeAutoDiffFromValueAndGradientMatrix` is way too long. I agree with Russ that `InitializeAutoDiff` is clear enough, with a single argument and a 2-argument version.

If we want to make it even shorter, I think we can remove `Matrix` from the name `ExtractValueMatrix` and `ExtractGradientMatrix`.

sherm1, 2021-08-24:
OK, here is an updated proposal with short names:

old name | new name | notes
-------- | -------- | --------
autoDiffToGradientMatrix | ExtractGradient
autoDiffToValueMatrix | ExtractValue
initializeAutoDiffGivenGradientMatrix | InitializeAutoDiff(value, gradient)| fix output parameter
initializeAutoDiff | InitializeAutoDiff(value[, num_derivatives]) | fix output parameter
initializeAutoDiffTuple | InitializeAutoDiffTuple
resizeDerivativesToMatchScalar | --- | unused; deprecate
gradientMatrixToAutoDiff | --- | unused; deprecate

A few notes:
- I'm not planning to change Discard[Zero]Gradient() or ExtractDoubleOrThrow()
- ExtractDoubleOrThrow() is limited to AutoDiffXd, the above utilities are more general so can (in theory) be used for second derivatives, fixed-size derivative, etc.

Any further thoughts?

RussTedrake, 2021-08-25:
I just took on more look to decide if I like `ExtractDouble` instead of `ExtractValue`.  At very least I would hope that the relationship between `ExtractValue` and `ExtractDoubleOrThrow` is clearly documented (perhaps one should even call the other).  The choice "value" comes from Eigen::AutoDiffScalar, but it's not a perfect match with the Value concept in Drake.  Conversely, the choice "gradient" differs from Eigen::AutoDiffScalar (which uses "derivatives").

I slightly favor `ExtractDouble` and `ExtractGradient`, since I don't think people are particularly familiar with AutoDiffScalar and that match doesn't contribute much.  But am happy with whichever you choose.

sherm1, 2021-08-31:
Looks like I should deprecated the `Eigen::Transform` specialization for `DiscardGradient()` and `DiscardZeroGradient()` (keeping the versions that apply to ordinary `Vector<AutoDiff>` and `Matrix<AutoDiff>` objects).

Does anyone think we should keep the Transform versions around?

RussTedrake, 2021-08-31:
Not me.

hongkai-dai, 2021-08-31:
No I don't think we should keep `Eigen::Transform` around as we have drake::RigidTransform.

sherm1, 2021-09-01:
OK, I will deprecate the specializations for `Discard[Zero]Gradient(Transform<AutoDiffScalar>)`.

sherm1, 2021-09-29:
Styleguide compatible names for autodiff utilities are now available in Drake master:
- InitializeAutoDiff()
- ExtractValue()
- ExtractGradient()
- 
(See table in issue description for mapping from old names.)

All call sites in Drake and most in Anzu have been updated. Please use the new names for new code now. The old ones will be deprecated shortly.