RobotLocomotion/drake #15606: Build Drake v0.33.0 for Colab Python 3.7

jwnimmer-tri, 2021-08-12:
We'd like create a binary release tarball for use with Google Colab, which uses python 3.7.

We should update https://github.com/RobotLocomotion/drake-python3.7 to reflect the v0.33.0 baseline and then create a new release tarball from it.

jwnimmer-tri, 2021-08-12:
FYI related https://github.com/RobotLocomotion/drake/pull/15516#issuecomment-896972776

BetsyMcPhail, 2021-08-13:
https://github.com/RobotLocomotion/drake-python3.7/tree/0.33.0 has been rebased on Drake's v0.33.0 release. (Note the new branch, there were quite a few conflicts, so let's leave main as is just in case).

BetsyMcPhail, 2021-08-16:
tarball is available here: https://drake-packages.csail.mit.edu/drake/rt/drake-python3.7-0.33.0-1.tar.gz

BetsyMcPhail, 2021-08-16:
@jwnimmer-tri I'm happy to do some testing if given some direction.

BetsyMcPhail, 2021-08-16:
As a note to myself, the python3.7 instance on AWS (currently stopped) was used to produce the tarball

jwnimmer-tri, 2021-08-17:
I don't think I need to test anything, but maybe @RussTedrake wants to?

RussTedrake, 2021-09-19:
Related to #1183.  
I've confirmed that running 
```
!pip install drake
from pydrake.geometry import Meshcat
```
works for me on colab.  Hoorah!

I will do more testing soon, but since I am using deepnote as my primary workflow this term, I will reduce the priority on this.

jwnimmer-tri, 2021-09-20:
The purpose of this issue was to build (already done) and test (not yet done) the tarball release of v0.33.0 vs the colab runtime, using the colab-specific branch of Drake.  This issue is not related to pip at all.

If testing that older tarball release is not interesting anymore (and it sounds like it's not), then we can just close the issue without additional testing, and revisit for v0.35.0 or some future revision, if necessary.

closed 2021-09-20 by RussTedrake