RobotLocomotion/drake #15613: iiwa: Should add in reasonable acceleration limits, and use actual velocity and effort limits

EricCousineau-TRI, 2021-08-13:
At present, IIWA14 URDFs (at least) do not appear to have actual hardware velocity and torque limits; instead, they're derived from 
https://github.com/IFL-CAMP/iiwa_stack/tree/2fa5bd5 (understandably)
Additionally, they do not have acceleration limits, which are useful for planning and control.

We have TODOs in Anzu to hoist these to Drake URDFs; this issue should be resolved by:
- Incorporating velocity and effort limits from KUKA
- Providing a reasonable estimate of acceleration limits (the actual achievable limits)
- Ideally, providing guidance on how to sync them against our different variants (xacro and hand-edited)

\cc @thduynguyen @calderpg-tri 

@azeey I'd like us to queue this up starting October.

EricCousineau-TRI, 2021-11-01:
Marco saw some difference in precision between prior version of ROS1 xacro and Noetic xacro in #16003.

marcoag, 2021-11-03:
After going all the way back the `kinetic` branch of `xacro` the generated `urdf` still differs from the original. Here is the diff: 
https://gist.github.com/marcoag/fb5216f34ead56e34c44fb462f904d61 .Trying to use previous versions throws problems of very old and not supported dependencies not found, making it very hard to run.

It seems that the [file was committed in 2016](https://github.com/RobotLocomotion/drake/commit/4e561f01e0b2cf0937e5a56796b43ffdb5e4402c) so it must have been generated with a `kinetic` version or before.

From [the files blame description](https://github.com/RobotLocomotion/drake/blame/master/manipulation/models/iiwa_description/urdf/iiwa14_polytope_collision.urdf) it seems that the files have been manually modified several times, making it impossible to have the generated files exactly match the current committed version. 

Also given that [this line](https://github.com/RobotLocomotion/drake/blob/7445fa973aa8571800cf4a6ccb547f47bb797c6e/manipulation/models/iiwa_description/urdf/iiwa14.xacro#L447) is supposed to generate something like: `<joint name="iiwa_tool0_joint" type="fixed">` and the committed file has `<joint name="tool0_joint" type="fixed">` tells me the file has not even been regenerated any time after being committed, even when the `xacro` file was modified.

The rest of the differences are reordering and precision (values are more precise now).

I guess this leaves us the following options:

1. **Remove the generated files**. As generated files (sames as a binary) keeping them in the repo will only take space and confuse people if they are not updated. In this case, the [README.md ](https://github.com/RobotLocomotion/drake/blob/master/manipulation/models/iiwa_description/README.md) should explain how to generate the `urdfs`.
2. **Manually modify the generated files** and update velocity and effort.
3. **Regenerate and commit** the new version of the generated files.

Suggestions?

EricCousineau-TRI, 2021-11-04:
My suggestion is to first do (3) as a separate PR, but ideally using `melodic` (since that is what we're using in Anzu). We should then commit that artifact to show the changes isolated to just the toolchain.

If we see something in the files that are hand-edited, we should try to hoist that back into the xacro, or just fork the model.

At some point, we should teach Drake to use `xacro` proper (assuming it's not too intertwined in ROS dependencies). We have this in Anzu.

calderpg-tri, 2021-11-04:
We have a number of different iiwa models, all nominally based on the same xacro model, but with significantly different collision geometries. I suspect that (2) is the only way to update them all, unless we refactor them all to share some base xacro?

EricCousineau-TRI, 2021-11-04:
Aye. At minimum, we should perhaps identify where divergence happens. Hopefully, (3) as a first check can tell us what's close and what's far?

jwnimmer-tri, 2021-11-04:
If memory serves, the comments in the folder & files about "Use xacro for this" are grossly misleading, and you are going to find in (3) that there are mountains of divergence.

Paying off that debt and fixing the comments / xacro is a fine project, but seems like a huge, time-sink distraction on the way to getting correctly limits typed into the model files.  Adding the limits by hand into all of the files in git seems super easy to implement and code review, and could merge quick, and thus unblock the progress we need here.  We can service the debt immediately after, if you think it's important.

EricCousineau-TRI, 2021-11-04:
Aye, in that case, (2) sounds better.

marcoag, 2021-11-09:
Just went ahead and added (2) to https://github.com/RobotLocomotion/drake/pull/16003. Are we still looking to teach Drake to use xacro proper? if so I assume that should be done on a different PR, is it?

marcoag, 2022-01-12:
@EricCousineau-TRI I was looking to move forward and send a PR on task 2 of this issue but wanted to confirm: Are the values from https://odr.chalmers.se/bitstream/20.500.12380/256658/1/256658.pdf acceptable for now?

EricCousineau-TRI, 2022-01-12:
Yup!

EricCousineau-TRI, 2022-03-07:
Going to carve out "variant" stuff to separate issue.

EricCousineau-TRI, 2022-03-07:
Closing this in lieu of #8946

closed 2022-03-07 by EricCousineau-TRI