RobotLocomotion/drake #15619: [doc] DRAKE_DEPRECATED can lead usage leads to badly formatted doxygen

SeanCurtis-TRI, 2021-08-17:
## Problem

To conform to style guide constrains (80 columns) it is common to spell a long deprecation message over several implicitly concatenated strings. For example, [`SceneGraph::get_pose_bundle_output_port()`](https://github.com/RobotLocomotion/drake/blob/5c7f033bdd492bcf239edfe2500fd5cbc342b666/geometry/scene_graph.h#L337-L344):

```c++
  DRAKE_DEPRECATED("2021-12-01",
                   "PoseBundle is no longer in use. Visualizers typically "
                   "connect to SceneGraph's QueryObject port.")
  const systems::OutputPort<T>& get_pose_bundle_output_port() const
```

It appears doxygen doesn't handle this practice with much elegance, producing the following result:

[image]

Note that it includes each string, with its enclosing double quotes in sequence.

(As a side note, the fact that it *also* includes quotes around the date is odd -- as if the date weren't a literal thing.)

SeanCurtis-TRI, 2021-08-17:
@jwnimmer-tri I've taken the liberty to assign this to you, at least in the hopes of it being dispositioned properly. My best effort at assigning labels may also require correction.

jwnimmer-tri, 2021-08-17:
I noticed this as well, a couple months ago.  My disposition at the time was "not worth spending any time on".  I suspect we best we could do is either:

(1) Post-process Doxygen's HTML output with a regex, to strip out the extra quotes.

(2) Change our line wrapping custom for these to remove the hard linebreaks (and still somehow placate the linter).  The outermost quotes would remain, but the interior quote pairs that interrupt the flow would be gone.

SeanCurtis-TRI, 2021-08-17:
I'm content with "not worth spending time on" as a resolution today as well. It's a minor symptom at worst and it'll disappear and reappear erratically in the documentation (following the vagaries of deprecation). The actual deprecation message isn't rendered any less clear. I just felt bad that I hadn't noticed in the past and didn't have any expectations to see that odd formatting.

closed 2021-09-28 by sherm1

jwnimmer-tri, 2021-09-29:
https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1_scene_graph.html#a637679130ce796cc4dfaa97ceb0e7559

[image]

sherm1, 2021-09-29:
Beautiful!

PR #15832 (2021-09-28): [doc] Remove extra quotation marks from Doxygen deprecation notes
Closes #15619.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15832)