RobotLocomotion/drake #15627: MBP segfaults on computing reaction forces when all joints are welds

edrumwri, 2021-08-19:
Following code triggers it:

```
#include <memory>

#include "drake/common/drake_assert.h"
#include "drake/multibody/plant/multibody_plant.h"

#include <gtest/gtest.h>

using drake::geometry::Box;
using drake::math::RigidTransform;
using drake::multibody::Body;
using drake::multibody::MultibodyPlant;
using drake::multibody::SpatialForce;
using drake::multibody::SpatialInertia;
using drake::multibody::UnitInertia;
using drake::multibody::WeldJoint;
using drake::systems::Context;
using drake::Vector3;

class WeldedBoxesTest : public ::testing::Test {
 protected:
  void SetUp() final {
    plant_ = std::make_unique<MultibodyPlant<double>>(0.0 /* arbitrary continuous time */);
    AddBoxes();
    plant_->Finalize();
    plant_context_ = plant_->CreateDefaultContext();
  }

  void AddBoxes() {
    const double kCubeSize = 1.0;
    const double kBoxMass = 2.0;
    const Vector3<double> p_BoBcm_B = Vector3<double>::Zero();
    const UnitInertia<double> G_BBcm = UnitInertia<double>::SolidBox(kCubeSize, kCubeSize, kCubeSize);
    const SpatialInertia<double> M_BBo_B = SpatialInertia<double>::MakeFromCentralInertia(kBoxMass, p_BoBcm_B, G_BBcm);

    // Create two rigid bodies.
    boxA_ = &plant_->AddRigidBody("boxA", M_BBo_B);
    boxB_ = &plant_->AddRigidBody("boxB", M_BBo_B);

    // Desired transformation for the boxes in the world.
    const RigidTransform<double> X_WA(Vector3<double>::Zero());
    const RigidTransform<double> X_WB(Vector3<double>(kCubeSize, 0, 0));
    const RigidTransform<double> X_AB = X_WA.inverse() * X_WB;

    // Pin boxA to the world and boxB to boxA with weld joints.
    weld1_ = &plant_->WeldFrames(plant_->world_body().body_frame(), boxA_->body_frame(), X_WA);
    weld2_ = &plant_->WeldFrames(boxA_->body_frame(), boxB_->body_frame(), X_AB);
  }

  void VerifyBodyReactionForces() {
    const auto& reaction_forces_F_CJc_Jc =
        plant_->get_reaction_forces_output_port().Eval<std::vector<SpatialForce<double>>>(*plant_context_);
    (void)reaction_forces_F_CJc_Jc;
  }

  const Body<double>* boxA_{nullptr};
  const Body<double>* boxB_{nullptr};
  std::unique_ptr<MultibodyPlant<double>> plant_{nullptr};
  const WeldJoint<double>* weld1_{nullptr};
  const WeldJoint<double>* weld2_{nullptr};
  std::unique_ptr<Context<double>> plant_context_;
};

TEST_F(WeldedBoxesTest, Go) { VerifyBodyReactionForces(); }
```

Backtrace:

[... 64 lines cut ...]
    at bazel-out/k8-dbg/bin/systems/framework/_virtual_includes/cache_entry/drake/systems/framework/cache_entry.h:175
#37 0x000055555611e9a7 in drake::systems::CacheEntry::Eval<drake::multibody::internal::AccelerationKinematicsCache<double> > (this=0x555557500030, context=...)
    at bazel-out/k8-dbg/bin/systems/framework/_virtual_includes/cache_entry/drake/systems/framework/cache_entry.h:159
#38 0x0000555556111f77 in drake::multibody::internal::MultibodyTreeSystem<double>::EvalForwardDynamics (this=0x5555574f1540, context=...)
    at bazel-out/k8-dbg/bin/multibody/tree/_virtual_includes/multibody_tree_core/drake/multibody/tree/multibody_tree_system.h:120
#39 0x000055555564aa7c in drake::multibody::MultibodyPlant<double>::CalcReactionForces (this=0x5555574f1540, context=..., F_CJc_Jc_array=0x55555750b480) at multibody/plant/multibody_plant.cc:3215
#40 0x00005555556f33e1 in drake::systems::ValueProducer::make_calc_mode_1<drake::systems::LeafSystem<double>, drake::multibody::MultibodyPlant<double>, drake::systems::Context<double>, std::vector<drake::multibody::SpatialForce<double>, std::allocator<drake::multibody::SpatialForce<double> > > >(drake::systems::LeafSystem<double> const*, void (drake::multibody::MultibodyPlant<double>::*)(drake::systems::Context<double> const&, std::vector<drake::multibody::SpatialForce<double>, std::allocator<drake::multibody::SpatialForce<double> > >*) const)::{lambda(drake::systems::ContextBase const&, drake::AbstractValue*)#1}::operator()(drake::systems::ContextBase const&, drake::AbstractValue*) const (this=0x555557502dc0, context=..., result=0x55555750b470)
    at bazel-out/k8-dbg/bin/systems/framework/_virtual_includes/value_producer/drake/systems/framework/value_producer.h:450
#41 0x000055555573f50f in std::_Function_handler<void (drake::systems::ContextBase const&, drake::AbstractValue*), drake::systems::ValueProducer::make_calc_mode_1<drake::systems::LeafSystem<double>, drake::multibody::MultibodyPlant<double>, drake::systems::Context<double>, std::vector<drake::multibody::SpatialForce<double>, std::allocator<drake::multibody::SpatialForce<double> > > >(drake::systems::LeafSystem<double> const*, void (drake::multibody::MultibodyPlant<double>::*)(drake::systems::Context<double> const&, std::vector<drake::multibody::SpatialForce<double>, std::allocator<drake::multibody::SpatialForce<double> > >*) const)::{lambda(drake::systems::ContextBase const&, drake::AbstractValue*)#1}>::_M_invoke(std::_Any_data const&, drake::systems::ContextBase const&, drake::AbstractValue*&&) (__functor=..., __args#0=..., 
    __args#1=@0x7fffffffd9f8: 0x55555750b470) at /usr/include/c++/9/bits/std_function.h:300
#42 0x000055555659833b in std::function<void (drake::systems::ContextBase const&, drake::AbstractValue*)>::operator()(drake::systems::ContextBase const&, drake::AbstractValue*) const (this=0x555557503530, 
    __args#0=..., __args#1=0x55555750b470) at /usr/include/c++/9/bits/std_function.h:688
#43 0x0000555556597f94 in drake::systems::ValueProducer::Calc (this=0x555557503510, context=..., output=0x55555750b470) at systems/framework/value_producer.cc:53
#44 0x000055555659620e in drake::systems::CacheEntry::Calc (this=0x5555575034e0, context=..., value=0x55555750b470) at systems/framework/cache_entry.cc:64
#45 0x000055555563626a in drake::systems::CacheEntry::UpdateValue (this=0x5555575034e0, context=...)
    at bazel-out/k8-dbg/bin/systems/framework/_virtual_includes/cache_entry/drake/systems/framework/cache_entry.h:329
#46 0x00005555556360fe in drake::systems::CacheEntry::EvalAbstract (this=0x5555575034e0, context=...)
    at bazel-out/k8-dbg/bin/systems/framework/_virtual_includes/cache_entry/drake/systems/framework/cache_entry.h:175
#47 0x0000555556511840 in drake::systems::LeafOutputPort<double>::DoEval (this=0x555557503590, context=...)
    at bazel-out/k8-dbg/bin/systems/framework/_virtual_includes/leaf_output_port/drake/systems/framework/leaf_output_port.h:105
#48 0x000055555560b1b8 in drake::systems::OutputPort<double>::Eval<std::vector<drake::multibody::SpatialForce<double>, std::allocator<drake::multibody::SpatialForce<double> > >, void> (this=0x555557503590, 
    context=...) at bazel-out/k8-dbg/bin/systems/framework/_virtual_includes/output_port/drake/systems/framework/output_port.h:119
#49 0x0000555555609cd7 in WeldedBoxesTest::VerifyBodyReactionForces (this=0x5555574ecd80) at multibody/plant/test/simulated_force_torque_measurement_test.cc:58
#50 0x00005555556063c6 in WeldedBoxesTest_Go_Test::TestBody (this=0x5555574ecd80) at multibody/plant/test/simulated_force_torque_measurement_test.cc:73
```

sherm1, 2021-08-19:
@amcastro-tri would you take a look at this please? It appears that the ArticulatedInertia calculation barfs when there is no articulation!

amcastro-tri, 2021-08-20:
Thanks for reporting. This is an interesting use case, essentially you are trying to compute the reaction forces in the "static condition" imposed by these welds. Very nice.
Thank you for providing the unit test, I will definitely use it in the fix PR.

edrumwri, 2021-08-24:
Hey @amcastro-tri can you provide an ETA on fixing this? We're depending on this feature (computing reaction forces for static bodies) to test a calculation. I don't feel confident providing a fix for this.

amcastro-tri, 2021-08-25:
I do not have an ETA since I don't really know what the underlying problem is. I might be able to say more today evening. For sure I will work on this on Thursday morning.

amcastro-tri, 2021-08-30:
Sorry for the delay all, working on diagnosing this today.

amcastro-tri, 2021-08-30:
Ok @edrumwri, at least there is an easy fix: Switch to discrete mode. 
In discrete mode your test works and the numerical values are correct.

Definitely there is a bug in the continuous case. I'll debug that next.

amcastro-tri, 2021-08-30:
I was able to find the problem. I will push a fix shortly.

amcastro-tri, 2021-08-30:
@edrumwri, fix in #15686.

PR #15686 (2021-08-31): Fixes BodyNode::GetJacobianFromArray() to return an empty map when nv=0.
Resolves #15627.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15686)
closed 2021-08-31 by sherm1