RobotLocomotion/drake #15631: pip install: Consider renaming wheel from pydrake to drake

BetsyMcPhail, 2021-08-19:
`pydrake` has already been taken on pypi.

[Slack Conversation](https://drakedevelopers.slack.com/archives/C028F9T6MSM/p1629301305056000)

[Another Slack Conversation](https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1629315725025600)

Working towards #1183

BetsyMcPhail, 2021-08-19:
As discussed at meeting on 8/19, the first step is for @mwoehlke-kitware to ensure we have `drake` name on pypi. Perhaps using a placeholder wheel with just a README.

mwoehlke-kitware, 2021-09-08:
So... the wheel *is* "drake", and permits `import drake`. Is there any reason for this to remain open?

jwnimmer-tri, 2021-09-08:
The invariant we need to achieve is that `pip install $foo` provides `import $foo` (and nothing else) -- that is, the name of the wheel and the name of the module agree, and we do not have any _additional_ modules hiding out next to it.

How y'all want to break out tasks into GitHub issues to meet that invariant is up to y'all.

If our premise is that `$foo == "drake"`, then the overall task is something like _Rename the 'pydrake' module to 'drake'_, but that will also have a lot of sub-tasks.

jwnimmer-tri, 2021-11-11:
We should probably do this in concert with #11802, e.g., have `drake.foo` be fewer (larger) modules out of the gate, and use the top-level module rename as a good lever to distinguish the two different sub-module layouts.