RobotLocomotion/drake #15632: pip install: License notices

BetsyMcPhail, 2021-08-19:
From @jwnimmer-tri on [slack](https://drakedevelopers.slack.com/archives/C028F9T6MSM/p1629220550048100):

Another (non-critical-path) task for this project is license notices.  For externals that were previously shared libraries from Ubuntu, which we are now linking statically, if their license requires a notice file (and most common licenses would do so), we need to collect those upstream notices into the wheel, similar to how we already do for the things that Drake itself builds from source:

```
jwnimmer@call-cps:~/Downloads$ find v0.32.0/drake/share/doc -type f
...
v0.32.0/drake/share/doc/ignition_utils/LICENSE
v0.32.0/drake/share/doc/ignition_utils/COPYING
v0.32.0/drake/share/doc/cds/LICENSE
v0.32.0/drake/share/doc/yaml-cpp/LICENSE
v0.32.0/drake/share/doc/ghc_filesystem/LICENSE
v0.32.0/drake/share/doc/meshcat_python/LICENSE.txt
v0.32.0/drake/share/doc/ibex/LICENSE
v0.32.0/drake/share/doc/ibex/COPYING.LESSER
v0.32.0/drake/share/doc/bot2-core-lcmtypes/LICENSE.txt
v0.32.0/drake/share/doc/bender/LICENSE
v0.32.0/drake/share/doc/bender/NOTICE
...
```

Working towards #1183

BetsyMcPhail, 2021-09-14:
Using https://github.com/RobotLocomotion/pip-drake/tree/master/image/dependencies/projects as my starting point of licenses to gather

BetsyMcPhail, 2021-09-14:
My notes so far:

bzip2 - https://www.sourceware.org/bzip2/ (BSD)
clp
double-conversion - https://github.com/google/double-conversion/blob/master/LICENSE (BSD)
eigen - https://gitlab.com/libeigen/eigen/-/blob/master/COPYING.README
gflags - https://github.com/gflags/gflags/blob/master/COPYING.txt (BSD)
ipopt
lapack (blas)- https://github.com/Reference-LAPACK/lapack/blob/master/LICENSE (BSD)
libjpeg-turbo https://github.com/libjpeg-turbo/libjpeg-turbo/blob/main/LICENSE.md
libtiff - https://libtiff.gitlab.io/libtiff/misc.html (BSD)
lz4 - https://github.com/lz4/lz4/blob/dev/LICENSE
msgpack - https://github.com/msgpack/msgpack-c/blob/master/LICENSE_1_0.txt
nlopt - https://github.com/stevengj/nlopt/blob/master/COPYING
patchelf - https://github.com/NixOS/patchelf/blob/master/COPYING (GPL3)
png - https://sourceforge.net/p/libpng/code/ci/master/tree/LICENSE
suitesparse - https://github.com/DrTimothyAldenDavis/SuiteSparse/blob/master/LICENSE.txt
tinyxml2 - https://github.com/leethomason/tinyxml2/blob/master/LICENSE.txt (Zlib)
xz - https://git.tukaani.org/?p=xz.git;a=blob;f=COPYING;h=20e60d5b2427334e1fec6701e7c5ad0da0bc8a5d;hb=HEAD
yaml-cpp - https://github.com/jbeder/yaml-cpp/blob/master/LICENSE (MIT)
zlib - https://github.com/madler/zlib (zlib)

jwnimmer-tri, 2021-09-14:
For reference, #4045 had a summary that was accurate in the past.  It's not totally up to date anymore, but it might be a helpful reference.

The overall objective of this issue is to have the `LICENSE` (and/or related) files shipped as part of the wheel, alongside the other `pydrake/doc/foo/LICENSE` files that are already there.  We also want it to be done in a way that is easy enough to maintain as dependencies ebb and flow over time.  We don't want to do a one-shot license compendium by hand that we'd need to revisit as a whole again every time.

One way to do this might be to have https://github.com/RobotLocomotion/pip-drake/blob/master/image/build-dependencies.sh copy the LICENSE files into `/opt/drake-dependencies/licenses/...` one by one as it builds each one, and then the whl builder just has to propagate them in bulk, without knowing anything about what dependencies were used.

BetsyMcPhail, 2021-09-23:
Of note here, `libxcrypt` is LGPLv2.1. Similar to #15757 for nlopt, we need to use the shared library or not use this library (Is this possible?)

jwnimmer-tri, 2021-09-23:
> Of note here, libxcrypt is LGPLv2.1. Similar to #15757 for nlopt, we need to use the shared library or not use this library ...

Please open a separate issue for this.  It _seems_ like we're already shipping it dynamically (when peeking at the a recent whl file), but the build script compiles both shared and static.  To be cautious, it seems like we should not be building any static archive, so that we don't accidentally mis-use it.

https://github.com/RobotLocomotion/pip-drake/blob/57dccacfe02d5dc6906df0c19693b3c858b12d7f/image/dependencies/projects/libxcrypt.cmake#L9-L10

mwoehlke-kitware, 2021-09-29:
Fixed by RobotLocomotion/pip-drake#20.

I'm not sure why static libxcrypt is enabled; I'll look into turning that off.

closed 2021-09-29 by mwoehlke-kitware