RobotLocomotion/drake #15633: pip install: Workaround LCM error on import pydrake.all

BetsyMcPhail, 2021-08-19:
`import pydrake.all` is currently failing with error:

```
root@a63705f6c01a:/build-wheel# python3 -c 'import pydrake.all'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/build-wheel/pydrake/all.py", line 35, in <module>
    from .perception import *
ImportError: ImportError: ImportError: ImportError: ModuleNotFoundError: No module named 'drake'
```

On [Slack ](https://drakedevelopers.slack.com/archives/C028F9T6MSM/p1628197575035800?thread_ts=1628196180.033800&cid=C028F9T6MSM), @jwnimmer-tri noted that:

Great!  The import drake is due to
... "These lcm messages remain a problem:    drake.lcmt... lcmtypes"_ ...
https://github.com/RobotLocomotion/drake/issues/1183#issuecomment-858702715
https://github.com/RobotLocomotion/drake/issues/14362

This issue is to try to work around the LCM error and see if any other errors appear

Working towards #1183

BetsyMcPhail, 2021-09-01:
Using this [patch](https://github.com/RobotLocomotion/pip-drake/commit/702891fffcce528af7963770b6c6f850253ae80f) got past the LCM errors.

I built the wheel on the Docker image. Then pip installed drake on the same image.

```
 python3 -c 'import pydrake.all'
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/pydrake/manipulation/simple_ui.py", line 7, in <module>
    import tkinter as tk
ModuleNotFoundError: No module named 'tkinter'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/pydrake/all.py", line 48, in <module>
    from .manipulation.all import *
  File "/usr/local/lib/python3.6/dist-packages/pydrake/manipulation/all.py", line 4, in <module>
    from .simple_ui import *
  File "/usr/local/lib/python3.6/dist-packages/pydrake/manipulation/simple_ui.py", line 9, in <module>
    import Tkinter as tk
ModuleNotFoundError: No module named 'Tkinter'
```

I tried to manually install the missing package:

`sudo apt-get install python3-tk `

```
root@94aa8068945d:/wheel/dist# python3 -c 'import pydrake.all'
root@94aa8068945d:/wheel/dist#
```

BetsyMcPhail, 2021-09-01:
FYI @mwoehlke-kitware - looks like we may have an additional dependency to deal with?

mwoehlke-kitware, 2021-09-01:
It's not a build dependency... I presume someone that does `pip install drake` will run into this? That would imply that `setup.py` needs to change, not the provisioning...

BetsyMcPhail, 2021-09-01:
Note on the [drake-python3.7]( https://github.com/RobotLocomotion/drake-python3.7/commit/6c65b8c3bb8ffb850cf94eaa9c49734555261c56) branch, targets that depend on tkinter were removed

BetsyMcPhail, 2021-09-02:
Starting with Python 3.7, Tkinter is included with the Python installation. We shouldn't have to worry about this on Colab.

jwnimmer-tri, 2021-09-07:
> Did commenting out the lcmtypes bindings end up working?

It seems like the answer was "yes".  Is that accurate?

If yes, that means we could ship the initial wheels without cross-language LCM framework ports.

BetsyMcPhail, 2021-09-07:
Yes, [this patch](https://github.com/RobotLocomotion/pip-drake/pull/12) allowed us to move past the LCM errors and successfully run `import pydrake.all`. (I haven't done any testing beyond that)

At the meeting last week we had the outstanding question, does applying the patch and not calling `BindCppSerializers()` leave us with a usable solution?

jwnimmer-tri, 2021-09-07:
It's sufficient to start shipping wheels.  Eventually we will need to fix it, but I don't think its a blocker.

BetsyMcPhail, 2021-09-14:
The workaround has been merged into pip-drake.

closed 2021-09-14 by BetsyMcPhail