RobotLocomotion/drake #15634: pip install: Acceptance tests

BetsyMcPhail, 2021-08-19:
From [Slack](https://drakedevelopers.slack.com/archives/C028F9T6MSM/p1628777843040700):

References to some good pydrake acceptance tests that we should run on the whl.

As in, manually copy these py files into some test image that has the whl installed, and python3 my_test_prog.py.
No need to automate any testing yet.
https://github.com/RobotLocomotion/drake-external-examples

https://github.com/RobotLocomotion/drake-external-examples/blob/main/drake_cmake_installed/src/find_resource/find_resource_example.py

https://github.com/RobotLocomotion/drake-external-examples/blob/main/drake_cmake_installed_apt/src/particle_test.py

Working towards #1183

jwnimmer-tri, 2021-09-23:
The victory condition we want is that the three of us can all run the same small suite of acceptance tests (import, getDrakePath, etc.).  Does not need to be 100% automated.  Will not run under CI.

jwnimmer-tri, 2021-11-04:
I think I'm happy to call this complete?  Please speak up if there's anything important we missed.  We have an import-all test (that covers loading) and a getDrakePath test (that covers resource-finding).  Automating these tests in CI is covered by the #15960 ticket.

closed 2021-11-04 by jwnimmer-tri