RobotLocomotion/drake #15635: pip install: Document pypi publication process

BetsyMcPhail, 2021-08-19:
Working towards #1183

mwoehlke-kitware, 2021-08-19:
So... I'm not even sure if there's a task here. Here's the complete publication process:
```
twine upload some.whl
```
I've been using the drake-robot account, the credentials for which are on AWS. Long term, however, whoever is going to "own" this should possibly have a personal account on PyPI.

jwnimmer-tri, 2021-08-19:
(1) We need a log notebook of how all of the _initial_ setup for the Drake pypi project occurred, i.e., how we claimed the name, any settings we changed, which account(s) have permissions, etc.  Maybe it's small, but it's non-zero.

(2) Credentials should live in the LastPass vault.  Toffee can help facilitate that.

(3) Even if the instructions are a one-liner, we still need to write them down.  Possibly the https://drake.mit.edu/release_playbook.html will be a good place.  However, I doesn't seem like its a one-liner, because we _also_ need to document how the whl file(s) should be created, prior to calling `twine`.

The victory condition for the issue is that a linux-capable core drake developer can be given a hyperlink (e.g., that playbook link) and follow the documented steps, and by the end there's a new whl on pypi.  They should not need to search any tips on the internet, or deduce how anything is supposed to work.

To the extent the document says "run this script" and we encode the steps into code instead of docs, all the better.

Given the prototyping nature of the current whl, we can't finish step (3) until we've ironed out all of the problems; maybe we can't even start it.   For (1) and (2), though, they seem unblocked to do anytime.

jwnimmer-tri, 2021-08-19:
For (1) we have a "Drake Build & Release" google drive, which might be a good home.  I'll make sure you're invited to that, in case you're not already.

mwoehlke-kitware, 2021-08-19:
> how we claimed the name

Upload something to it.

> any settings we changed

None, so far. The first uploader becomes the initial owner who, in our case, should probably add other owners. (OTOH, that's drake-robot, which is usable by multiple people. OTGH, someone probably ought to change the password to that — to anything with shared access, really — any time someone leaves the group.)

> which account(s) have permissions

This is probably going to be fluid.

Should this start with "create an e-mail account for the robot"? (If "yes", I didn't do that, so don't know how it was done...)

jwnimmer-tri, 2021-08-19:
I don't even know what `drake-robot` is, so explaining that in the document would be a start.  (Or linking to another document, in case the term is already defined elsewhere and I've missed it.)

mwoehlke-kitware, 2021-08-19:
https://pypi.org/user/drake-robot/

mwoehlke-kitware, 2021-08-24:
Here's an attempt. I'm not sure where this should "live" (new document?).

----

1. Build the wheels using the `build-wheels.sh` script from [pip-drake](https://github.com/RobotLocomotion/pip-drake). This will copy the built wheels to the current directory on the host machine.
2. Upload the wheels using `twine upload some.whl` on a machine with a reasonably recent twine (Bionic is too old) using a PyPI account that has suitable permissions for [the PyPI project](https://pypi.org/project/drake), e.g. [drake-robot](https://pypi.org/user/drake-robot/) (credentials are on AWS).

jwnimmer-tri, 2021-09-07:
> I'm not sure where this should "live" (new document?).

To capture the initial setup (the (1) part), I think a new google doc in the Build & Release shared drive would work well.

To explain release activitives (the (3) part), I think we can worry about the final documentation location and fine-tuning once we have more of the supporting code merged to master.  However, I suspect that https://drake.mit.edu/release_playbook.html will be the best place for any portions that are "do these steps for every numbered release".  For now, all we really need to do for this issue is to make sure to keep track of those pieces sufficiently well such that we collect them later, and to make sure that anyone we ask to test the "build a wheel" scripts knows what to run.

jwnimmer-tri, 2021-09-09:
For (1), there is a draft up at https://docs.google.com/document/d/17D0yzyr0kGH44eWpiNY7E33A8hW1aiJRmADaoAlVISE/edit for review.

jwnimmer-tri, 2021-09-16:
I will review the document.

jwnimmer-tri, 2021-09-23:
I reviewed the document and left comments inline, though I didn't try running anything yet.  It'll suffice as a snapshot of what we're doing now.

Once I start running things to attempt a beta release, I'll rework it to be more of a boring recipe to follow in the flavor of https://drake.mit.edu/release_playbook.html.