RobotLocomotion/drake #15636: pip install: Upstream drake.patch

BetsyMcPhail, 2021-08-19:
Link to current patch https://github.com/RobotLocomotion/pip-drake/blob/master/image/pip-drake.patch

[Slack conversation ](https://drakedevelopers.slack.com/archives/C028F9T6MSM/p1629215811045000)

Working towards #1183

jwnimmer-tri, 2021-09-07:
@BetsyMcPhail and/or @mwoehlke-kitware please take the branch I noted in the prior message and get it tested and merged.  You can push to my branch, or just fork your own; either is fine by me.

mwoehlke-kitware, 2021-09-08:
It needs additional changes:
```diff
diff --git a/tools/workspace/vtk/repository.bzl b/tools/workspace/vtk/repository.bzl
index 3473dcd42..3caf67fa5 100644
--- a/tools/workspace/vtk/repository.bzl
+++ b/tools/workspace/vtk/repository.bzl
@@ -639,6 +639,11 @@ licenses([
         ],
     )
 
+    if os_result.is_manylinux:
+        vtk_glew_library = ":vtkglew"
+    else:
+        vtk_glew_library = "@glew"
+
     file_content += _vtk_cc_library(
         os_result,
         "vtkRenderingOpenGL2",
@@ -664,7 +669,7 @@ licenses([
             ":vtkCommonTransforms",
             ":vtkRenderingCore",
             ":vtksys",
-            "@glew",
+            vtk_glew_library,
             "@opengl",
         ],
     )
@@ -687,7 +692,13 @@ licenses([
         deps = ["@zlib"],
     )
 
-    file_content += _vtk_cc_library(os_result, "vtksys")
+    if os_result.is_manylinux:
+        file_content += _vtk_cc_library(os_result, "vtksys", linkopts=['-ldl'])
+
+        file_content += _vtk_cc_library(os_result, "vtkglew")
+
+    else:
+        file_content += _vtk_cc_library(os_result, "vtksys")
 
     # Glob all files for the data dependency of //tools:drake_visualizer.
     file_content += """
```

Does that look reasonable?

(I'm also testing on my multi-python branch which needs additional changes, but I'm not sure how to do some of them in a selectable manner. Also, I think it makes sense those should be at least a separate commit, if not separate PR.)

jwnimmer-tri, 2021-09-08:
> Does that look reasonable?

Yes.  (Once the full PR is ready for code review, then we can nitpick any details.)

> (I'm also testing on my multi-python branch which needs additional changes, but I'm not sure how to do some of them in a selectable manner. Also, I think it makes sense those should be at least a separate commit, if not separate PR.)

Agreed.

jwnimmer-tri, 2021-09-08:
Also just to be clear, the goal here is to smooth out the interaction between `drake` and `pip-drake` repositories.  To the extent that the patch is large and touches many files, it will likely suffer merge conflicts over time.  We should work to upstream the changes that are most likely to cause merge conflicts, and/or those most likely to require ongoing maintenance within Drake itself.  It's fine if we retain a small `drake.patch` file within `pip-drake`, if that is the path of least resistance.

jwnimmer-tri, 2021-09-09:
Once #15667 lands, _and_ we have rebased `pip-drake` to build against it, we can close this issue.

closed 2021-09-23 by mwoehlke-kitware