RobotLocomotion/drake #15638: Rename FixedOffsetFrame::SetPoseInBodyFrame() and add GetFixedPoseInParentFrame()

calderpg-tri, 2021-08-19:
From conversation with @joemasterjohn-TRI, `FixedOffsetFrame::SetPoseInBodyFrame()` actually sets the pose in the _parent_ frame, not the body frame, and should be named`SetPoseInParentFrame()` (and corresponding documentation update).

To effectively use `SetPoseInParentFrame()`, a method `GetFixedPoseInParentFrame()` should be added so that the pose in parent frame can be easily determined, rather than using `GetFixedPoseInBodyFrame()` and needing to know the body->parent frame transform.

Assigned high priority as this functionality is used by online calibration at TRI, and a recent switch from having camera frames relative to body frames to relative to other fixed offset frames turned up this issue.

jwnimmer-tri, 2021-11-12:
Ping @joemasterjohn?

jwnimmer-tri, 2022-01-10:
Searching Drake for "priority: high" and landed here.  This still seems like a serious problem, where the function is not doing what it claims to do?  What is the plan here?

\CC @sherm1

joemasterjohn, 2022-01-10:
Sorry @jwnimmer-tri, missed the previous ping. I'll have a patch in for this shortly.

jwnimmer-tri, 2022-05-16:
@joemasterjohn @rpoyner-tri re-ping?  I just found this TODO in Anzu again today.

hongkai-dai, 2022-10-06:
Any update on this issue? I would say that this function name is a landmine. I thought that if I call
```cc
frame.SetPoseInBodyFrame(frame.CalcPoseInBodyFrame(context));
```
nothing should change.

sherm1, 2022-10-06:
Paul @mitiguy can you take a look at this?

mitiguy, 2022-10-26:
Starting this now.

mitiguy, 2022-10-27:
PR #18198 resolves issue #15638.

closed 2022-11-08 by mitiguy