RobotLocomotion/drake #15652: build error

edrumwri, 2021-08-23:
Latest install_prereqs.sh has been run, this is on Ubuntu 20:

```
└─ $ ▶ bazel build -c dbg manipulation/util:geometry_inspector
Extracting Bazel installation...
Starting local Bazel server and connecting to it...
INFO: Analyzed target //manipulation/util:geometry_inspector (173 packages loaded, 8271 targets configured).
INFO: Found 1 target...
ERROR: /home/drum/drake-new/bindings/pydrake/BUILD.bazel:57:37: Action bindings/pydrake/documentation_pybind.h failed: (Exit 1): mkdoc failed: error executing command bazel-out/host/bin/tools/workspace/pybind11/mkdoc -DDRAKE_COMMON_SYMBOLIC_DETAIL_HEADER -DEIGEN_MPL2_ONLY -DHAVE_CSTDDEF '-DFMT_HEADER_ONLY=1' '-DFMT_NO_FMT_STRING_ALIAS=1' -DSPDLOG_COMPILED_LIB ... (remaining 743 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
external/eigen/include/_usr_include_eigen3/Eigen/Core:82:12: fatal error: 'new' file not found
Traceback (most recent call last):
  File "/home/drum/.cache/bazel/_bazel_drum/2059ccae995efcc54d793ceb88203d5b/sandbox/linux-sandbox/1197/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 841, in <module>
    main()
  File "/home/drum/.cache/bazel/_bazel_drum/2059ccae995efcc54d793ceb88203d5b/sandbox/linux-sandbox/1197/execroot/drake/bazel-out/host/bin/tools/workspace/pybind11/mkdoc.runfiles/drake/tools/workspace/pybind11/mkdoc.py", line 801, in main
    raise RuntimeError(
RuntimeError: Parsing headers using the clang library failed with 0 error(s) and 1 fatal error(s)
----------------
Note: The failure of target //tools/workspace/pybind11:mkdoc (with exit code 1) may have been caused by the fact that it is running under Python 3 instead of Python 2. Examine the error to determine if that appears to be the problem. Since this target is built in the host configuration, the only way to change its version is to set --host_force_python=PY2, which affects the entire build.

If this error started occurring in Bazel 0.27 and later, it may be because the Python toolchain now enforces that targets analyzed as PY2 and PY3 run under a Python 2 and Python 3 interpreter, respectively. See https://github.com/bazelbuild/bazel/issues/7899 for more information.
```

Looks like the error has something to do with Eigen, not python.

edrumwri, 2021-08-23:
Feel free to retitle this generically titled issue.

jwnimmer-tri, 2021-08-23:
If fact that `#include <new>` (a C++ standard library header) is failing indicates that something is seriously wrong with the compiler when extracting documentation comments.  It is not specific to Python nor Eigen; those are just the first thing it happens to hit.

I will see if I can reproduce the problem, but given that Drake CI is passing, this probably indicates a machine-local configuration setting that is somehow defeating our build.  If you can share details about your machine configuration, perhaps the can isolate the culprit, and then figure out how to avoid it.

(0) What revision of Drake?

(1) Are any of the environment variables `CC`, or `CXX`, or `LD_LIBRARY_PATH` set in the terminal that is running this build?

(2) What (if any) extra Ubuntu compiler packages are installed, beyond what Drake's install_prereqs uses (e.g., is `g++-8` installed?).

(3) Which version(s) of Clang are installed into `/usr/bin`?

(4) What (if any) Clang standard libraries are installed (Ubuntu packages whose names begin with `libc++`).

(4) Is anything installed into `/usr/local`?

edrumwri, 2021-08-24:
Thanks for the help! Please find responses inline below:

> (1) Are any of the environment variables `CC`, or `CXX`, or `LD_LIBRARY_PATH` set in the terminal that is running this build?
> 

No.

> (2) What (if any) extra Ubuntu compiler packages are installed, beyond what Drake's install_prereqs uses (e.g., is `g++-8` installed?).
> 

g++-7 and g++-9 are installed; latter is default.

> (3) Which version(s) of Clang are installed into `/usr/bin`?
> 

clang++-10, -9, -8, and -6. 10 is default.

> (4) What (if any) Clang standard libraries are installed (Ubuntu packages whose names begin with `libc++`).
> 

```
libc++1-10/focal,now 1:10.0.0-4ubuntu1 amd64 [installed,automatic]
libc++1/focal,now 1:10.0-50~exp1 amd64 [installed]
libc++abi1-10/focal,now 1:10.0.0-4ubuntu1 amd64 [installed,automatic]
```

> (4) Is anything installed into `/usr/local`?

Yes- apriltag library & tools, Franka Panda library (libfranka) & tools, LCM, librealsense2, opencv4, openmp, google benchmarks library, glog.

jwnimmer-tri, 2021-08-24:
I haven't been able to reproduce it yet, even after installing those packages.  (However, I wasn't able to install clang-6 -- that package is not available in Focal, as best I can tell.)

Here's a patch that will dump out the include paths in use while libclang is trying to extract documentation:

```patch
--- a/tools/workspace/pybind11/mkdoc.py
+++ b/tools/workspace/pybind11/mkdoc.py
@@ -787,6 +787,7 @@ def main():
             eprint("Parse headers...")
         index = cindex.Index(
             cindex.conf.lib.clang_createIndex(False, True))
+        parameters.extend(["-###"])
         translation_unit = index.parse(
             glue_filename, parameters,
             options=cindex.TranslationUnit.PARSE_DETAILED_PROCESSING_RECORD)
```

When I add that to my successful Focal build, here's the relevant subset of paths that appear.  I'm curious if you end up with different paths:

```
-resource-dir lib/clang/9.0.1
-internal-isystem /../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9
-internal-isystem /../lib/gcc/x86_64-linux-gnu/9/../../../../include/x86_64-linux-gnu/c++/9
-internal-isystem /../lib/gcc/x86_64-linux-gnu/9/../../../../include/x86_64-linux-gnu/c++/9
-internal-isystem /../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/backward
-internal-isystem /usr/local/include
-internal-isystem /usr/include/clang/9.0.1/include/
-internal-externc-isystem /usr/include/x86_64-linux-gnu
-internal-externc-isystem /include
-internal-externc-isystem /usr/include
```

edrumwri, 2021-08-24:
Ah, I think this points to the culprit:

```
-resource-dir lib/clang/9.0.1
-internal-isystem /usr/lib/gcc/x86_64-linux-gnu/10/../../../../include/c++
-internal-isystem /usr/lib/gcc/x86_64-linux-gnu/10/../../../../include/c++/x86_64-linux-gnu
-internal-isystem /usr/lib/gcc/x86_64-linux-gnu/10/../../../../include/c++/backward
-internal-isystem /usr/local/include
-internal-isystem /usr/include/clang/9.0.1/include/
-internal-externc-isystem /usr/include/x86_64-linux-gnu
-internal-externc-isystem /include
-internal-externc-isystem /usr/include
```

gcc-10 was installed on this machine. My apologies for not seeing this when looking for g++ packages.

jwnimmer-tri, 2021-08-24:
That's it.  It seems you have `gcc-10` installed, but not `g++-10`.  Clang always tries to use the most recent GCC when looking for library headers, so it sees 10 and chooses that.  It didn't realize that only the C headers were available in 10, but the C++ headers were missing.

I think if you either additionally install `g++10`, or else uninstall `gcc-10`, then you'll be fine.

I looked into this in more detail a few months ago when we had some GCC 8 problems on Bionic.  It seems that there is _no mechanism_ by which we can tell Clang to choose a different GCC.  So installing extra Ubuntu packages hazards the risk of breaking the Drake build.

jwnimmer-tri, 2021-08-24:
I think the action for Drake here is to try to produce a better error message.

edrumwri, 2021-08-24:
Agreed. Removing gcc-10 hasn't yet fixed it- the headers are sticking around using the patch you gave me- but I think I'm on my way to getting it corrected. Thanks for your help @jwnimmer-tri

jwnimmer-tri, 2021-08-24:
Ah, okay.  If it helps, these was the "undo" command I used locally, to remove the effects of `g++10` being installed:

`sudo apt remove cpp-10 g++-10 gcc-10 libasan6 libgcc-10-dev libstdc++-10-dev`

edrumwri, 2021-08-25:
That did it. Thanks Jeremy!

jwnimmer-tri, 2021-09-21:
FYI As of #15798, here's the error message that comes out on Focal if gcc-10 is installed without g++-10 also being installed.  Maybe it will help with the googles.

```
ERROR: .../bindings/pydrake/BUILD.bazel:57:37: Action bindings/pydrake/documentation_pybind.h failed: (Exit 1): mkdoc failed: error executing command bazel-out/host/bin/tools/workspace/pybind11/mkdoc -DDRAKE_COMMON_SYMBOLIC_DETAIL_HEADER -DEIGEN_MPL2_ONLY -DHAVE_CSTDDEF '-DFMT_HEADER_ONLY=1' '-DFMT_NO_FMT_STRING_ALIAS=1' -DSPDLOG_COMPILED_LIB ... (remaining 742 argument(s) skipped)

Use --sandbox_debug to see verbose messages from the sandbox
bazel-out/k8-opt/bin/bindings/pydrake/documentation_pybind.h.tmp_artifacts/mkdoc_glue.h:1:10: fatal error: 'optional' file not found
clang version 9.0.1-12 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: 
 "clang" "-cc1" "-triple" "x86_64-pc-linux-gnu" "-fsyntax-only" "-disable-free" "-disable-llvm-verifier" "-discard-value-names" "-main-file-name" "foo" "-mrelocation-model" "static" "-mthread-model" "posix" "-mdisable-fp-elim" "-fallow-editor-placeholders" "-fmath-errno" "-masm-verbose" "-mconstructor-aliases" "-munwind-tables" "-fuse-init-array" "-target-cpu" "x86-64" "-dwarf-column-info" "-debugger-tuning=gdb" "-resource-dir" "lib/clang/9.0.1" "-isystem" "lcmtypes" "-isystem" "bazel-out/k8-opt/bin/lcmtypes" "-isystem" "external/lcm" "-isystem" "bazel-out/k8-opt/bin/external/lcm" "-isystem" "external/lcm/lcm" "-isystem" "bazel-out/k8-opt/bin/external/lcm/lcm" "-isystem" "external/eigen/include/_usr_include_eigen3" "-isystem" "bazel-out/k8-opt/bin/external/eigen/include/_usr_include_eigen3" "-isystem" "external/fmt/include" "-isystem" "bazel-out/k8-opt/bin/external/fmt/include" "-isystem" "external/optitrack_driver/lcmtypes" "-isystem" "bazel-out/k8-opt/bin/external/optitrack_driver/lcmtypes" "-isystem" "external/clp/include/_usr_include_coin" "-isystem" "bazel-out/k8-opt/bin/external/clp/include/_usr_include_coin" "-isystem" "external/nlopt/include" "-isystem" "bazel-out/k8-opt/bin/external/nlopt/include" "-isystem" "external/ibex/opt/libibex/2.7.4/include" "-isystem" "bazel-out/k8-opt/bin/external/ibex/opt/libibex/2.7.4/include" "-isystem" "external/ibex/opt/libibex/2.7.4/include/ibex" "-isystem" "bazel-out/k8-opt/bin/external/ibex/opt/libibex/2.7.4/include/ibex" "-isystem" "external/ibex/opt/libibex/2.7.4/include/ibex/3rd" "-isystem" "bazel-out/k8-opt/bin/external/ibex/opt/libibex/2.7.4/include/ibex/3rd" "-isystem" "external/ipopt/include/_usr_include_coin" "-isystem" "bazel-out/k8-opt/bin/external/ipopt/include/_usr_include_coin" "-isystem" "external/vtk/include/vtk-8.2" "-isystem" "bazel-out/k8-opt/bin/external/vtk/include/vtk-8.2" "-isystem" "external/double_conversion/include" "-isystem" "bazel-out/k8-opt/bin/external/double_conversion/include" "-isystem" "external/libjpeg/include" "-isystem" "bazel-out/k8-opt/bin/external/libjpeg/include" "-isystem" "external/libpng/include/_usr_include_libpng16" "-isystem" "bazel-out/k8-opt/bin/external/libpng/include/_usr_include_libpng16" "-isystem" "external/libtiff/include/_usr_include_x86_64-linux-gnu" "-isystem" "bazel-out/k8-opt/bin/external/libtiff/include/_usr_include_x86_64-linux-gnu" "-isystem" "external/opengl/include" "-isystem" "bazel-out/k8-opt/bin/external/opengl/include" "-isystem" "external/yaml_cpp/include" "-isystem" "bazel-out/k8-opt/bin/external/yaml_cpp/include" "-iquote" "." "-iquote" "bazel-out/k8-opt/bin" "-iquote" "external/lcm" "-iquote" "bazel-out/k8-opt/bin/external/lcm" "-iquote" "external/eigen" "-iquote" "bazel-out/k8-opt/bin/external/eigen" "-iquote" "external/fmt" "-iquote" "bazel-out/k8-opt/bin/external/fmt" "-iquote" "external/optitrack_driver" "-iquote" "bazel-out/k8-opt/bin/external/optitrack_driver" "-iquote" "external/clp" "-iquote" "bazel-out/k8-opt/bin/external/clp" "-iquote" "external/nlopt" "-iquote" "bazel-out/k8-opt/bin/external/nlopt" "-iquote" "external/ibex" "-iquote" "bazel-out/k8-opt/bin/external/ibex" "-iquote" "external/libblas" "-iquote" "bazel-out/k8-opt/bin/external/libblas" "-iquote" "external/liblapack" "-iquote" "bazel-out/k8-opt/bin/external/liblapack" "-iquote" "external/zlib" "-iquote" "bazel-out/k8-opt/bin/external/zlib" "-iquote" "external/ipopt" "-iquote" "bazel-out/k8-opt/bin/external/ipopt" "-iquote" "external/vtk" "-iquote" "bazel-out/k8-opt/bin/external/vtk" "-iquote" "external/double_conversion" "-iquote" "bazel-out/k8-opt/bin/external/double_conversion" "-iquote" "external/liblz4" "-iquote" "bazel-out/k8-opt/bin/external/liblz4" "-iquote" "external/liblzma" "-iquote" "bazel-out/k8-opt/bin/external/liblzma" "-iquote" "external/libjpeg" "-iquote" "bazel-out/k8-opt/bin/external/libjpeg" "-iquote" "external/libpng" "-iquote" "bazel-out/k8-opt/bin/external/libpng" "-iquote" "external/libtiff" "-iquote" "bazel-out/k8-opt/bin/external/libtiff" "-iquote" "external/glew" "-iquote" "bazel-out/k8-opt/bin/external/glew" "-iquote" "external/opengl" "-iquote" "bazel-out/k8-opt/bin/external/opengl" "-iquote" "external/x11" "-iquote" "bazel-out/k8-opt/bin/external/x11" "-iquote" "external/spdlog" "-iquote" "bazel-out/k8-opt/bin/external/spdlog" "-iquote" "external/tinyxml2" "-iquote" "bazel-out/k8-opt/bin/external/tinyxml2" "-iquote" "external/yaml_cpp" "-iquote" "bazel-out/k8-opt/bin/external/yaml_cpp" "-D" "__MKDOC_PY__" "-D" "DRAKE_COMMON_SYMBOLIC_DETAIL_HEADER" "-D" "EIGEN_MPL2_ONLY" "-D" "HAVE_CSTDDEF" "-D" "FMT_HEADER_ONLY=1" "-D" "FMT_NO_FMT_STRING_ALIAS=1" "-D" "SPDLOG_CO [... cut ...] ilt in the host configuration, the only way to change its version is to set --host_force_python=PY2, which affects the entire build.

If this error started occurring in Bazel 0.27 and later, it may be because the Python toolchain now enforces that targets analyzed as PY2 and PY3 run under a Python 2 and Python 3 interpreter, respectively. See https://github.com/bazelbuild/bazel/issues/7899 for more information.
----------------
Target //bindings/pydrake:documentation_pybind.h failed to build
```

PR #15798 (2021-09-21): [mkdoc] Report a more specific error when libstdc++ is broken
Closes #15652.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/15798)
closed 2021-09-21 by jwnimmer-tri