RobotLocomotion/drake #15660: plant: Should provide more convenient access to scoped names

EricCousineau-TRI, 2021-08-24:
Aiming to focus on this after September.

From convo w/ @calderpg-tri, usage of `drake::multibody::parsing::GetScopedFrameByName` is non-discoverable, and could possibly live in `MultibodyPlant` itself in some form.
Personally, I'm ambivalent, but it would make easier migration of use sites.

calderpg-tri, 2021-08-24:
To expand on our conversation, if we intend scoped frame and body names to be first-class queries, they should be as discoverable as the current `Mbp::Get{Body,Frame}ByName` methods, rather than being shunted off somewhere else.

A related topic in this conversation is that scoped frames are perhaps most important in terms of interop with other tools, namely ROS, which use string frame names. For any complex model with non-unique body/frame names, scoped names are the only way to use string-only names. Right now, in Drake and Anzu we have a set of ad-hoc {Body, Frame}<->scoped name and scoped name<->ROS-compatible name methods, but in the future we'll want a consistent set of these methods available in Drake for the community to use.

jwnimmer-tri, 2022-05-02:
I believe @rpoyner-tri has been poking at scoped names recently, in terms of collision filtering (#14785).  For my part, I think the most important facet of anything related to scoped names is a clear and bulletproof specification for what they denote.