RobotLocomotion/drake #15695: Update setup.py to not hard-code the Drake version.

mwoehlke-kitware, 2021-09-01:
Currently, `setup.py` hard-codes the Drake version. Given that this is already out of date, and a presumably-unnecessary maintenance burden, we should add logic to obtain the version number from the Drake build instead.

(At least for now, however, we should also have a nice way to tack on an alpha/beta build number; e.g. as of writing, this would be `'a4'`.)

jwnimmer-tri, 2021-09-02:
Related to #14509.

mwoehlke-kitware, 2021-12-29:
This was (mostly) addressed by #16139. Although that requires passing the *complete* version when building the wheels, it is unclear if we have any intention of ever extracting partial version information from the repository, so perhaps this should be closed.

jwnimmer-tri, 2022-01-04:
Yes, this is good enough for now.  The version number is not hard-coded anymore.

closed 2022-01-04 by jwnimmer-tri